bigtreetech/BIGTREETECH-SKR-mini-E3 #2: SKR Mini E3 USB Connection

ilakuli, 2019-07-08:
Hello,

My SKR Mini E3 keeps disconnecting from any USB Device after about 15 seconds. Tested on Windows PC and Octoprint. In Windows, the board shows up and connects normally but after 15 seconds it just disconnects. In Octoprint the same thing happens. Firmware was configured and compiled according to the List provided on Github. Have noticed the issue with conflicting serial port information and tried both ways. Nothing solved the issue. Any help would be appreciated.

Thanks.

vyruz1986, 2019-07-08:
I can confirm the same behavior on my E3 Mini board with the latest build from Marlin directly (at commit 2b1c2a0db)

Msq001, 2019-07-09:
I will test it

Msq001, 2019-07-09:
Can you provide firmware. bin file? I tested the latest version of marlin, no such behavior, and I uploaded a workable firmware. bin in [here](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/tree/master/firmware), you can try to use this to see if it works properly.

ilakuli, 2019-07-09:
[Firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/files/3371168/Firmware.zip)
attached is the firmware.bin.

I will test your firmware but I cannot use it permanenlty since my Ender 3 hast a different hotend and auto bed leveling.

Thanks,

ilakuli, 2019-07-09:
Your firmware resolved the issue. Can you provide the source code so I can add auto bed level and compile myself?

Msq001, 2019-07-09:
this firmware.bin is compiled from old version Marlin, and the source code have be replaced by new verison,  this issue maybe the Marlin's bug,  I will locate and fix it as soon as possible,  when I fix this, I will tell you and  upload the workable source code in here.

vyruz1986, 2019-07-09:
Could everyone please state which commit of marlin they are building and
testing?

I have this issue with commit 2b1c2a0db

Will try later today with the bin file provided by Msq001, thanks

vyruz1986, 2019-07-09:
Here's a zip (with firmware.bin , github doesn't allow *.bin uploads) which demonstrates the problem.
[firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/files/3374956/firmware.zip)

It was compiled by setting `#define SERIAL_PORT_2 -1` and with marlin at commit ab99a807

Msq001, 2019-07-10:
hi~  I have update source code, LCD12864 is ok, usb connect to printrun is ok

Msq001, 2019-07-10:
It's  just download form Latest version https://github.com/MarlinFirmware/Marlin/commit/a23021b823477b907f8b5e658afc3bbb3165be31

vyruz1986, 2019-07-10:
Below are 2 pictures comparing `configuration.h` and `configuration_adv.h` between the source code in this repository(https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/tree/2a167ef96c9170c63288a3c1d69b6dea9d913506/firmware/Marlin-2.0.x-SKR-Mini-E3), and MarlinFirmware/Marlin@a23021b , as you can see there are way more changes than what you currently have described in the instructions on how to flash Marlin. Likely if we apply all these changes to the latest Marlin sources, it will work as well. I'll give that a try tonight and report back here.
**Configuration_adv.h**
[image]

**Configuration.h**
[image]

vyruz1986, 2019-07-10:
Ok so after testing, I've done the following and have a completely working Ender 3 with the BBT SKR E3 Mini board.

1. Clone the latest version of Marlin (I used MarlinFirmware/Marlin@4e479caf61fde35121351e33276aed673a365b7f)
2. In the repo you just cloned, copy the contents of `config\examples\Creality\Ender-3\` to `Marlin`
3. Make the changes described below to `Configuration.h` and `Configuration_adv.h`

### Configuration.h

```c++
#define STRING_CONFIG_H_AUTHOR "(none, default config)" // Who made the changes.
#define SERIAL_PORT 2
#define SERIAL_PORT_2 -1
  #define MOTHERBOARD BOARD_BIGTREE_SKR_MINI_E3
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define E0_DRIVER_TYPE TMC2209
```

### Configuration_adv.h

```c++
  #define  Y_SLAVE_ADDRESS 1
  #define  Z_SLAVE_ADDRESS 2
  #define E0_SLAVE_ADDRESS 3
```

This works for my printer.

vyruz1986, 2019-07-10:
Apparently I cheered too early :) The control knob still isn't working. Printing over USB works fine, but pressing the control button causes the board to freeze up :(

Msq001, 2019-07-11:
> Apparently I cheered too early :) The control knob still isn't working. Printing over USB works fine, but pressing the control button causes the board to freeze up :(
> 
> Edit: Also the USB connection is still dropping after ~15 seconds, so this specific issue isn't really fixed it seems.

this issue please comment out `#define SPEAKER` in configuration.h file

Msq001, 2019-07-11:
and I wil update your feedback in readme.md,  thanks for your feedback !

vyruz1986, 2019-07-12:
Control knob issue was fixed by commenting out `#define SPEAKER` as suggested by @Msq001 , thanks!

Xenon-2510, 2019-07-14:
Any update on this issue? I have got the board up and running but have an issue with the X axis stepper driver and cannot debug as the serial connection doesn't work.

Msq001, 2019-07-18:
> Any update on this issue? I have got the board up and running but have an issue with the X axis stepper driver and cannot debug as the serial connection doesn't work.

Have you installed Maple USB port driver on your computer? Can you debug with this [firmware. bin](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/blob/master/firmware/firmware.bin)

Xenon-2510, 2019-07-18:
I will check this firmware out later today, I have installed the Maple driver on my pc but I am also trying to use Rpi3 with Octoprint. I will feedback the outcome.

Xenon-2510, 2019-07-18:
Hi,

So i tried the firmware you suggested and unfortunately it did not work, I then tried a firmware that I was sent through facebook earlier this week by Zyfio (see attached) this worked fine with the serial connection both from my PC using pronterface and my Rpi3 using Octoprint. So i know that the board works and that it is a firmware issue, I have attached the working firmware for you to try but I am afraid I do not have the source.

I am hoping that this is of some use, please let me know if there is anything else that I can do to support.

[firmware_working_serial.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/files/3408044/firmware_working_serial.zip)

Msq001, 2019-07-19:
Hi， Even if you updated the firmware. bin that we have compiled here, the serial port is still unrecognizable?

Msq001, 2019-07-19:
SO SORRY for this issue, USB Serial can't recognize or unstable because of: USB slave needs to connect a pull-up resistor to VCC in D + to tell USB host it's own device type, we use a P-MOS to control whether the 1.5k resistance is pulled up or not, and the GPIO of P-MOS should "output" a stable low level, but It was "input" state before, so the USB device could not recognize or be unstable.
I have updated the firmware.bin and source code now, It should workable now

nicstoun, 2019-07-19:
Hi, i think soon buy this board ..i track this issues ,still board don't have stable firmware for everyday use ? 

Regards

Msq001, 2019-07-19:
the SD card print, and usb serial is ok now in my tested,  but I need more feedback, this source code and firmware.bin should workable and I have Creat a PR to fix this issue in Marlin master branch https://github.com/MarlinFirmware/Marlin/pull/14679

nicstoun, 2019-07-19:
> the SD card print, and usb serial is ok now in my tested, but I need more feedback, this source code and firmware.bin should workable and I have Creat a PR to fix this issue in Marlin master branch [MarlinFirmware/Marlin#14679](https://github.com/MarlinFirmware/Marlin/pull/14679)

Ok, for that i see only usb have some issues now , if flash firmware provided  later i get board all be fine right ?

Xenon-2510, 2019-07-19:
I have tried the source firmware this afternoon, seem to be getting inconsistent E1 Thermal runaway issues on boot. Do you need to add in a grace period for this board?

Msq001, 2019-07-19:
@Xenon-2510 This seems to happen only when you connect to USB CDC serial at startup, right? If wait  4-5s for motherboard to startup completed, then connecting USB CDC Serial is normal ?

Msq001, 2019-07-19:
> Ok, for that i see only usb have some issues now , if flash firmware provided later i get board all be fine right ?

yes, It should be fine.

Xenon-2510, 2019-07-19:
@Msq001 - yes, it seems to only do it when the printer is turned on with the USB plugged in. Is there anyway this can be rectified as this didn't happen with the stock board? I like to leave Octoprint connected so I can remotely turn the print on and start a print from the Web interface.

vyruz1986, 2019-07-19:
I tried building the latest bigtreetech/Marlin repository, and can confirm the USB issue is fixed 👍 
Thanks BTT!

Msq001, 2019-07-20:
@Xenon-2510  Please wait 4-5s, after motherboard startup successed, then connect USB CDC Sereial will be ok,  if you don't want to wait,  Please uncomment this
`#define BOGUS_TEMPERATURE_FAILSAFE_OVERRIDE ` in here, 
 https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/blob/1016c47352a5a30e4ea25dda2652418618295655/firmware/Marlin-2.0.x-SKR-Mini-E3/Marlin/src/pins/pins_BIGTREE_SKR_MINI_E3.h#L35
this is just a temporary solution.
I will continue to debug this issue, and when I find the final solution, I will update it.

Msq001, 2019-07-20:
@vyruz1986 Thanks for your feedback and patience！

nicstoun, 2019-07-20:
> yes, It should be fine.
Can you open theme to explain in details all options for this board. I read for DIAG   instead of Endstop,
 blocking rotation detection  ,that sound interesting but not find info in manual. 
Thank you!

ilakuli, 2019-07-23:
Just to give some Feedback: My Mini E3 is now fully running with working USB Connection so Issue solved. Let me know whether I should close this issue or keep it open.

nicstoun, 2019-07-24:
What about Eeprom settings i read this chip not support this options ? How safe calibrated settings ?

Msq001, 2019-07-24:
it can be saved in SD card file  "eeprom.dat".

nicstoun, 2019-07-24:
> it can be saved in SD card file "eeprom.dat".

Ok, so that is solution now

Msq001, 2019-07-24:
yeah. we are considering to modify stm32f103rc to stm32f103re. it have 512KB flash. can to emulation eeprom.  but not confirm yet

nicstoun, 2019-07-24:
> yeah. we are considering to modify stm32f103rc to stm32f103re. it have 512KB flash. can to emulation eeprom. but not confirm yet

Good to know , hope this happened soon :)

reloxx13, 2019-07-31:
dont close this if its not fixed!

a workaround is not a fix!

reloxx13, 2019-08-02:
@bigtreetech pls reopen

thomasgerman, 2019-08-12:
why can not I connect the board to usb?  I loaded the last marlin, when I connect the cable hangs up the board and the hotend fan whistles.  where is the problem?

mejtoogood, 2019-08-17:
Board still gives thermal runaway error when USB connected at boot. Not prepared to use workaround which degrades safety. Fix was in pipeline 24 days ago, what is status?

bigtreetech, 2019-08-22:
look here https://github.com/MarlinFirmware/Marlin/pull/15026, we have fixed the usb cdc connection bug of  temperature. and have update source code

thomasgerman, 2019-08-22:
Must i all uint_16 Switch to uint_8?
Or what?

Gesendet mit der WEB.DE iPhone App

reloxx13, 2019-08-23:
> Must i all uint_16 Switch to uint_8? Or what? Gesendet mit der WEB.DE iPhone App Am 22.08.19 um 09:15 schrieb BigTreeTech
> […](#)

click on the link and read the changes.

petrik0621, 2019-08-28:
> @petrik0621 <https://github.com/petrik0621>
>
>    1. first, download this folder.
>    2. After you update the normal firmware, insert motherboard USB into
>    PC, You should see a maple device with a yellow exclamation mark in the
>    Device Manager
>    3. Right-click for Update Driver Software.
>    4. Select Browse PC Local Files for Update Driver
>    5. Select the folder has be download.
>    6. Start update driver...

@Msq001 I can just open the folder, there is no download button.

Msq001 <notifications@github.com> ezt írta (időpont: 2019. aug. 28., Sze,
7:55):

petrik0621, 2019-08-28:
Hi. I need the Maple USB driver for the PC can anyone send me pls? I can't find it.

petrik0621, 2019-08-28:
@Msq001  thank, I found that too, but i don't know how to instal it :( I'm new

petrik0621, 2019-08-28:
@Msq001 I can just open the folder, there is no download button.

petrik0621, 2019-08-28:
@Msq001 [image]
[image]

petrik0621, 2019-08-28:
@Msq001 ok thanks but what now?
I have Windows 8 :(
[image]

reloxx13, 2019-08-28:
google "windows 8 install driver without digital signature"

reloxx13, 2019-08-28:
@bigtreetech u can close this now

Msq001, 2019-08-28:
@petrik0621 https://github.com/bigtreetech/BIGTREETECH-SKR-MINI-V1.1/tree/master/firmware/mapleDrv-USBdriver

Msq001, 2019-08-28:
@petrik0621
1. first,  download this folder.
2. After you update the normal firmware, insert motherboard USB into PC, You should see a maple device with a yellow exclamation mark in the Device Manager
3. Right-click for Update Driver Software.
4. Select Browse PC Local Files for Update Driver
5. Select the folder has be download.
6. Start update driver...

Msq001, 2019-08-28:
I upload folder in here too https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/tree/master/firmware .
you can download the Whole repo
[image]

closed 2019-09-07 by bigtreetech

Silversurfer79, 2019-11-29:
Maybe a silly question, is there a bin file that i can load from SD or do i need have the SKR connect via USB to Adr or a PC to do this?

Silversurfer79, 2019-12-01:
?

NotNiki, 2020-05-22:
Hi, I connect the USB to a pc/octoprint it works but when i sent M305 for PID autotune my connection will always close and won't open again. Does anybody know what to do? Thanks, Adam.

[1 more reply]