bigtreetech/BIGTREETECH-SKR-mini-E3 #3: SKR Mini E3 display connection

vyruz1986, 2019-07-08:
Hi,

Which connection should we use on the original Ender3 12864 LCD display?
If I use EXP3 (same as default on ender3) I get backlight with blank screen, button makes clicking sound.
If I use EXP1 & EXP2 I get no backlight, and sometimes if I click the backlight comes on very briefly and it makes a silent clicking sound.

Msq001, 2019-07-09:
should use EXP3, i will test it

Msq001, 2019-07-09:
I have tested. not work for lcd12864， It seems a marlin bug, I will locate and fix it as soon as possible. and I have uploaded a firmware.bin for skr mini e3，It is configured according to Ender3, you can user this .bin before I fixed this issue

Msq001, 2019-07-10:
I have uploaded source code in this repository. you can try it

nicstoun, 2019-07-29:
> I have uploaded source code in this repository. you can try it

I wait my E3 mini, what is stable version that i flash on ,any recommendation ?   I don't use BL touch 
Thank you!

chodenut, 2019-10-13:
I received and installed my Mini E3 board today and have this issue still (Version 1.2).  I have a stock Ender 3 and with the ribbon cable for the display attached, there is no light on the board.  When I disconnect the light, the red light comes on the board.  I downloaded the latest firmware.bin file from here under 1.2 and the second red light flashed after a moment, indicating firmware had been loaded.  I then powered off and reattached the ribbon cable for the display.  After powering on the display still did not function and no lights on the board came on.

TheDude1970, 2019-10-29:
I received and installed my Mini E3 board today and have this issue still (Version 1.2). I have a stock Ender 3 and with the ribbon cable for the display attached, there is no light on the board. When I disconnect the light, the red light comes on the board. I downloaded the latest firmware.bin file from here under 1.2 and the second red light flashed after a moment, indicating firmware had been loaded. I then powered off and reattached the ribbon cable for the display. After powering on the display still did not function and no lights on the board came on.

I have the same problem

chodenut, 2019-10-29:
I connected the same board to another ender 3 with even worse results.  At the suggestion of support, i connected only the main power and the display ribbon cable.  The damn thing worked!  I then connected each plug or wire pair one at a time, power cycling to confirm it still worked. I find it questionable that this technique worked at the least.  Of note is that i only had the center screw securing the board while going through this process.  I am thinking that one of the screws moght be shorting board layers or touching a pad on the board causing the issue.  Regardless, the board is installed and working for now.  Test prints are good and we are back on line.  Hope you get yours working too!

DankInstitute, 2019-10-30:
I am experiencing this as well! I have 2 different SKR mini E3 boards and cannot for the life of me get them to display anything. I have tried the display EXP 1 and EXP3 and can get the screen to light up but no text.