bigtreetech/BIGTREETECH-SKR-mini-E3 #4: Marlin Configuration.h needs additional changes

vyruz1986, 2019-07-09:
After following your instructions [here](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/blob/master/firmware/readme.md) I still had to change the following lines to get the firmware to work correctly:
```c++
...
#define INVERT_X_DIR true //Change false to true
...
#define DEFAULT_AXIS_STEPS_PER_UNIT { 80, 80, 400, 93 } //Change 4000 to 400
```

Without these changes, the X-axis is moving in wrong direction, and steps/mm for Z-axis are way too much (1mm = 1cm movement)

jacek-ortiz-zocdoc, 2019-07-11:
Did you copy the configs from the Marlin config/examples folder for your printer? My settings seemed to be correct for the Ender 3.

jaceknyc, 2019-07-11:
Did you copy the configs from the Marlin config/examples folder for your printer? My settings seemed to be correct for the Ender 3.

reloxx13, 2019-07-16:
this is also in the origin marlin repo
https://github.com/MarlinFirmware/Marlin/issues/14594

Rene046, 2019-07-23:
Strange i get an error "Environment BIGTREE_SKR_MINI            [FAILED]"

syampolsky, 2019-11-28:
For Ender-5, settings from Marlin/examples folder are the opposite of what's needed. Here's what it looks like:
`#define INVERT_X_DIR false`
`#define INVERT_Y_DIR false`
`#define INVERT_Z_DIR true`