bigtreetech/BIGTREETECH-SKR-mini-E3 #6: BLTouch Configuration

Xenon-2510, 2019-07-21:
I am using a BLTouch clone in my setup and I am not seeing a trigger using the "Probe" connector. Is the something that I need to configure? I have tried it in the Z-Min connector and it works as expected although not consistently, when doing a G29 it will trigger 6 out of 9 times (3*3 grid) I am not sure whether there is a difference in the resistance between the two connectors but as the Manual suggests using the "Probe" connector I thought I would try that first and see if any improvement is achieved.

jaceknyc, 2019-07-25:
Having the same issue with more frequent failures. Assumed it was because I was using a clone for so I bought an original v3.1 so waiting for that.

Commenting out `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` and putting the probe in the probe connector didn't work as I lost my X,Y endstops. Issuing a `M119` showed only the z probe but not the x/y at that point. Also tried explicitly setting `#define Z_MIN_PROBE_PIN    PC14` to no avail. 

Have you had any success?

Xenon-2510, 2019-07-25:
sadly no, I am looking to get an Oscilloscope on the signal to look at how the trigger differs from a successful probe to an unsuccessful probe. I also had the same issue with the X/Y endstops not reporting to an  `M119` what I did was to define `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` but then swap the PC14 and PC2 in the boards pin.h file. This then allowed me to use the Probe connector but I still had the same issues.

Let me know how you get on with your official BLTouch as I too have a clone (3DTouch V1.2)

jaceknyc, 2019-07-28:
Well that was disappointing. Bought a BLTouch 3.1 (verified genuine) made the marlin config setting changes and still crashing into the bed 👎 

Going to contact Antclabs to see if they have any ideas. Maybe @Msq001 has any ideas?

tomoinn, 2019-07-29:
I've got this working with a genuine BLTouch 3.0 by swapping it into the z-min endstop header on the board and leaving `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` enabled. I had the same issue with losing X and Y endstops when trying to use the pin 14 header on the board, but this works fine now so unless there's a good reason you don't want to use the z endstop that should be enough (I removed mine when I got the bltouch, didn't really see the need for it)

Msq001, 2019-07-30:
hello, the reason of losing X/Y/Z endstops in here https://github.com/MarlinFirmware/Marlin/issues/14732. 
but I'm not sure why bltouch crash into bed

jaceknyc, 2019-07-31:
@Msq001  Oh interesting! The BL Touch crashing into the bed was a combination of a few problems that were mostly my fault:
1. The probe was not the correct height from the bed (corrected)
2. The probe was not level (corrected)
3. The drivers were overheating because (and this is speculation until support gets back to me) the spread jumper was in the wrong position on the E3 Mini. It is now running well and cool to the touch.

I'm now able to consistently probe with the genuine BL Touch but still not with the clone so that is good.

tomoinn, 2019-08-01:
@jaceknyc I've also got the genuine BLTouch working fine (using z-stop connector), can I ask which jumper location you've got for the spread jumper? I've not seen any issues yet but better to know before they happen!

jaceknyc, 2019-08-02:
@tomoinn - set it to GND/Spread but I only changed that because I never got the board working without overheating in the first place.

analytist, 2019-08-07:
/**
 * The BLTouch probe uses a Hall effect sensor and emulates a servo.
 */
#define BLTOUCH << to use BLtouch, I have to comment out here, is that correct?

mejtoogood, 2019-08-10:
The readme needs to be updated with instructions for configuring the BL Touch header on the board.

I'd like to use the BL Touch header on the board but I've been unable to figure out how, nor have I been able to find anyone who knows how. I'm using the Z Endstop header but have inconsistent results - the probe will occasionally crash into the bed.

eklee, 2019-08-17:
Any updates on this? Is this a hardware issue (PROBE header not working), is that why version 1.1 of the board is released?

sky8000, 2019-08-21:
Hi,

I got my BL Touch v3.1 working with SKR mini E3 on my Ender 5 using the lastest Marlin bugfix-2.0.x and making the changes below.

On file **pins_BIGTREE_SKR_MINI_E3.h**:
- Changed `#define Z_STOP_PIN PC2` to `#define Z_STOP_PIN PC14`.

On file **Configuration.h**:
- Uncommented `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`.
- Uncommented `#define BLTOUCH`.
- Set the appropriate values on `X_PROBE_OFFSET_FROM_EXTRUDER`. `Y_PROBE_OFFSET_FROM_EXTRUDER` and `Z_PROBE_OFFSET_FROM_EXTRUDER`.
- Uncommented `#define AUTO_BED_LEVELING_BILINEAR` (or use your preferred ABL method). 
- Uncommented `#define Z_SAFE_HOMING`. Although this is not required, it avoids damaging your printer by ensuring that Z homing is always done inside the print bed area.

Additional info:
- The value of `Z_PROBE_OFFSET_FROM_EXTRUDER` should be defined carefully to avoid smashing the print head into the bed. When not sure about the offset just set the default to 0 and fine tune it using **Probe Z Offset** from the printer's configuration menu.

brew99, 2019-08-21:
> Hi,
>
> I got my BL Touch v3.1 working with SKR mini E3 on my Ender 5 using the lastest Marlin bugfix-2.0.x and making the changes below.
>
> On file **pins_BIGTREE_SKR_MINI_E3.h**:
>
> * Changed `#define [...]

So you are using the "Probe" connector now but still need to uncomment Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN?

sky8000, 2019-08-21:
> 
> 
> > Hi,
> > I got my BL Touch v3.1 working with SKR mini E3 on my Ender 5 using the lastest Marlin bugfix-2.0.x and making the changes below.
> > On file **pins_BIGTREE_SKR_MINI_E3.h**:
> > 
> > * Changed `#define Z_STOP_PIN PC2` to `#define Z_STOP_PIN PC14`.
> > 
> > On file **Configuration.h**:
> > 
> > * Uncommented `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`.
> > * Uncommented `#define BLTOUCH`.
> > * Set the appropriate values on `X_PROBE_OFFSET_FROM_EXTRUDER`. `Y_PROBE_OFFSET_FROM_EXTRUDER` and `Z_PROBE_OFFSET_FROM_EXTRUDER`.
> > * Uncommented `#define AUTO_BED_LEVELING_BILINEAR` (or use your preferred ABL method).
> > * Uncommented `#define Z_SAFE_HOMING`. Although this is not required, it avoids damaging your printer by ensuring that Z homing is always done inside the print bed area.
> > 
> > Additional info:
> > 
> > * The value of `Z_PROBE_OFFSET_FROM_EXTRUDER` should be defined carefully to avoid smashing the print head into the bed. When not sure about the offset just set the default to 0 and fine tune it using **Probe Z Offset** from the printer's configuration menu.
> 
> So you are using the "Probe" connector now but still need to uncomment Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN?
> 
> Right now, I'm using the "Zstop" connector without having to edit the pins_BIGTREE_SKR_MINI_E3.h file.

Yes, that's because I changed `Z_STOP_PIN` to act also as a probe.

LPRM, 2019-08-23:
Has anyone got BLTouch clone working?

schwaberdoodles, 2019-08-25:
> Hi,
> 
> I got my BL Touch v3.1 working with SKR mini E3 on my Ender 5 using the lastest Marlin bugfix-2.0.x and making the changes below.
> 
> On file **pins_BIGTREE_SKR_MINI_E3.h**:
> 
> * Changed `#define Z_STOP_PIN PC2` to `#define Z_STOP_PIN PC14`.
> 
> On file **Configuration.h**:
> 
> * Uncommented `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`.
> * Uncommented `#define BLTOUCH`.
> * **EDIT:** Uncommented `#define ENDSTOP_INTERRUPTS_FEATURE`.
> * Set the appropriate values on `X_PROBE_OFFSET_FROM_EXTRUDER`. `Y_PROBE_OFFSET_FROM_EXTRUDER` and `Z_PROBE_OFFSET_FROM_EXTRUDER`.
> * Uncommented `#define AUTO_BED_LEVELING_BILINEAR` (or use your preferred ABL method).
> * Uncommented `#define Z_SAFE_HOMING`. Although this is not required, it avoids damaging your printer by ensuring that Z homing is always done inside the print bed area.
> 
> Additional info:
> 
> * The value of `Z_PROBE_OFFSET_FROM_EXTRUDER` should be defined carefully to avoid smashing the print head into the bed. When not sure about the offset just set the default to 0 and fine tune it using **Probe Z Offset** from the printer's configuration menu.

Thank you @sky8000! I can confirm this configuration worked for me on an ender3 and an old BLTouch from ANTLABS in 2018 (not sure which version).

LPRM, 2019-08-29:
Got 3DTouch working with @sky8000 settings + enable #define BLTOUCH_FORCE_SW_MODE

sky8000, 2019-08-30:
Last night homing axis failed for the first time with my current configuration and the print head was smashed against the printer bed damaging the magnetic build plate. Posted some pictures in this link https://imgur.com/gallery/YJkVZGS.

Still don't know why this happened but I'll be running a few tests the next couple of days to try to find the root cause of the problem.

I've also contacted BIGTREETECH support and reported the issue.

donovanlaoh, 2019-09-06:
> Last night homing axis failed for the first time with my current configuration and the print head was smashed against the printer bed damaging the magnetic build plate. Posted some pictures in this [...]

First i like to thank you for your setup. It works (mostly). I also experiencing the same problem, sometimes the extruder go thoroughly without stop. 
After some trial and error, keep this setting commented : 

#define ENDSTOP_INTERRUPTS_FEATURE.

But it still dont remove the issues completely. Mostly this issues happen when the starting Z position is above 30mm.

sky8000, 2019-09-06:
> First i like to thank you for your setup. It works (mostly). I also experiencing the same problem, sometimes the extruder go thoroughly without stop.
> After some trial and error, keep this setting [...]

Thanks for the feedback. I am still making additional tests and debugging the problem.

I'll share config files here: https://github.com/sky8000/SKR-Mini-E3-Ender-5-Config-Files-Marlin-2

sky8000, 2019-09-06:
It seems a design problem in the 5V rail of the board might be causing ABL sensors to fail. This issue was reported by user Bratr Rytíř Salage on BIGTREETECH SKR MINI E3&E3 DIP User Group.

Facebook link: https://www.facebook.com/groups/322956191976815/permalink/377307899874977/

Snapshot of the post for people without Facebook: https://i.imgur.com/99kHqls.png

donovanlaoh, 2019-09-07:
Is adding 100nf cap between vcc and gnd in bltouch connector will help?

ChritstianAlexander, 2019-09-07:
Had trouble with my BLTouch and found this to work fine:-
https://youtu.be/-XUQKQnUNig
His description has links to his Marlin files. I downloaded them and they worked with BLTouch when connected as per his suggested schematic.

Settings I changed to get it working on my Tevo Black Widow.
I adjusted the bed size
Invert my steppers
Invert endstops X and Y
Adjust the boundaries for probing..

The Nozzle height is not setup yet but at lest my BLTouch is working.

Hope this helps.

donovanlaoh, 2019-09-08:
i've put 100nf cap between vcc and ground in the servo connector. Tried to print a few, starting Z position before auto homing from 190 cm. None are fail.

Rene046, 2019-09-08:
i got with my 3dtouch a bit further, almost working just from time to time the pin won't jump up if touching build plate...and wont give pulse.

what i changed is in pin_BIGTREE_SKR_MINI.h
#define X_STOP_PIN         PC0
#define Y_STOP_PIN         PC1
#define Z_STOP_PIN         PC14 //<<<<<<< was PC2

//
// Z Probe must be this pins
//
#define Z_MIN_PROBE_PIN    PC2   //<<<<<<< was PC14

sky8000, 2019-09-08:
> 
> 
> Is adding 100nf cap between vcc and gnd in bltouch connector will help?

I tested that today and it didn't make any difference, I still had the same issues.

Chaotnix, 2019-09-08:
While looking at the Schematics I noticed that the Probe and Endstop connectors have 3.3V. I commented out #define BLTOUCH_SET_5V_MODE and saved it to BLTouch eeprom. 
Since then I had no failed probes an no crashes. I Have BLTouch V3.0

lococnc, 2019-09-08:
> #define BLTOUCH_SET_5V_MODE

Would this also work on a V3.1?

Chaotnix, 2019-09-09:
It should work on V3.1 to..

sky8000, 2019-09-09:
> While looking at the Schematics I noticed that the Probe and Endstop connectors have 3.3V. I commented out #define BLTOUCH_SET_5V_MODE and saved it to BLTouch eeprom.
> Since then I had no failed probes an no crashes. I Have BLTouch V3.0

I did test this mode (open drain) and it still fails randomly on my printer with a BL Touch 3.1.

Chaotnix, 2019-09-09:
Did you write it to eeprom of BLTouch?
You can do this via display if voltage menue is enabled. Or by commenting out #define BLTOUCH_FORCE_MODE_SET. If you use #define BLTOUCH_FORCE_MODE_SET don't forget to comment it again and reflash after first boot.

sky8000, 2019-09-09:
> 
> 
> Did you write it to eeprom of BLTouch?
> You can do this via display if voltage menue is enabled. Or by commenting out #define BLTOUCH_FORCE_MODE_SET. If you use #define BLTOUCH_FORCE_MODE_SET don't forget to comment it again and reflash after first boot.

Yes, I also have BL Touch voltage menu enabled that I use for tests and debugging.

lococnc, 2019-09-10:
> Yes, I also have BL Touch voltage menu enabled that I use for tests and debugging.

Have you connected a octopi or other 5v source to the USB port?

sky8000, 2019-09-10:
> Have you connected a octopi or other 5v source to the USB port?

No. I did all tests using the printer's power supply with no power source on USB port.

Rene046, 2019-09-10:
> i got with my 3dtouch a bit further, almost working just from time to time the pin won't jump up if touching build plate...and wont give pulse.
> 
> what i changed is in pin_BIGTREE_SKR_MINI_E3.h
> #define X_STOP_PIN PC0
> #define Y_STOP_PIN PC1
> #define Z_STOP_PIN PC14 //<<<<<<< was PC2
> 
> //
> // Z Probe must be this pins
> //
> #define Z_MIN_PROBE_PIN PC2 //<<<<<<< was PC14
> 
> This is working for me with servo cable to servo connector black/white wire to Probe connector.
> and Zmin switch still working. lucky enough...
> 
> and don't forget to look at this link.. for other settings..
> https://www.antclabs.com/bltouch-v3

Wel finally got my 3dtouch working perfect, did what i wrote uphere, and solved the misreading by adding a 100Nf capacitor on the servo plus and ground pin...

JimmySnails, 2019-09-12:
I'm experiencing the same problem. I did use the Z endstopstop port though. As you guys described it, it works very inconsistently. 
The problems is, that when it doesn't work, it continues on printing without knowing it's z-min and it scratched my ultrabase bed very hard..... :- /

I do have a RPI connected to my e3 mini and there's a loopback (display is on when i disconnect the printers power),, but this didn't fix anything for me. LCD still flickers occasionally and a lot of probe errors. my worst so far was 7 failures out of 10.

Did you guys think of adding a buck converter to power the 5v for the probe and just connect the signal? 

Is adding a capacitor a "real fix"? or just an improvement? 

Should i keep the hardware switch for the z stop and move the probes z min to the probe slot? What's the advantage of doing that? - Would this keep the nozzle from crashing into the bed in case the probe fails?

From what i read both z stops would be treated equally?

Pukington, 2019-09-20:
@Rene046 How has your capacitor fix held up, still working well?  Any failures since?  I'm getting fed up with the high failure rate of a bltouch.  So you used a single 100nf cap, between which pins?

Rene046, 2019-09-21:
Hi Pukington.

Everything still working perfect. i also twisted the cables .

Pukington, 2019-09-24:
@Rene046 Thanks, I've put a 100nf between the GND (brown) and VCC (red) pins on my BLTouch and have been printing flawlessly all afternoon.  I'll use your code above to move the pin to it's correct motherboard spot to tidy up too.  Thanks again.

taeraeyttaejae, 2019-09-27:
I have managed to get my printer working so that it has a working z-endstop in stock configuration as a (ease of mind) safety AND a working 3dtouch ( looking like this https://imgur.com/a/UslwYp6 ).
I haven't used any kind of capacitor on the servo wires.

I have a CR-10 mini(300 x 220 x 300 build area) with Heavy Duty customisable modular e3D V6 mount ( https://www.thingiverse.com/thing:2494642 ) with 3dtouch on the left side. I have a 1 month old black pcb SKR mini e3. I have had 10-15 succesful probings with 5x5 mesh bilinear leveling.

My 3dtouch signal is in the PROBE connector and servo is in SERVO, Z-min is in it's own connector like in stock configuration.

When attaching the wires to board, i first followed bltouch wiring diagram that is different from 3Dtouch - 3Dtouch wiring: Z-MIN SIGNAL(white), GROUND(black), servo signal (yellow), servo 5V(red), servo ground (green) as opposed to bltouch pinout. Please take notice what version you have and wire accordingly.

Thanks for @Rene046 for pointing out what i was doing wrong in my attempts before i read your comments!

Here are dumps from configuration.h and pins_BIGTREE_SKR_MINI_E3.h. Please go through the files with diff so that you can pull out some unnecessary changes for your use case, like my PID tune settings and such.

`/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2019 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */

/**
 * Creality CR-10 Mini
 * X=300mm  Y=220mm  Z=300mm
 * E3DV6 Hotend
 * Titan Extruder
 * CR10_STOCKDISPLAY (RAMPS-compatible with single 10-pin plug)
 */

#define CONFIGURATION_H_VERSION 020000

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
[... 2323 lines cut ...]
    #define LCD_PINS_D4    PB9

  #else
    #error "Only CR10_STOCKDISPLAY is currently supported on the BIGTREE_SKR_MINI_E3."
  #endif

#endif // HAS_SPI_LCD

//
// SD Support
//
#define HAS_ONBOARD_SD

#ifndef SDCARD_CONNECTION
  #define SDCARD_CONNECTION ONBOARD
#endif

#define ON_BOARD_SPI_DEVICE 1    //SPI1
#define ONBOARD_SD_CS_PIN  PA4   // Chip select for "System" SD card

// eeprom
#define FLASH_EEPROM_EMULATION
#define EEPROM_PAGE_SIZE     (0x800) // 2KB
#define EEPROM_START_ADDRESS (0x8000000 + (256 * 1024) - 2 * EEPROM_PAGE_SIZE)
#define E2END                (EEPROM_PAGE_SIZE - 1)`

pusewicz, 2019-09-27:
@taeraeyttaejae I just did what you did, and it works fine. Gotta adjust the Z endstop though, which is a pain.

taeraeyttaejae, 2019-09-27:
What i had to do was to make sure that Z-endstop does not trigger before probe does, and the linked carriage with adjustable "wings" helped out a lot in that. The Z-endstop works like extra safety in my setup, so it throws probing error and disconnects serial if z-endstop triggers when probing is in progress.

gazcbm, 2019-10-06:
What board version is everyone using? I have just received a v1.2 board just wondered if this is still valid

sky8000, 2019-10-07:
> 
> 
> What board version is everyone using? I have just received a v1.2 board just wondered if this is still valid

Hi, mine is v1 and the problem reported here was identified in this version of the board.

I noticed that last week people were posting on social networks that they received v1.2 but I can't find any information about this newer version of the board.

Can you test the board a give us some feedback?

ManiRX100, 2019-10-07:
> So you are using the "Probe" connector now but still need to uncomment Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN?
>
> Right now, I'm using the "Zstop" connector without having to edit the [...]

Hi
If you got the BL touch to work, I guess you did not get the "TMC Connection error" message on your board with the marlin 2.0 bug fix. Currently I have an open thread #32 , wherein few of us are facing the TMC error with the new SKR MINI E3 board. Can you support us in this regard? We followed the same procedure as given by  Bigtreetech. Thanks in advance

pusewicz, 2019-10-07:
@ManiRX100 Do you have the board connected via USB?

ManiRX100, 2019-10-07:
> 
> 
> @ManiRX100 Do you have the board connected via USB?

No.. I connected the board to the printer. I am compiling the Marlin to a .bin file and trying to update the firmware.

wieltar, 2019-10-09:
should this also work for my e3 dip board?

Pukington, 2019-10-09:
> should this also work for my e3 dip board?

I fixed mine with fitting a 100nF between VCC and GND, that worked, but others both here and elsewhere have stated that enabling BLTOUCH_FORCE_SW_MOD­E works just as well.  And yes, I have an E3 Dip.

lococnc, 2019-10-10:
Is that a 100nf Ceramic Cap?

Pukington, 2019-10-10:
@lococnc yes, sorry I edited my post, 100nF ceramic capacitor, [here's](https://imgur.com/a/6z3gWxG) how I tested it.

Mesmer88, 2019-10-10:
> 
> 
> I have managed to get my printer working so that it has a working z-endstop in stock configuration as a (ease of mind) safety AND a working 3dtouch ( looking like this https://imgur.com/a/UslwYp6 ).
> I haven't used any kind of capacitor on the servo wires.
> 

Is there something specific in Marlin for this setup? I also want to add a BLTouch in this fashion, i.e also keep the original Z-stop for safety.

sakoula, 2019-10-10:
thanks! I will order it!!!

tilk, 2019-10-11:
I managed to get BLTouch working with Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN commented out. The reason for disappearing endstops described by @tomoinn is rather esoteric, here is the full issue I reported to marlin:
https://github.com/MarlinFirmware/Marlin/issues/15528
My solution is rather hacky, maybe the Marlin developers will find a better one.

bigtreetech, 2019-10-12:
hello everyone, please try [this](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V1.2/readme.md#if-you-enable-bltouch), we have tested a lot times,  everything of Bltouch works well.

davehardy20, 2019-10-12:
@bigtreetech can you confirm please, which terminals on the SKR E3 mini the BLTouch should be connected for this method you have shown. My working BLTouch Smart 3.1 is currently connected to 'servos' and the Z Endstop terminal with the 'Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN' defined in the firmware. Are you using the 'Probe' terminal? A diagram or picture would be helpful. Thanks

gazcbm, 2019-10-12:
https://github.com/gazcbm/Marlin-2.0.x-SKR-Mini-E3-v1.2/blob/master/docs/1570702845960.png

tilk, 2019-10-12:
It turns out the issue with Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN is solved upstream, in particular this file:
`Marlin/src/HAL/HAL_STM32F1/fastio.h`
@bigtreetech please merge the solution.

mankis2000, 2019-10-24:
Hello, i have SKR mini E3 V1.2 and failing to work 3dtouch, i am able to stop it crashing by deploying the probe and holding it deployed, only that can stop it. Tried some wire changes and firmware but no luck, any suggestions?

mankis2000, 2019-10-24:
Forgot to mention that 3dtouch is newer with plastic pin not metal

[37 more replies]

closed 2020-04-06 by Xenon-2510