bigtreetech/BIGTREETECH-SKR-mini-E3 #10: Cura 4.2 USB issues under linux, workaround included

tomoinn, 2019-08-05:
Under linux, Cura 4.0 and 4.2 both have trouble connecting to the USB serial port. This manifests as a lock-up of the UI on the printer and a failure to detect the board in Cura. The permissions on /dev/ttyACMxxx are fine, and arduino serial monitor can connect and send GCode

I found that if I commented out the following section in Cura's AutoDetectBaudJob.py file it then worked perfectly, so I assume there's something in the interaction between this code and the mini E3's USB serial stack that's unhappy. Haven't seen the same issue on windows. The code I now have commented out is:

```python
def run(self) -> None:
        Logger.log("d", "Auto detect baud rate started.")
        wait_response_timeouts = [5, 15, 30]
        wait_bootloader_times = [5, 15, 30]
        write_timeout = 3
        read_timeout = 3
        tries = 2

        # programmer = Stk500v2()
        serial = None
        # try:
        #     programmer.connect(self._serial_port)
        #     serial = programmer.leaveISP()
        # except ispBase.IspError:
        #     programmer.close()
```

Note the commented out references to the ISP library. I think this is really a Cura bug / issue, but it's only affected me since swapping to this mainboard so I guess others may hit it as well.

tomoinn, 2019-08-05:
Related issue filed at Cura - https://github.com/Ultimaker/Cura/issues/6154

jtnix, 2019-10-13:
Thanks for reporting this. I encountered the same issue and behavior on Cura 4.2.1 connecting to SKR mini E2 v1.2 running the new firmware. I figured it was a Mac/USB thing and almost started messing with the Configuration.h settings then I saw this.

Editing the same file which is in `Ultimaker Cura.app/Contents/Resources/plugins/plugins/USBPrinting/AutoDetectBaudJob.py` on Mac OS as suggested fixed this issue. I can now connect to and control my modded Ender 3 Pro from Cura.

lanvezhon, 2020-01-08:
Thanks for reporting this bug. I get same pb using cura 4.4.1 appimage under ubuntu trying to connect to anet A8 PLUS V1.6.