bigtreetech/BIGTREETECH-SKR-mini-E3 #12: Build error: SoftwareSerial.h: No such file or directory

eklee, 2019-08-11:
Is anyone able to build the firmware from this repo? I am getting this error:

Compiling .pioenvs\BIGTREE_SKR_MINI\src\src\feature\bedlevel\abl\abl.cpp.o
In file included from Marlin\src\module/stepper_indirection.h:47:0,
                 from Marlin\src\module/stepper.h:221,
                 from Marlin\src\Marlin.cpp:37:
.piolibdeps\BIGTREE_SKR_MINI\TMCStepper_ID5513\src/TMCStepper.h:18:11: fatal error: SoftwareSerial.h: No such file or directory

************************************************************************
* Looking for SoftwareSerial.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SoftwareSerial.h"
* Web  > https://platformio.org/lib/search?query=header:SoftwareSerial.h
*
************************************************************************

  #include <SoftwareSerial.h>
           ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\BIGTREE_SKR_MINI\src\src\Marlin.cpp.o] Error 1
In file included from Marlin\src\feature\../module/stepper_indirection.h:47:0,
                 from Marlin\src\feature\../module/stepper.h:221,
                 from Marlin\src\feature\babystep.cpp:30:
.piolibdeps\BIGTREE_SKR_MINI\TMCStepper_ID5513\src/TMCStepper.h:18:11: fatal error: SoftwareSerial.h: No such file or directory

eklee, 2019-08-11:
I downloaded the official Marlin repo, applied changes from this repo, and it now builds successfully.

closed 2019-08-11 by eklee

SergeySn, 2019-08-21:
@eklee which changes did you apply from the official Marlin repo?  Did you just copy the SoftwareSerial.h file?

@bigtreetech Could you please fix the BIGTREETECH-SKR-mini-E3- repo, so that a fresh copy would compile?

eklee, 2019-08-21:
@SergeySn See here for steps to modify Marlin repo: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/blob/master/firmware/readme.md

SergeySn, 2019-08-21:
@eklee I have followed your instructions, and it compiles now.  Thanks a lot.