bigtreetech/BIGTREETECH-SKR-mini-E3 #13: PCB Schematic

LemaruX, 2019-08-14:
Please release the schematic of the PCB. You have released a schematic for other models but not any of the mini versions.

This will help in troubleshooting and repairing faulty boards.

pyunramura, 2019-08-16:
Waiting for this for the Mini E3 DIP as well. It would go a long way to sorting out what the board is capable of and helping with any issues that may arise like you mention.

AletheianAlex, 2019-08-18:
Yes, the open source community is eagerly awaiting schematics for the mini boards. In addition, they will solve many potential questions and make debugging much much easier.

thisiskeithb, 2019-08-31:
They've been uploaded:
- [Mini 1.1 schematic](https://github.com/bigtreetech/BIGTREETECH-SKR-MINI-V1.1/blob/master/hardware/SKR-mini-V1.1sch.pdf)
-  [Mini E3 schematic](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/blob/master/Hardware/SKR-mini-E3-SCHpdf.pdf)
- [Mini E3 DIP schematic](https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/blob/master/Hardware/SKR-E3-DIP-V1.0-SCH.pdf)

AletheianAlex, 2019-09-01:
Thanks for posting the update Keith!  Just got the notification.

bojanpotocnik, 2019-10-28:
I would suggest closing this issue as it is resolved.

closed 2019-10-28 by LemaruX