bigtreetech/BIGTREETECH-SKR-mini-E3 #14: [QUESTION] SD mount by USB

reloxx13, 2019-08-15:
Heya,

should it be possible to mount the SD card by USB on a unix/windows PC?

Im trying to find a way to update the firmware by USB rather than the SD Card.

For this i would like to copy the firmware over USB to the SD card.

E.g. the SKR Pro 1.1 does work like this.

BR

bigtreetech, 2019-09-07:
It's done in here https://github.com/MarlinFirmware/Marlin/pull/15180.
But more testing is needed.

reloxx13, 2019-09-07:
thank you, looking forward for the merge.

reloxx13, 2019-09-08:
[image]

closed 2019-09-08 by reloxx13

dseven, 2019-12-12:
Can't get this to work in 2.0.x-bugfix. I've built for STM32F103RC_bigtree_512K_USB, and both Windows and Linux (Pi running OctoPi) see a disk device, but they don't seem to be seeing a partition table on it, so it can't be mounted. Connecting the card via a USB card reader works fine (so the card itself seems OK).

```
root@octopi:~# fdisk -l /dev/sda
Disk /dev/sda: 120 MiB, 125829120 bytes, 245760 sectors
Disk model: SD Flash Disk
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
root@octopi:~#
```

What am I missing?

Phisik, 2020-01-13:
I experience exactly the same problem as dseven above with my SKR MINI E3 v1.2 (TMC2209 version). I have build the latest Marlin 2.0.1 with platform.io  for STM32F103RC_bigtree_512K_USB  environment.  I can see the SD card reader in OS. Both Win10 & Armbian see Composite USB device & COM port. Both OSes correctly detect SD card capacity but see no partition on it. Neither  SD card can be formatted from OS.  Are there any defines to be switch on for SD reader to work properly? Or this is  some usb code bug?

Here is Linux output for STM32 card reader:
```
root@orangepi:/# ls /dev/sd*
/dev/sda
```
```
root@orangepi:/# fdisk -l /dev/sda
Disk /dev/sda: 120 MiB, 125829120 bytes, 245760 sectors
Disk model: SD Flash Disk
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
```
[config_file.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4055094/config_file.zip)

dseven, 2020-01-13:
I started to open a new issue on this, but it pointed me to [this existing one](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/129). The workaround there seems to be working for me.