bigtreetech/BIGTREETECH-SKR-mini-E3 #15: No COM/Serial port visible

SergeySn, 2019-08-21:
I have followed the instructions on https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/blob/master/firmware/readme.md to compile the firmware.
But I wanted to debug it, so I connected via SWD connector and pressed the PlatformIO Upload button.  The firmware was uploaded successfully.  But when I try to connect to the board via the virtual COM port via USB - I don't see any COM ports.  
And before I uploaded my firmware, I was able to see a COM port, and I could connect to it, and send gcode commands, and everything was working.
And now, even if I put the firmware.bin file onto the SD Card - it doesn't take it.  The LED doesn't blink, and FIRMWARE.CUR doesn't appear.
What might be wrong?

ceppelli, 2019-09-10:
Hi @SergeySn 

I'd like to do some "live debug" of the SRK mini E3. From what I read I need a ST-LINK USB jtag, right? I seen different versions of it (v1, v2, v3), could you point me to the right one?

I'm using PlatformIO on a Mac Book, do you know if it is possible to use for debugging?

Have you find a official guide from bigtreetech for debugging their STM32 boards?

Thanks in advance!!!!
Luca

SergeySn, 2019-09-11:
@ceppelli 1. I am using ST-LINK from Nucleo STM32F446RE board. (Not sure about the version)
2. I don't know much about Mac - I am using Linux Mint 18.3
3. I have not seen any official guide from BigTreeTech for debugging their STM32 boards.

thubot, 2019-10-01:
Flashing from SD is done by the bootloader. When you flashed via SWD, you likely erased/overwrote the bootloader.

I think the board might use a modified [stm32duino-bootloader](https://github.com/rogerclarkmelbourne/STM32duino-bootloader). If you want to get the flashing from SD functionality back, you need to flash a suitable bootloader via SWD.

Not sure how the serial issue is related to that though.

SergeySn, 2019-10-01:
@thubot Thanks.  That makes a lot of sense.

dinamitemic, 2019-11-05:
Anyone solved this? I've made same steps as author and I'm exactly in the same situation

dinamitemic, 2019-11-07:
I've found the solution. The problem was with a wrong starting address when programmed from st-link.
`[Here](https://github.com/bigtreetech/BIGTREETECH-SKR-MINI-V1.1/issues/7#issuecomment-549972479)`you can find all the details among with the **stock SKR MINI v1.1 bootloader.**

wisesokol, 2020-05-27:
> 
> 
> I've found the solution. The problem was with a wrong starting address when programmed from st-link.
> [Here](https://github.com/bigtreetech/BIGTREETECH-SKR-MINI-V1.1/issues/7#issuecomment-549972479) you can find all the details among with the **stock SKR MINI v1.1 bootloader.**

Hi.
stock SKR MINI v1.1 bootloader suitable for skr mini e3 v1.2 ?

SergeySn, 2020-05-27:
Not for debugging with ST-Link.  At least I had problems with it. So, I have uploaded a new one, and then it worked.
> stock SKR MINI v1.1 bootloader suitable for skr mini e3 v1.2 ?

closed 2020-05-27 by SergeySn

wisesokol, 2020-05-27:
I want to replace RCT6 with RET6 c 512Kb on my skr mini e3 v1.2. I will solder it. Do I definitely need ST link v2 or will UART TTL CP2102 be enough?
It is flashed through the tft port as I understand it.

foxdan, 2020-11-21:
Possibly stupidly as soon as my SKR Mini E3 1.2 arrived first thing I did was hook up to the SWD port and flash Marlin mainline. I then couldn't get a virtual serial port from the STM32 USB port.

After doing a complete chip erase and trying again to run it without the bootloader it worked "sometimes". At first I thought there was some trickery in the bootloader that initialised it correctly for this board and it was conflicting with the UART for the TMC drivers.

Turned out is was always working only the board does not have a diode setup to stop USB power backpowering the rest of the board, so make sure after writing to the chip via an ST-Link you then test it by powering the board from it's 12/24v input.

wisesokol, 2020-11-21:
https://youtu.be/w2hJrD5voBY