bigtreetech/BIGTREETECH-SKR-mini-E3 #16: Fan0 and Fan1 mixed compared to stock Ender 3

weakset, 2019-08-21:
I didn't find 1:1 similar example pictures, but I hope you can see what I mean. This board is drop-in replacement, but if user copies wiring from stock, his part cooling fan is always 100% and marlin controls the case fan.

Stock board:
[image]
Fan0 next to endstops: 12V and GND
Fan1: 12V and ControlPin

SKR mini-E3:
[image]
Fan0 next to endstops: PA8 and GND
Fan1: GND and VBB (12V)

weakset, 2019-08-23:
The source of this mixup:
In stock board both fan headers are connected and work identical. We tested this with multimeter instead of trying to find schematics. That is bad design for case fan, it should always be on even if part cooling is off or low.

SKR-mini-E3 did try to fix this, but it isn't well documented. Users should be advised which header fans should be connected to,  because they are different than on stock board.

The-slunk, 2019-08-23:
So the only difference is that the parts fan and control box fan headers have new switched?

weakset, 2019-08-23:
They did fix one fan header so that it is always 100%, but they didn't label it on pcb or tell that to users any other way. And using stock wiring as reference as most people do, you'd most likely plug parts fan to that 100% header.

Good fix, but bad documentation.

weakset, 2019-08-24:
I did finally find the documentation. There actually is manual on github, but wiring page doesn't help you. You have to know how to read Pins page.
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/blob/master/Hardware/SKR%20MINI%20E3%20V1.0%20manual.pdf

So technically the solution was RTFM, I still suggest you document it easier to understand.

closed 2019-08-24 by weakset