bigtreetech/BIGTREETECH-SKR-mini-E3 #17: Non-stop continuous beeping with Ender3 12864 LCD buzzer, when powered with USB

delafer, 2019-08-25:
After flashing a new firmware version and testing in usb mode, connected via pc-usb, so with 5V only power supply and without any sensors / motors attached -> Ender3 12864 LCD buzzer produces constant continuous beeping. After resetting / flashing original stock firmware (supplied on sd card) continuous beeping sound disappears. 
I have commented out #define SPEAKER in configuration.h file, it doesn't help.
I have a latest marlin version from 25.08.2019 with https://github.com/MarlinFirmware/Marlin/pull/15026 bugfix, it doesn't help also.
behavior of stock firmware powered via usb in test mode is different and stock firmware ,doesn't produce any annoing constant buzzer sounds, if skr mini e3 is turned on.

closed 2019-08-25 by delafer

CineXMike, 2019-08-27:
What was the issue?
I am having the same problem.

delafer, 2019-08-28:
> What was the issue?
> I am having the same problem.

Du you have skr e3 mini or skr e3 dip version with removable drivers ?

retikulumx, 2019-12-11:
Hi. I'm having the same issue. And yes, the dip version with TMC2209 on it.
Is it a TMC related problem? It seems so since the beeping starts before Marlin init.