bigtreetech/BIGTREETECH-SKR-mini-E3 #18: When starting print button is randomly triggered and making beeping sounds.

CineXMike, 2019-08-27:
Hi,

When i start a print the button of the screen keeps being pressed and released and it goes in and out of all kinds of menu's until i reboot the printer.

Does anyone have some advice?

Greetings,
Mike

CineXMike, 2019-08-27:
Correction, 
This only happens with the heated bed on. 
As soon as i turn that on it starts going nuts.

CineXMike, 2019-09-04:
Thanks allot gonna test!

lococnc, 2019-09-30:
I have this happen as well. It seems random. On one print I ran it did this every few seconds. Today I ran an 8 hour print and had this happen only once.

adsaxton, 2021-07-24:
This is indicative of low current from your power supply. If you're powering a Raspberry Pi from the printer's power supply, try using an external power supply for it. I ended up replacing the stock power supply with a 500w model.