bigtreetech/BIGTREETECH-SKR-mini-E3 #19: X_SLAVE_ADDRESS and E0_SLAVE_ADDRESS are the same

creamywhite, 2019-08-27:
In Configuration_adv.h:
`  #define  X_SLAVE_ADDRESS 0
  #define  Y_SLAVE_ADDRESS 1
  #define  Z_SLAVE_ADDRESS 2
  #define X2_SLAVE_ADDRESS 0
  #define Y2_SLAVE_ADDRESS 0
  #define Z2_SLAVE_ADDRESS 0
  #define Z3_SLAVE_ADDRESS 3 //<---
  #define E0_SLAVE_ADDRESS 0 //<---
  #define E1_SLAVE_ADDRESS 0
  #define E2_SLAVE_ADDRESS 0
  #define E3_SLAVE_ADDRESS 0
  #define E4_SLAVE_ADDRESS 0
  #define E5_SLAVE_ADDRESS 0`

It seems wrong to me. Previously it was like that: 
`  #define Z3_SLAVE_ADDRESS 0
  #define E0_SLAVE_ADDRESS 3`

Chaotnix, 2019-08-28:
I noticed this also, seems somebody changed the wrong line. In the line above Z3 is set 

to 3. I left E0 as 3 and Z3 as 0. This works flawlessly.

thisiskeithb, 2019-08-31:
I put in a PR to address the incorrect slave address for E0: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/pull/25

creamywhite, 2019-09-01:
> I put in a PR to address the incorrect slave address for E0: #25

Thanks!

closed 2019-09-04 by creamywhite