bigtreetech/BIGTREETECH-SKR-mini-E3 #21: Feature Request: Get Neopixel working

Chaotnix, 2019-08-28:
I would like to use Neopixels on my Printer.

By default Adafruit Neopixel is in lib_ignore in plattform.io for STM32F1 boards, because of compile errors regarding gpio_set, gpio_clear and time funktions. Also Neopixel has no support for STM32F1 cpu yet.

I tried to copy LPC1768 parts and change to TARGET_STM32F1. This lead me to the same compile errors. 

I also tried to use bigtreetech/Adafruit_NeoPixel. This compiles ok, but unfortunately does not work.

Is there anything, we can do to use Neopixels?

mmende, 2019-10-20:
I have the same issue but made some progress.
One main issue seems to be the DCDC5V v1.0 power module not delivering any output. I plan to use a 24 neopixel ring. As the documentation says [here](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V1.2/The%20Notement%20of%20BTT%20SKR%20MINI%20E3%20V1.2.pdf) for more than 8 LEDs the jumper should be on the left so that the power is not provided by the board 5V power supply but rather from the DCDC5V v1.0 power module which should be able to supply enough power for up to 30 pixels. However when the jumper is on the left I measure 0 voltage between +5V and ground in this configuration. Fortunately I had a 8 pixel strip lying around to test the on board power supply configuration and tada the NEOPIXEL_STARTUP_TEST was visible and I could control the pixels with the M150 gcode. The question is now how to get the DCDC5V v1.0 power module working.

For the software part:
To be able to compile it besides uncommmenting `Adafruit NeoPixel@1.2.5` and adding `https://github.com/bigtreetech/Adafruit_NeoPixel` I also had to remove `Adafruit Neopixel` from lib_ignore under the `[env:STM32F103RC_bigtree]` section otherwise Neopixel.h would be missing. I then enabled NEOPIXEL_LED in Configuration.h and set NEOPIXEL_PIN to PC7 as stated in the pinout.

Veema23, 2019-10-24:
im a bit confused with this neopixel stuff as well, I have everything set up in my configuration files and have compiled the .bin but the rag strip that I have has four wires coming from it and the skr mini e3 only has three pins.what will go where for this to work?

Veema23, 2019-10-24:
Ok never mind I see I was looking at the wrong pinout on the board...