bigtreetech/BIGTREETECH-SKR-mini-E3 #22: Stepper motor current control

mustafaarif, 2019-08-28:
Stepper motors on my Ender 3 are getting very hot using Skr mini E3 board. I am using the default firmware which was provided. I am wondering which parameters I should tune in configuration_adv.h for tmc2209?

Chaotnix, 2019-08-28:
Then it might be a good idea to adjust  the (X|Y|Z)_MAX_CURRENT to 800ma  in @section tmc.

mustafaarif, 2019-08-29:
@brew99 This is exactly what I was looking for. I was able to make changes in @section tmc_smart and steppers are now behaving normal. Before updating the firmware, I tried different values using;

// Modify stepper motor current in milliamps
M906 X550 Y550 Z550
M906 T0 E650

My current values are X,Y,Z (600) and E0 (650).

Thank you!

closed 2019-08-29 by mustafaarif

thisiskeithb, 2019-08-31:
FYI: I just put in a PR to the main Marlin repo to add default RMS currents for a stock Ender-3: https://github.com/MarlinFirmware/Marlin/pull/15115.

sugar012, 2019-09-04:
> Yes, I found that also. The issue is that the stock motors are rated at peak currents or 0.84A (X,Y,Z) and 1.0A (E). I got an email from the motor manufacturer who stated that they rate their motors in "peak current", unlike many others that rate in "RMS".
> 
> The default current in Marlin 2.0 bugfix (and I assume BTT firmware) is 800mA RMS, which equates to 1130mA peak. This is why the motors are getting hotter than needed.
> 
> Adjust the current in Configuration_adv.h (@section tmc_smart) in around line 1647, #define X_CURRENT (change from 800 to 550). Do this for X,Y,Z. Change the E_CURRENT to about 650.
> 
> I'm running these numbers and the motors are warm, but you can easily keep your hand on them, and they are not missing steps at these currents either.
> 
> I've gone through this on the r/BIGTREETECH

thanks you mate. i was looking for this

sugar012, 2019-09-05:
i found that with 650 the E stepper still run at 65°! Since the extruder is in direct contact with it the heat make the filament to soften and then it clogs right before enter the bowden tube. Im trying to step down the current to find the nice compromise. How do i know when is too low? It miss steps?

mustafaarif, 2019-09-05:
If the current is too low, it will miss steps and you will see missed layers or holes in your prints. Use pronterface to manually change the current by issuing M906 T0 E650 or lower. The real test of motor torque can only be done while printing as the filament has to push through nozzle and stepper motor needs more current to achieve that level of torque. 

Extruder current 650 has been working totally fine for me for long hours of print. Did you upgrade firmware on your skr-mini E3 board recently? There was a bug in Marlin SKR E3 branch where the slave address of Extruder was the same as X stepper. This was causing Extruder to heat up enormously as driver was unable to control the extruder current. This bug has been fixed in latest release. Let me know if you need more help with that.

sugar012, 2019-09-05:
Many Thanks mustafa. The board arrived 2 days ago but i installed what i found here in the github: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/tree/master/firmware/Marlin-2.0.x-SKR-Mini-E3 . Just download it yesterday and installed. My current "slave" setup is this:

`#define  X_SLAVE_ADDRESS 0
  #define  Y_SLAVE_ADDRESS 1
  #define  Z_SLAVE_ADDRESS 2
  #define X2_SLAVE_ADDRESS 0
  #define Y2_SLAVE_ADDRESS 0
  #define Z2_SLAVE_ADDRESS 0
  #define Z3_SLAVE_ADDRESS 3
  #define E0_SLAVE_ADDRESS 0
  #define E1_SLAVE_ADDRESS 0
  #define E2_SLAVE_ADDRESS 0
  #define E3_SLAVE_ADDRESS 0
  #define E4_SLAVE_ADDRESS 0
  #define E5_SLAVE_ADDRESS 0`

is because E0 is the same as X ?

Thanks again

mustafaarif, 2019-09-05:
Not your fault. It was the bug in the skr-mini-e3 repo and its fixed now with merge #25  . Please make below changes;
  #define Z3_SLAVE_ADDRESS 0
  #define E0_SLAVE_ADDRESS 3
Recompile and upgrade the firmware. It should solve the Extruder heating problem.

sugar012, 2019-09-06:
all seems to work fine now! Stepper is now cool and the print works great. Thanks Mustafa!

Ragpuss, 2020-04-05:
I just replaced my stepper motors for 1500ma ones on XYZ as the old ones were getting noisy , what would be a good setting for those ? I was thinking something around 900ma ?

MoMoPouetDev, 2020-06-04:
Hi @thisiskeithb, I noticed that the change for the slave address is available for the v1.0, not need on the v1.2?

thisiskeithb, 2020-06-04:
> Hi @thisiskeithb, I noticed that the change for the slave address is available for the v1.0, not need on the v1.2?

Correct because the SKR Mini E3 V1.2 uses software serial. Related to that: you’ll see slave addresses have been added to the SKR Mini E3 V2.0 since BigTreeTech went back to using hardware serial.

MoMoPouetDev, 2020-06-04:
> Correct because the SKR Mini E3 V1.2 uses software serial. Related to that: you’ll see slave addresses have been added to the SKR Mini E3 V2.0 since BigTreeTech went back to using hardware serial.

Ok so I need to keep all SLAVE_ADDRESS to 0?

Thanks

thisiskeithb, 2020-06-04:
> Ok so I need to keep all SLAVE_ADDRESS to 0?

For the V1.2, yes. You can verify against the working SKR Mini E3 configs in the main [Marlin Configurations repo](https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/BigTreeTech) as well.

MoMoPouetDev, 2020-06-04:
OK, great.
Thank you very much.

It's because I had a print problem on z axis so I looking for a solution.
I had on current at 800 on all axis so I change it thanks to your PR, so now I do a test again. Hope this is the answer.

Thanks again

M.

MoMoPouetDev, 2020-06-04:
> > Ok so I need to keep all SLAVE_ADDRESS to 0?
> 
> For the V1.2, yes. You can verify against the working SKR Mini E3 configs in the main [Marlin Configurations repo](https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/BigTreeTech) as well.

OK, great.
Thank you very much.

It's because I had a print problem on z axis so I looking for a solution.
I had on current at 800 on all axis so I change it thanks to your PR, so now I do a test again. Hope this is the answer.

Thanks again

M.

Gertsch, 2020-08-10:
So I compiled my own firmware, took the settings from the config samples (SKR mini 1.2) and changed some settings according to this guide: https://www.reddit.com/r/ender3/comments/hymv70/marlin_20x_guide_skr_mini_e3_v12_ender_3/

My steppers get really hot to the touch, and the test cube looks really weird - like a staircase, but in z direction: https://photos.app.goo.gl/6XrWJyNCWwzhCtei9
Any thoughts what I could try?

LVNeptune, 2020-08-12:
> So I compiled my own firmware, took the settings from the config samples (SKR mini 1.2) and changed some settings according to this guide: https://www.reddit.com/r/ender3/comments/hymv70/marlin_20x_guide_skr_mini_e3_v12_ender_3/
> 
> My steppers get really hot to the touch, and the test cube looks really weird - like a staircase, but in z direction: https://photos.app.goo.gl/6XrWJyNCWwzhCtei9
> Any thoughts what I could try? Could Octoprint interfere with this?

You are having the same issues as I am. I used the previous guide for a few month old Marlin version and my motors are all 55+ which everyone keeps saying is "normal" ...No, it isn't. You should be able to touch the motor without it burning you. They were warm on the stock board. I am almost at the point where I may go back to the stock board because of this. I am going to try and drop the current on Y to 450 and see what happens.

mustafaarif, 2020-08-12:
@LVNeptune 450 seems too low for Y axis.
@LVNeptune @Gertsch Though I don't have experience with SKR v1.2, but if you can upload your current Marlin source (configuration.h, configuration_adv.h, src/pins/stm32f1/pins_BTT_SKR_MINI_E3_V1_2.h), I can have a quick look to see if I can spot any issue? Also, can you share M503?

oschni, 2020-09-01:
I have the same issue. Running Marlin 2.0.6 with BTT SKR mini 1.2 configuration files and a setting for a BLTouch.

Stepper motors for X, Y and Z are getting hot. Really hot.  The attached filament guide melted a while ago.

I am very close to install the original Creality Silent Board again.

```
[...]
Send: M503
Recv: echo:  G21    ; Units in mm (mm)
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:; Filament settings: Disabled
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo: M92 X80.40 Y80.00 Z400.00 E135.41
Recv: echo:; Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z5.00 E25.00
Recv: echo:; Maximum Acceleration (units/s2):
Recv: echo:  M201 X500.00 Y500.00 Z100.00 E5000.00
Recv: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P500.00 R500.00 T500.00
Recv: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
Recv: echo:  M205 B20000.00 S0.00 T0.00 J0.08
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:  G29 W I0 J0 Z0.09500
Recv: echo:  G29 W I1 J0 Z0.09250
Recv: echo:  G29 W I2 J0 Z0.11000
Recv: echo:  G29 W I3 J0 Z0.11250
Recv: echo:  G29 W I4 J0 Z0.08750
Recv: echo:  G29 W I0 J1 Z0.07750
Recv: echo:  G29 W I1 J1 Z0.03750
Recv: echo:  G29 W I2 J1 Z0.02250
Recv: echo:  G29 W I3 J1 Z0.02500
Recv: echo:  G29 W I4 J1 Z-0.03500
Recv: echo:  G29 W I0 J2 Z0.05250
Recv: echo:  G29 W I1 J2 Z0.02000
Recv: echo:  G29 W I2 J2 Z0.01500
Recv: echo:  G29 W I3 J2 Z0.02750
Recv: echo:  G29 W I4 J2 Z-0.01750
Recv: echo:  G29 W I0 J3 Z-0.00750
Recv: echo:  G29 W I1 J3 Z-0.00500
Recv: echo:  G29 W I2 J3 Z0.02750
Recv: echo:  G29 W I3 J3 Z0.06250
Recv: echo:  G29 W I4 J3 Z0.07000
Recv: echo:  G29 W I0 J4 Z0.04000
Recv: echo:  G29 W I1 J4 Z0.05250
Recv: echo:  G29 W I2 J4 Z0.10000
Recv: echo:  G29 W I3 J4 Z0.13750
Recv: echo:  G29 W I4 J4 Z0.14000
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H185 B45 F255
Recv: echo:  M145 S1 H240 B110 F255
Recv: echo:; PID settings:
Recv: echo:  M301 P35.82 I4.10 D78.25
Recv: echo:  M304 P84.12 I14.78 D319.22
Recv: echo:; Z-Probe Offset (mm):
Recv: echo:  M851 X-45.00 Y-4.00 Z-2.90
Recv: echo:; Stepper driver current:
Recv: echo:  M906 X580 Y580 Z580
Recv: echo:  M906 T0 E650
Recv: 
Recv: echo:; Driver stepping mode:
Recv: echo:  M569 S1 X Y Z
Recv: echo:  M569 S1 T0 E
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.00
Recv: echo:; Filament load/unload lengths:
Recv: echo:  M603 L350.00 U400.00
Recv: ok
```

oschni, 2020-09-03:
Today I flashed the board again with the firmware from yesterday. Only the procedure of flashing are helping that the steppers are not getting hot.

Is it possible that there are some registers set different then the situation after a cold boot? 
My Solution is, the SD Card will remain into the controller board and after every cold/warm boot it will "re-flash" the board. We'll see how long the board will live when it will be flashed every boot.

hamphh73, 2020-09-06:
I also had problems with hot motors and layer shift on skr mini 1.2. The solution for me was the answer from @thisiskeithb (thanks!!!). In Configuration_adv.properties all SLAVE_ADDRESS must be 0 (for the version 1.2 board),

Jaxx0174, 2020-09-28:
I just installed the SKR Mini E3 v2.0 on my Ender 5 pro. I made my own firmware configuration with input from the premade configs on some issues. First I had issues with steps. It was set to 80 for both x and y, but after trial and error I found that x needs to be 41.5 and y is 300. X is set to 800 and needs to probably go up but I haven't messed with those yet. What I'm experiencing is the motors get extremely hot while printing, especially the X axis more than the others. And as they heat up they get very noisy, not just motor noises but squeaking as well, which did not happen with the stock board. I thought maybe the current was too high, but RMS was set to 580 for x and z, and 650 for y and e. The btt config has them all at 800, and since they are 1.5a motors, the rms should be at like 1000 or so, so they should be way under on voltage. I tried setting them to 800 and started a benchy, but stopped after a few layers as they were starting to get noisy again

john-clark, 2021-02-06:
I purchased https://www.amazon.com/gp/product/B081DHW839

BIGTREETECH Diect SKR Mini E3 V1.2 32bit

I flashed 

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V1.2/firmware-bltouch-for-z-homing.bin

Stepper motors on X and Y are very noisy and too hot to touch, 80c.

Will this be fixed or should I return product? OEM board worked without overheating steppers.

Jaxx0174, 2021-02-06:
It looks like I got this email because you replied to a thread I was in on GitHub. What I found out my problem was, was that the stepper motor drivers were not set correctly in Marlin. It's been awhile since I looked into it, but do a search for driver and find the stepper motors and make sure they are set correctly, should be 2209 I believe, but double check for yourself.Sent from my Galaxy

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V1.2/firmware-bltouch-for-z-homing.bin

john-clark, 2021-02-15:
So if I am reading your post right, the default firmware btt has on their website is not compatible with the oem steppers?

Jaxx0174, 2021-02-15:
Not exactly. Just that it wasn’t configured properly for their board. At least for the version I got. Here is a screenshot of the code that I’m currently using. Its under the Configuration.h file. When I first tried to use the new board, these were set to TMC2208, which is what the stepper drivers on the OEM board are. However the BTT board use TMC2209. Once I switched these to 2209, I no longer had any issues with the motors overheating.

If yours are set to 2209 already then I would say that your problem is different than the one that I had.

From: John Clark
Sent: Monday, February 15, 2021 5:00 PM
To: bigtreetech/BIGTREETECH-SKR-mini-E3
Cc: Jaxx0174; Comment
Subject: Re: [bigtreetech/BIGTREETECH-SKR-mini-E3] Stepper motor current control (#22)

So if I am reading your post right, the default firmware btt has on their website is not compatible with the oem steppers?
—

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

chrisconnew, 2021-09-16:
Been having this problem with very hot steppers - the extruder to the point it's warming the filament up and i have just had to remove a pigtailed bit of pla which required fully disassembling the extruder and connectors to get it out.

Tried recommendations online  previously to reduce them eg 580 for the extruder but that is the current i had when i got filament heated at the extruder.
Dropped to 350xyz and 420e and so far all test prints are fine, even with retraction on the extruder stepper is only 30c, with the others a few degrees lower. Fingers crossed