bigtreetech/BIGTREETECH-SKR-mini-E3 #23: Can not build bin file

PeeZee, 2019-08-29:
hello, I followed the instructions https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/tree/master/firmware , XYZE0_DRIVER_TYPE I have TMC2208_STANDALONE
[image]

I got these error messages ... in image

Where is the problem ?

styxxen80, 2019-08-29:
I get the exact same error when trying to compile the firmware

PeeZee, 2019-08-29:
problem solved with "**MOTHERBOARD BOARD_BIGTREE_SKR_E3_DIP**" :-D

styxxen80, 2019-08-29:
so we should use an incorrect board in order to be able to compile?
Because it doesn´t work with **"BOARD_BIGTREE_SKR_MINI_E3"** 

I was able to get it to compile by changing platformio.ini file to point to the latest repository for U8glib. (line 29)
lib_deps =
U8glib-HAL=**https://github.com/MarlinFi.../U8glib-HAL/archive/bugfix.zip**

and also, you need to update the folder Marlin-2.0.x-SKR-Mini-E3\Marlin\src\lcd\menu with the contents from the latest Vanilla Marlin (which has renamed **menu_sd** to **menu_media**

brew99, 2019-08-30:
I think the OP must have the DIP version, since they mentioned having 2208 drivers.   Posted in wrong repo is my guess.   That being said, I haven't tried compiling the BTT firmware here for the mini E3 board

minzukas, 2019-10-10:
have the same:

> Executing task in folder Marlin-2.0.x-SKR-Mini-E3-v1.2: platformio run <

Error: Unknown environment names 'BOARD_BIGTREE_SKR_MINI'. Valid names are 'melzi, STM32F103R_fysetc, teensy31, rambo, at90usb1286_cdc, megaatmega1280, mks_robin, BIGTREE_SKR_PRO, mks_robin_nano, STM32F103V_longer, STM32F103R, STM32F407VE_black, mks_robin_mini, at90usb1286_dfu, sanguino_atmega1284p, STM32F4, STM32F7, ARMED, BIGTREE_BTT002, STM32F103R_bigtree, megaatmega2560, teensy35, DUE, include_tree, jgaurora_a5s_a1, STM32F103R_bigtree_USB, DUE_USB, DUE_debug, SAMD51_grandcentral_m4, esp32, sanguino_atmega644p, malyanm200, LPC1768, LPC1769, linux_native, melzi_optiboot, mks_robin_lite, fysetc_f6_13'
The terminal process terminated with exit code: 1