bigtreetech/BIGTREETECH-SKR-mini-E3 #24: LCD backlight is flickering when heaters are active

creamywhite, 2019-08-29:
I have a stock Ender 3 display and it's backlight flickers (changes intensity) slightly if hotend or heatbed is powered. I suspect it's a hardware issue - is there any known fix for this?

hopatz, 2019-09-01:
Same problems here

creamywhite, 2019-09-02:
I was able to solve the issue by soldering 4.7 uF capacitor across +5V and GND pins of EXP1 connector. Seems that the board has some electric design flaws after all. My BLTouch (which worked flawlessly before on a stock Ender 3 board) also would randomly deploy its pin during prints until I added 100 nF decoupling capacitor across PA1 anf GND pins of SERVO connector.

sky8000, 2019-09-03:
Hi, that happened to me too.

Check if you have PID enabled for hot bed and disable it (comment #define PIDTEMPBED ). This should also fix your problem.

creamywhite, 2019-09-03:
> Hi, that happened to me too.
> 
> Check if you have PID enabled for hot bed and disable it (comment` #define PIDTEMPBED` in **Configuration.h**). This should also fix your problem.

Thanks for the suggestion. However, disabling a useful feature (which bed PID is) to workaround a possible electrical design issue seems wrong to me.

sky8000, 2019-09-04:
> Thanks for the suggestion. However, disabling a useful feature (which bed PID is) to workaround a possible electrical design issue seems wrong to me.

I agree that it is an useful feature but not a requirement for the heated bed where there's no need for very high precision temperatures. Anyway I noticed that Bigtreetech has an external heated bed power module called BTMOS v2.0 that supports current up to 15A (https://www.aliexpress.com/item/32874506353.html). I wonder if the module would solve this problem.

creamywhite, 2019-09-04:
> I agree that it is an useful feature but not a requirement for the heated bed where there's no need for very high precision temperatures. Anyway I noticed that Bigtreetech has an external heated bed [...]

First of all, it is debatable whether heatbed temperature stability is necessary or not. Secondly, using bang-bang heating method puts more load on electrical components. Yes, the MOSFET that is used for bed heating control (IPD036N04L) is rated well above 10A that Ender 3's heatbed can draw, so there shouldn't be any issues. I'm just a bit frustrated that SKR Mini E3 is designed badly in some aspects causing users to seek workarounds if they want to use proven useful features.

JimmySnails, 2019-09-12:
@creamywhite i noticed the same thing but without PWM. `#define PIDTEMPBED` is disabled per default and still my LCD brightness jumps every now and then. It doesn't happen too often but it's annoying anyway and i'd like to fix that.

So i'd like to try adding capacitor like you did. I'm just confused you said you added it on EXP1? My LCD is connected via EXP3 (like on the stock board). So did you add the capacitor between the empty pins on the LCD screen? or on the ribbon cable? If you used the ribbon cable, which colors are  +5V and GND ?

Thank you!

creamywhite, 2019-09-15:
> @creamywhite i noticed the same thing but without PWM. `#define PIDTEMPBED` is disabled per default and still my LCD brightness jumps every now and then. It doesn't happen too often but it's [...]

Sorry for a late answer. I soldered a capacitor directly between 5V and GND pins of EXP1 connector on the board. You can refer to this diagram to locate them https://cdn.shopify.com/s/files/1/1619/4791/files/SKR-mini-E3_05.jpg?v=1562825161. Also you can solder a capacitor of a higher value than 4.7 uF. As I noticed it won't solve the issue completely, but the screen will flicker less.

kirelam, 2019-09-26:
Hi,

I am considering to buy the SKR mini E3 for a while now. 

But i read that there seems to be an issue with the 5V voltage regulator. It seems to leak 1.5V to the USB port and it doesn't seem to be stable, where it causes strange beheviour of some components, like the BLtouch. Is there a new revision planned, or is there a modification known here? i am willing to do this, but i do not know what i have the change or add to the board. I have some Solder Skills, so that is no problem!

Many regards,
Kirelam

barko99, 2020-04-16:
Hey there,

even this discussion is a bit old, i want to comment my issues with the skr regarding EMI:

After fiddling around with uneven extrusion over height, (changing nearly everything regarding z axis) i switched over to check if my uneven extrusion is in just one layer. Same here.  More or less periodically, i have a lowered extrusion and z axis. I changed everything regarding extrusion (motor, extruder, bowden, nozzle, etc. etc.)

After a lot of head scratching, i noticed that i have these issues when the heatbed is heating. It is reproducible, so while printing, i can change to a higher bed temp, and suddenly i have the issues with lower extrusion feedrate and my z axis drops down slightly. 

I do not have solved the issue right now, i want to try an external mosfet the following days and switching off the PID bed.

Best regards and wishes

Johannes

txt4nk, 2020-05-14:
Did anyone ever completely solve this?

barko99, 2020-05-14:
Honestly, i can't really help you with display flickering. My above issue was caused (or exaggerated) by my swap heatbed. 

I had an Fsytec magnetic heatbed installed. I changed nearly everything while trying to find the issue. (external mosfet, external mosfet for heatbed and hotend, new PSU, second new PSU (one for heatbed, one for hotend and SKR, .... .... .... )

The last thing i thought of was the heatbed. I couldn't see why and how this could effect my SKR. Still, while not knowing what the issue is with this bed i swaped to the stock bed again. The stock bed and SKR works now perfectly in my case.

Regards

txt4nk, 2020-05-14:
When I had my E3 mini, I was on an oem heat bed. After the E3 mini fried, I installed the th3d ezboard lite along with a spool3d forged aluminum heat bed with integrated magnets (amazing bed, btw!) And i never once had this issue. As soon as I put the E3 DIP in the printer, the problem returned. I have a meanwell psu on the way to replace the landy, along with a bunch of ferrite chokes. If that doesn't work ill end up soldering a resistor on to EXP1 and may even rewire my printer with all shielded cables.. kind of ridiculous though, I guess thats why the ezboard costs so much..

txt4nk, 2020-05-14:
So I got a response to the issue I created about this that I have yet to see, and could possibly help..
https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/issues/72#issuecomment-628454817

txt4nk, 2020-05-15:
well, that comment I linked to 100% worked. I've had the screen sitting to the side of the printer, not touching the frame at all and the screen has not spazzed out even once. It happened on every single print before this. I am designing a sturdy buffer\stand off to prevent metals from touching so I can re-attach it to the frame without EMI issues.

txt4nk, 2020-05-16:
Here's the buffer\stand off I created. It works perfectly, I have not had a single glitch since installing this. It's currently 3mm thick, it's sturdy enough but it does have a little more flex than i'd like so I'm going to revise it and go to 6mm. I can provide the STL for whoever has an Ender 5.

[image]
[image]

paul-1, 2020-05-16:
I think most just remove the screen, and protect the edge of the screen with electrical tape.

creamywhite, 2020-05-16:
@txt4nk, it nice that you've found a solution to your issue, but I'm afraid it has nothing to do with my issue (and this topic respectively) in terms of its cause and solution. The thing is, in my case, the LCD screen's backlight will flicker slightly when heatbed power is regulated through PWM even when the LCD screen board is not attached to anything (save for EXP cable).

txt4nk, 2020-05-16:
Ah.. I guess I misunderstood the issue. Still seems to be some sort of EMI in both cases. Have you tried ferrite rings or chokes? Could also try shielded cables.

creamywhite, 2020-05-16:
> Ah.. I guess I misunderstood the issue. Still seems to be some sort of EMI in both cases. Have you tried ferrite rings or chokes? Could also try shielded cables.

I did not and I believe they won't help (might try just out of curiosity if I find one). As I stated in my previous messages here, the issue is mitigated to a degree by a capacitor across VCC and GND pins.

txt4nk, 2020-05-16:
I'd throw some chokes on the PSU to main board wires, main board to bed and main board to hot end.  IF the issue is something a choke would stop, those three points would stop it regardless of the source, whether it be the PSU or the main board. Chokes are super cheap, so it's worth trying at least.