bigtreetech/BIGTREETECH-SKR-mini-E3 #26: Hybrid mode skr e3 mini problems at Y axis

hopatz, 2019-09-01:
Test hybrid mode using skr e3 mini tmc2209 and got problems at Y axis wont go silent... run at 60mm/s, threshold mode is 120mm/s

X and Z axis is silent.. but no for Y axis, try increasing threshold value but doesnt help

tomoinn, 2019-09-04:
The default firmware doesn't have hybrid mode enabled 

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/blob/7b306d89096a82895b7fff2cdd3997b47930d026/firmware/Marlin-2.0.x-SKR-Mini-E3/Marlin/Configuration_adv.h#L1901-L1922

Is the section you need, and you'd need to uncomment `#define HYBRID_THRESHOLD`

Are you sure you're not in stealthchop mode though? Could just be a noisy mechanism, use `M569` from a terminal to get the current operating modes for each driver?

ceppelli, 2019-09-10:
Hi @hopatz

I'm not sure if it is related with your problem, but I found a issue with the Y_SLAVE_ADDRESS configuration for the issue #29. Maybe you could take a look on it

Hope this help.