bigtreetech/BIGTREETECH-SKR-mini-E3 #28: E0 defaults to Stealthchop, causing issues with geared extruders

tomoinn, 2019-09-04:
I found I was getting inconsistent extrusion with my BMG clone, switching the extruder driver back into spreadcycle cleaned it up so presumably under stealthchop it's missing steps. I've fixed it by adding 

```
M569 S0 E
```

To my start gcode in cura, but it's probably better to have it off by default in the firmware.

brianredbeard, 2019-09-29:
Likely this may just need better tuning of the settings.  Back in March there was a thread in Klipper with similar issues.  @elcojacobs engaged in quite a bit of yeoman's work to figure this out:

https://github.com/KevinOConnor/klipper/issues/1363

elcojacobs, 2019-09-29:
Yep, I'm running the extruder with stealthchop, which is possible with the right settings for the drivers. I looked at the datasheet and did some tests and found settings that worked well and did not overheat the drivers. The missing steps are due to the driver overheating and resetting itself.

You need to tune the aggressiveness and then you can run silently and with great performance. See the thread @brianredbeard linked for my current settings. The loudest thing on my Ender 3 are the fans.

tomoinn, 2019-09-29:
Thanks for that - I'll take a look and see whether I can reproduce that with Marlin.

brianredbeard, 2019-10-05:
@tomoinn were you able to reproduce the issue?   I see that it's still open on the repository.

tomoinn, 2019-10-06:
I wasn't able to configure Marlin in the way you've got Klipper set up, there don't appear to be the equivalent parameters for the TMC2209 drivers. Could be that I'm missing something buried in the driver code and it's possible, but there's nothing obvious. For now I'd stick with my original 'disable stealthchop' option, especially when paired with linear advance.

tomoinn, 2019-10-06:
Seems a lot of people have this issue, or one related. https://github.com/MarlinFirmware/Marlin/issues/14634 is an upstream bug, there are conflicting reports of using different modes, pulse lengths, square wave pulse etc etc. I'm waiting on a linear rail kit so will be doing a full rebuild next week anyway, I may have time to experiment with this a bit more while I recalibrate after that.