bigtreetech/BIGTREETECH-SKR-mini-E3 #32: TMC Connection Error

ManiRX100, 2019-10-06:
Hi
I followed the same procedure as given for the Marlin Firmware upgrade and got the firmware file. But when I upgrade the board, I get "TMC Connection Error" and the motors are not working properly. I have a BL Touch also installed. And also the board shown in the supplier site and the one I received are a little different near the USB end. Attached is the image.  Need help in this regard. 
[image]

davehardy20, 2019-10-06:
This is the same for me, I received the board yesterday, and I followed the same procedure as ManiRX100, I have the error 'TMC Connection Error' on the display, however my motors are working!
My BLTouch is not however working properly, when the print starts up the BLTouch self-tests and then once Marlin is booted it does not respond to the commands in the menu ie Reset, deploy etc.

ManiRX100, 2019-10-06:
Hi Dave,
Even I received the board yesterday. The whole day today I was trying to figure this out. Still was not able to solve. My motors are also working.. But in one direction + and not the other way. It doesnt move after 0.00 value.

mrBarbecue, 2019-10-06:
Had the same problem. Removed the red jumper as shown in the picture, so the display showed "Ready". Unfortunately, there is no manual showing us, which jumper configuration means what, but I'm sure, it's about the supply voltage.

ManiRX100, 2019-10-06:
Which jumper should we remove? There is another one at the other end of the board. Actually I removed the one shown in the image I uploaded, but still got the same error.

Paddzr, 2019-10-06:
Tried removing the jumper, still got TMC connection error.

davehardy20, 2019-10-06:
I see a firmware was released 4 hrs ago, marked as 1.2 but cannot try it out as I have no time at the moment

Paddzr, 2019-10-06:
> I see a firmware was released 4 hrs ago, marked as 1.2 but cannot try it out as I have no time at the moment

Tried compiling it, but arduino and platformio both fail.

glenmorangie007, 2019-10-06:
Paddzr - you must install Git client - I've tried this direct from the Zip file and it fails to compile

Paddzr, 2019-10-06:
> Paddzr - you must install Git client - I've tried this direct from the Zip file and it fails to compile

So I installed it but didn't do anything... Tried again and restarted after. Got this error now:

```
.pio\libdeps\STM32F103R_bigtree\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp: In member function 'void Adafruit_NeoPixel::show()':
.pio\libdeps\STM32F103R_bigtree\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp:1736:3: error: 'Pio' was not declared in this scope
   Pio            *port;
   ^~~
.pio\libdeps\STM32F103R_bigtree\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp:1736:19: error: 'port' was not declared in this scope
   Pio            *port;
                   ^~~~
.pio\libdeps\STM32F103R_bigtree\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp:1736:19: note: suggested alternative: 'word'
   Pio            *port;
                   ^~~~
                   word
Compiling .pio\build\STM32F103R_bigtree\libba8\SlowSoftI2CMaster\SlowSoftI2CMaster.cpp.o
.pio\libdeps\STM32F103R_bigtree\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp:1737:12: error: 'WoReg' does not name a type
   volatile WoReg *portSet, *portClear, *timeValue, *timeReset;
            ^~~~~
.pio\libdeps\STM32F103R_bigtree\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp:1740:3: error: 'pmc_set_writeprotect' was not declared in this scope
   pmc_set_writeprotect(false);
   ^~~~~~~~~~~~~~~~~~~~
.pio\libdeps\STM32F103R_bigtree\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp:1741:35: error: 'TC3_IRQn' was not declared in this scope
   pmc_enable_periph_clk((uint32_t)TC3_IRQn);
                                   ^~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\STM32F103R_bigtree\lib578\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp.o] Error 1
=============================================================================================================================================== [FAILED] Took 43.81 seconds =
```

glenmorangie007, 2019-10-06:
Snap PADDZR - I've emailed the results to BigTREE !!

glenmorangie007, 2019-10-06:
PADDZR

at least you now have GIT installed so you can compile

Paddzr, 2019-10-06:
well, mostly. Currently got either board with stepper motors but no BLTouch or board with BLTouch but no stepper motors what so ever. I don't think prints will turn out great without stepper motors being enabled.

I'll wait few days and see how it goes. Maybe someone else comes up with a solution!

glenmorangie007, 2019-10-06:
I'm getting tempted about ripping the damn thing out and going back to my original Ender 3 Board !!

davehardy20, 2019-10-06:
I have put my Creality 1.1.5 board back in for the time being until a solution is found, Bigtreetech need to provide us some decent documentation for this board, if they don't a really awesome upgrade is just gonna go by the wayside. Come on Bigtreetech step up

martyboi83, 2019-10-06:
I am also having the TMC connection error, it is the new board with the different pins.

davehardy20, 2019-10-06:
Regarding the firmware, I took a look at it and it looks like a new upstream version with new board names, platfom.ini default_envs = STM32F103RC_bigtree from Marlin. So I grabbed the latest Marlin 2.0_bugfix and configured it with the above default envs value, and set the other settings as per the setting from the ReadMe.md in this repository. And compiled the firmware in VsCode. Which completed successfully. If someone could try this out on the new SKR E3 Mini board it might help finding a fix. I have pushed the copy of my Marlin 2.0 bugfix build and firmware to my Github here - https://github.com/davehardy20/Marlin_2.0_Bugfix_SKR_E3_Mini_Test, the compiled firmware is in the .pio\build\STM32F103RC_bigtree folder.

Paddzr, 2019-10-06:
> Regarding the firmware, I took a look at it and it looks like a new upstream version with new board names, platfom.ini default_envs = STM32F103RC_bigtree from Marlin. So I grabbed the latest Marlin 2.0_bugfix and configured it with the above default envs value, and set the other settings as per the setting from the ReadMe.md in this repository. And compiled the firmware in VsCode. Which completed successfully. If someone could try this out on the new SKR E3 Mini board it might help finding a fix. I have pushed the copy of my Marlin 2.0 bugfix build and firmware to my Github here - https://github.com/davehardy20/Marlin_2.0_Bugfix_SKR_E3_Mini_Test, the compiled firmware is in the .pio\build\STM32F103RC_bigtree folder once compiled or I put my version in the Releases of my Github repo - https://github.com/davehardy20/Marlin_2.0_Bugfix_SKR_E3_Mini_Test/releases/tag/1.0

Both are 404 right now.

davehardy20, 2019-10-06:
Srry forgot to make it Public, should be fine now

Paddzr, 2019-10-06:
#define Y_DRIVER_TYPE was set to TMS not TMC, 

Tried your bin and then tried compiling the new one from your test source. Both resulted with the TMC Connection Error message.

davehardy20, 2019-10-06:
I have no errors when I build the firmware, only a deprecated library warning at the beginning. Make sure you are online and VSCODE can download any requirements

Paddzr, 2019-10-06:
I think they're warnings more than errors, they're yellow, not red. It compiles it successfully. I've tried few different TMC numbers just to test but all come with the same error.

I've emailed them about it, they just need to release the source code that now comes on the SD card as that one works, but majority of us got to modify it to fit...

Paddzr, 2019-10-06:
https://github.com/Paddzr/SKR-Mini-E3-Stock-FIRMWARE.BIN

Not sure if it's possible to reverse engineer this.

ManiRX100, 2019-10-07:
The new marlin compiled successfully but with the following warnings:
1. .pio\libdeps\STM32F103RC_bigtree\U8glib-HAL_ID1932\src\clib\u8g_dev_ssd1306_128x32.c:153:22: warning: 'u8g_dev_ssd1306_128x32_univision_init_seq' defined but not used [-Wunused-const-variable=]
 static const uint8_t u8g_dev_ssd1306_128x32_univision_init_seq[] PROGMEM = { 
2. C:\users\manil\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\STM32ADC\src\utility\util_adc.c:10:30: warning: 'adc_result' initialized and declared 'extern'
 extern volatile unsigned int adc_result = 0;

When I upload this firmware, I am not getting characters in my display. Its just getting on with a blank blue screen.  Hope Bigtreetech comes up with a solution soon for everyone!

ManiRX100, 2019-10-07:
Hi
Can anyone post the image of the new board? I think I have misplaced the two jumpers. I will try to remove the BL Touch and install the Z-Stop switch. Run the stock firmware uploaded by Paddzr and check.

davehardy20, 2019-10-07:
[image]
This my un-modified board.

ManiRX100, 2019-10-07:
> Regarding the firmware, I took a look at it and it looks like a new upstream version with new board names, platfom.ini default_envs = STM32F103RC_bigtree from Marlin. So I grabbed the latest Marlin [...]

Still the same TMC connection error. What are we supposed to do? Any ideas that Bigtreetech might upload their latest working code in GitHub? Anyone get a reply from them?

davehardy20, 2019-10-07:
The repo has had some updates, since yesterday and BTT have made reference to the the v1.2.zip, which contain firmware. I have tried to compile this and it fails, see the new issue #33.

ManiRX100, 2019-10-07:
> 
> 
> The repo has had some updates, since yesterday and BTT have made reference to the the v1.2.zip, which contain firmware. I have tried to compile this and it fails, see the new issue #33.

Yes.. I too tried with that now and noticed your new thread also. I get an error "command 'platformio-ide.build' not found".

Stew-rt, 2019-10-07:
I don't get this error showing up if I compile against the version newly uploaded to github.
However, I can only confirm that the steppers work as this is all I have tested, as I'm having a seperate issue where I can't get it to power on if my BLTouch is plugged in to either the PROBE or Z-STOP pins. (this was also true of firmware before this version)

Paddzr, 2019-10-07:
> I don't get this error showing up if I compile against the version newly uploaded to github.
> However, I can only confirm that the steppers work as this is all I have tested, as I'm having a seperate [...]

Hey, swap your cables around, I had the same issue. The layout doesn't seem to be the same on the 3 pin connector. If your bltouch comes with quick detach cables, just rotate that.

I do however have issues compiling my code....

This is the error: IOError: [Errno 2] No such file or directory: 'c:\\Users\\patry\\Desktop\\Marlin-bugfix-2.0.x\\BIGTREETECH-SKR-mini-E3-master\\firmware\\Marlin-2.0.x-SKR-Mini-E3-v1.2\\.pio\\libdeps\\STM32F103R_bigtree\\_tmp_installing-qeeuxs-package\\SailfishLCD-master\\examples\\FlashForgeInterfaceBoardLCD\\FlashForgeInterfaceBoardLCD.ino':

davehardy20, 2019-10-07:
> I don't get this error showing up if I compile against the version newly uploaded to github.
> However, I can only confirm that the steppers work as this is all I have tested, as I'm having a seperate [...]

This is from the FaceBook group page - Hello,here is BIGTREETECH SKR MINI E3 V1.2 notes: We had upgraded some place for SKR MINI E3 V1.2,more detail pls see our file. If your SKR MINI E3 is V1.2 version,pls use our new firmware and we has been uploaded to our github.
About some users are unstable when using BLTOUCH V3.1,Please try :
Use the Z-Stop (PC2) connector. It is more stable than Probe (PC14) connector.
In configuration.h, delete // as below.
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN'

Paddzr, 2019-10-07:
Ok so it compiles, stepper motors are active, now BL touch is giving error. When I try to home it just stops and knocks my octoprint. If I home on the printer it just stays STOPPED. It appears to go up a tiny bit. maybe Z stop issue? I did as dave mentioned and uncommented the #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN'

Stew-rt, 2019-10-07:
Looks like I've been bitten by a change of heart between board versions.
I've just seen the newly uploaded schematic.
Old Board SERVOS socket: |PA1|GND|+5V|
1.2 Board SERVOS socket: |GND|+5V|PA1|
(PROBE and Z-STOP remain the same, |GND|PC14| / |GND|PC2|)
(Just for future reference if anyone has this issue)

ManiRX100, 2019-10-08:
> Looks like I've been bitten by a change of heart between board versions.
> I've just seen the newly uploaded schematic.
> Old Board SERVOS socket: |PA1|GND|+5V|
> 1.2 Board SERVOS socket: |GND|+5V|PA1| [...]

Are there any modifications needed in the "SKR MINI E3 V1.2" marlin for the BL Touch additionally apart from the changes made by BTT? Can you send the connection image in board for reference please?

ManiRX100, 2019-10-08:
Got it to work currently without BL Touch. The nozzle fan is no noisy compared to before.  I connected it to the fan pin near the X-end stop connector. Is that correct?

glenmorangie007, 2019-10-08:
Connect the nozzle fan to the outer connect and the case cooler to the inner connector

glenmorangie007, 2019-10-08:
Has anyone got a working firmware now which has been tried and tested ??

Paddzr, 2019-10-08:
I’ve got it working. Printed a benchy at .12 layer with .2mm nozzle at 150mm/sec. turned out quite well. With bltouch.

davehardy20, 2019-10-08:
@Paddzr please could you share what you did to get this going? Thanks

ManiRX100, 2019-10-08:
@Paddzr  Yes.. Please share us. BL touch pin connection and the changes in Marlin.. It will be of immense help.

glenmorangie007, 2019-10-08:
Big tree tech have come back and said the firmware will now compile so hopefully I can get the board working properly

Paddzr, 2019-10-08:
Redownload the zip file and it should compile.

ManiRX100, 2019-10-08:
At last made it to work with Bl Touch. The latest firmware from BTT compiles with no issues so far. Thanks everyone for the immense support. Shall I close this thread?

ManiRX100, 2019-10-08:
> 
> 
> Got it to work currently without BL Touch. The nozzle fan is so noisy compared to before. I connected it to the fan pin near the X-end stop connector. Is that correct? Also, can anyone say how to add this board to the Arduino IDE so that I can compile the firmware from Arduino IDE. I checked through the Get board info, it says "Unknown Board".

In between, can anyone suggest how to add this board to Arduino IDE?

glenmorangie007, 2019-10-08:
@ManiRX100 how can you make the BLTouch do the Auto Home from the bed as apposed to going all the way to the x endstop where it is currently doing it ??

glenmorangie007, 2019-10-08:
found it - you have to enable Z_SAFE_HOMING

closed 2019-10-09 by ManiRX100

teoyh, 2019-10-10:
I was having the TMC connection error , i saw the firmware posted by Paddzr 

https://github.com/Paddzr/SKR-Mini-E3-Stock-FIRMWARE.BIN

I downloaded it and flash it to my board and the TMC connection error was gone. 

But my usb connection is not working , how do i compile a new firmware that will not have TMC connection error problem but will get my usb to work in Cura. 

I am using vscode + platformio to compile my firmware , i encounter issue when doing the compilation, is i could not find a right name for the env_default = BIGTREE_SKR_MINI

When i use BIGTREE_SKR_MINI it always throw me error say name is not valid.

ManiRX100, 2019-10-10:
@teoyh . The right name is : env_default = STM32F103R_bigtree.
Download the latest firmware here: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/Marlin-2.0.x-SKR-Mini-E3-v1.2.zip

teoyh, 2019-10-10:
> @teoyh . The right name is : env_default = STM32F103R_bigtree.
> Download the latest firmware here: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/Marlin-2.0.x-SKR-Mini-E3-v1.2.zip
> This firmware is the latest that worked for the most of us.

Thank you so much i will go and try

glenmorangie007, 2019-10-10:
you don't need to make any changes to the SKR Mini E3 v1.2 unless you are using the BLTouch then use the 

Use the Z-Stop (PC2) connector. It is more stable than Probe (PC14) connector.
In configuration.h, delete // as below.
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

check the wiring as the pin configuration for the 3 pin Servo connection is different on this new board

also worth noting that if you are using the BL Touch, you also need to enable the #define Z_SAFE_HOMING or the BL touch will try and autohome at the Endstop on the x axis

glenmorangie007, 2019-10-10:
have you install GIT client - you need this to compile !

teoyh, 2019-10-10:
Oh I see I never install GIT client, I will try again but tomorrow ...very sleepy now, good night

martyboi83, 2019-10-10:
HI, I am still having issues with the TMC connection error? I have compiled the newest v1.2 release in  platformio. I can get it to compile and have burned the firmware to the board using the details in the readme.md (changing the enviroment to STM32F103R_bigtree and MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V1_2) 

I get a few errors in yellow when it compiles:
pio\libdeps\STM32F103R_bigtree\U8glib-HAL_ID1932\src\clib\u8g_dev_ssd1306_128x32.c:153:22: warning: 'u8g_dev_ssd1306_128x32_univision_init_seq' defined but not used [-Wunused-const-variable=]
 static const uint8_t u8g_dev_ssd1306_128x32_univision_init_seq[] PROGMEM = {
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\STM32F103R_bigtree\U8glib-HAL_ID1932\src\clib\u8g_dev_ssd1306_128x32.c:83:22: warning: 'u8g_dev_ssd1306_128x32_adafruit2_init_seq' defined but not used [-Wunused-const-variable=]
 static const uint8_t u8g_dev_ssd1306_128x32_adafruit2_init_seq[] PROGMEM = {
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\STM32F103R_bigtree\U8glib-HAL_ID1932\src\clib\u8g_dev_ssd1306_128x32.c:48:22: warning: 'u8g_dev_ssd1306_128x32_adafruit1_init_seq' defined but not used [-Wunused-const-variable=]
 static const uint8_t u8g_dev_ssd1306_128x32_adafruit1_init_seq[] PROGMEM = {
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pio\build\STM32F103R_bigtree\libcda\U8glib-HAL_ID1932\clib\u8g_dev_ssd1306_64x48.c.o
Compiling .pio\build\STM32F103R_bigtree\libcda\U8glib-HAL_ID1932\clib\u8g_dev_ssd1309_128x64.c.o
.pio\libdeps\STM32F103R_bigtree\U8glib-HAL_ID1932\src\clib\u8g_dev_ssd1306_128x64.c:146:22: warning: 'u8g_dev_ssd1306_128x64_univision_init_seq' defined but not used [-Wunused-const-variable=]
 static const uint8_t u8g_dev_ssd1306_128x64_univision_init_seq[] PROGMEM = {
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\STM32F103R_bigtree\U8glib-HAL_ID1932\src\clib\u8g_dev_ssd1306_128x64.c:44:22: warning: 'u8g_dev_ssd1306_128x64_adafruit1_init_seq' defined but not used [-Wunused-const-variable=]
 static const uint8_t u8g_dev_ssd1306_128x64_adafruit1_init_seq[] PROGMEM = {
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\users\martyboi\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\STM32ADC\src\utility\util_adc.c:10:30: warning: 'adc_result' initialized and declared 'extern'
 extern volatile unsigned int adc_result = 0;

I have sent a M122 in pronterface and all I get back is
SENDING:M122
Testing X connection... OK
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW

This is driving me mad now, any help would be greatly appreciated

glenmorangie007, 2019-10-10:
@martyboi83 see here :- my compiled FIRMWARE for my Ender 3 - change things in your printer menu before printing - especially the Esteps for your extruder

https://github.com/glenmorangie007/Test

martyboi83, 2019-10-10:
Awesome! thank you, it is working with my v3.1 bltouch

Any idea why i cant get my firmware to work?

ManiRX100, 2019-10-11:
> HI, I am still having issues with the TMC connection error? I have compiled the newest v1.2 release in platformio. I can get it to compile and have burned the firmware to the board using the details [...]
Yellow colored are warnings, not errors. I got the same warnings, but the firmware works.

ManiRX100, 2019-10-11:
@martyboi83 Yellow colored are warnings, not errors. I got the same errors, but the firmware works fine.

Arganaz, 2019-10-11:
Just comment  EEPROM_SETTINGS as defined in [https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2)
TMC ERROR will be gone.

glenmorangie007, 2019-10-11:
@Arganaz EEPROM_SETTINGS are defined in mine as the release originally has it in and I don't get any TMC Errors

wstrzalka, 2019-11-02:
I had the same. Ensure you're powering the board with main power not via USB. Otherwise you'll get "TMC Connection error" regardless what version you try.

[20 more replies]