bigtreetech/BIGTREETECH-SKR-mini-E3 #33: Cannot compile the new v1.2 firmware.

davehardy20, 2019-10-07:
I have tried to compile the new firmware as I need to change the BLTouch offsets in relation to the nozzle as I have a non-standard setup. I've downloaded and extracted the firmware made the changes for the offset and attempted to compile. The process fails as shown below;

[image]
 
The library is download successfully as shown below;

[image]

Stew-rt, 2019-10-07:
It Compiled for me if I added

```
[env:STM32F103R_bigtree]
...
lib_ignore        = Adafruit NeoPixel, SPI
...
```

to `platformio.ini`

as I don't need NeoPixel Support...

davehardy20, 2019-10-07:
great it compiled for me now, thanks @Stew-rt

Paddzr, 2019-10-07:
I wish I had a clue what I was doing... When I unzip the v1.2 it just comes up with missing lib: 
IOError: [Errno 2] No such file or directory: 'c:\\Users\\patry\\Desktop\\Marlin-bugfix-2.0.x\\BIGTREETECH-SKR-mini-E3-master\\firmware\\Marlin-2.0.x-SKR-Mini-E3-v1.2\\.pio\\libdeps\\STM32F103R_bigtree\\_tmp_installing-xsoi9e-package\\SailfishLCD-master\\examples\\FlashForgeInterfaceBoardLCD\\FlashForgeInterfaceBoardLCD.ino':

I'm not sure what I'm doing but went to library search in platformio and checked for updates but didn't find anything of the sorts.

bigtreetech, 2019-10-09:
Hello friend, it is fixed yesterday

dborio, 2019-10-19:
I am trying to compile V1.2 and this is what is happening any help would be great.

Processing STM32F103RC_bigtree (platform: ststm32; framework: arduino; board: genericSTM32F103RC)
-------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option

CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 5.6.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: tool-stm32duino 1.0.1, toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing Arduino-L6470
Using cache: C:\Users\Dan\.platformio\.cache\e0\a85bf8ed5b8232b743f62637d1e2c2e0
Unpacking...
IOError: [Errno 2] No such file or directory: 'C:\\Users\\Dan\\Desktop\\3D Printer Stuff\\Firmware\\BIGTREETECH-SKR-mini-E3-master\\firmware\\V1.2\\Marlin-2.0.x-SKR-Mini-
E3-V1.2\\.pio\\libdeps\\STM32F103RC_bigtree\\_tmp_installing-ldj3lp-package\\Arduino-L6470-dev\\examples\\two_motors_synchronized\\two_motors_synchronized.ino':
  File "C:\Users\Dan\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Dan\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Dan\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\Users\Dan\.platformio\platforms\ststm32\builder\main.py", line 91:
    target_elf = env.BuildProgram()
  File "C:\Users\Dan\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Dan\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 122:
    _build_project_deps(env)
  File "C:\Users\Dan\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 47:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Dan\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Dan\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1043:
    project.install_dependencies()
  File "C:\Users\Dan\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 878:
    lm.install(uri)
  File "c:\users\dan\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 359:
    force=force)
  File "c:\users\dan\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 701:
    track=True)
  File "c:\users\dan\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 483:
    self.unpack(dlpath, tmp_dir)
  File "c:\users\dan\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 216:
    raise e
========================= [FAILED] Took 7.31 seconds =========================
Environment              Status    Duration
-----------------------  --------  ------------
megaatmega2560           IGNORED
megaatmega1280           IGNORED
at90usb1286_cdc          IGNORED
at90usb1286_dfu          IGNORED
DUE                      IGNORED
DUE_USB                  IGNORED
DUE_debug                IGNORED
LPC1768                  IGNORED
LPC1769                  IGNORED
melzi                    IGNORED
melzi_optiboot           IGNORED
rambo                    IGNORED
sanguino_atmega644p      IGNORED
sanguino_atmega1284p     IGNORED
STM32F103RE              IGNORED
STM32F103RC_fysetc       IGNORED
STM32F103RC_bigtree      FAILED    00:00:07.306
STM32F103RC_bigtree_USB  IGNORED
STM32F4                  IGNORED
STM32F7                  IGNORED
ARMED                    IGNORED
STM32F103VE_longer       IGNORED
mks_robin                IGNORED
mks_robin_lite           IGNORED
mks_robin_mini           IGNORED
mks_robin_nano           IGNORED
jgaurora_a5s_a1          IGNORED
STM32F407VE_black        IGNORED
BIGTREE_SKR_PRO          IGNORED
BIGTREE_BTT002           IGNORED
teensy31                 IGNORED
teensy35                 IGNORED
STM32F103CB_malyan       IGNORED
esp32                    IGNORED
fysetc_f6_13             IGNORED
linux_native             IGNORED
SAMD51_grandcentral_m4   IGNORED
include_tree             IGNORED
==================== 1 failed, 0 succeeded in 00:00:07.306 ====================
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
Marlin-2.0.x-SKR-Mini-E3-V1.2
Process Palette: PIO Build
Command	
Output	stream	
Process Palette: PIO Clean
Command	
Output	stream	
Process Palette: PIO Upload
Command	
Output	stream	
Process Palette: PIO Upload (traceback)
Command	
Output	stream	
Process Palette: PIO Upload using Programmer
Command	
Output	stream	
Process Palette: PIO Test
Command	
Output	stream	
Process Palette: PIO Debug
Command	
Output	stream	
Process Palette: PIO Remote
Command	
Output	stream

meltonpieman, 2019-10-19:
Hi deborio
I have exactly the same problem. I have tried various other places for help and the best I got was a hex file from BTT, but I want to be able to compIle my own hex!

Gummibaer, 2019-10-19:
i had the Problem with Atom too. VSCode works for me but now i compile this under Linux in Command Line with pio only. (works great)

dborio, 2019-10-21:
I have tried in both Atom and VSCode.  I get the same error in both any help would be greatly appreciated as I am trying to compile the firmware to work with my needs.

dborio, 2019-10-21:
In file included from Marlin\src\HAL\HAL_STM32F1\HAL_SPI.cpp:35:0:
Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:30:10: fatal error: ../HAL/HAL.h: No such file or directory
 #include "../HAL/HAL.h"
          ^~~~~~~~~~~~~~
compilation terminated.
In file included from Marlin\src\HAL\HAL_STM32F1\Servo.cpp:26:0:
Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:30:10: fatal error: ../HAL/HAL.h: No such file or directory
 #include "../HAL/HAL.h"
          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_bigtree\src\src\HAL\HAL_STM32F1\Servo.cpp.o] Error 1
*** [.pio\build\STM32F103RC_bigtree\src\src\HAL\HAL_STM32F1\HAL_SPI.cpp.o] Error 1
In file included from Marlin\src\HAL\HAL_STM32F1\HAL.cpp:31:0:
Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:30:10: fatal error: ../HAL/HAL.h: No such file or directory
 #include "../HAL/HAL.h"
          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32F103RC_bigtree\src\src\HAL\HAL_STM32F1\HAL.cpp.o] 
Error 1
===================== [FAILED] Took 8.08 seconds =====================

This is my new error.

dborio, 2019-10-21:
I fixed the issues I just download the Default Marlin 2.0 Firmware and started with that.  Seems to be working just fine in VSCode now compiling anyway I will get it flashed to the printer after this print finishes.

rmaltais, 2019-11-02:
Common error in windows is the path. Use **simple path** with no space (use insteed underscore  
'' _ '' ) for the root like `d:\path_to_firmware\`. So check if you are install git software and python too. The compile work fine in linux and mac because no problem with path translate.