bigtreetech/BIGTREETECH-SKR-mini-E3 #35: USB Drivers Windows 10

glenmorangie007, 2019-10-09:
He all

I can't get the MAPLE USB Drivers to install under Windows 10 - I am running Windows 10 Pro 64 bit and every time U try to install the drivers for MAPLE it displays "The Third-Party inf does not contain digital signature information" so the drivers won't install - when launching Cura 4.3 the printer control panel hangs and I have to disconnect the printer, switch it off and start it up again without the USB cable inserted. Any advice please so I can print to the SKR MINI E3 V1.2 via Cura

Thanks

ManiRX100, 2019-10-09:
I also use Windows 10. But the OS recognized the device and showed that the drivers are already up to date. In your case you may try disabling the "DIGITAL SIGNATURE CHECKING" in the start up settings.

glenmorangie007, 2019-10-10:
I have done all that but whenever I connect the Ender 3 to be used in Cura - The Ender 3 locks up and has to be restarted without the USB Cable attached - the only way I can print is slicing the file first then uploading that to the TF card and printing from the TF card

ManiRX100, 2019-10-10:
I had this issue first when I connected through the USB. The buzzer started and the printer just froze. I had to reset the printer again. I am not sure exactly whether this is the solution or it is related to it. I feel that the board has some power grounding issues. Are you using the stock two pin power chord? I changed it to a three pin power chord which has an "earth" pin and it solved the issue for me.

ManiRX100, 2019-10-10:
> 
> 
> I have done all that but whenever I connect the Ender 3 to be used in Cura - The Ender 3 locks up and has to be restarted without the USB Cable attached - the only way I can print is slicing the file first then uploading that to the TF card and printing from the TF card

@glenmorangie007 I got the issue again when I tried using the USB connection in Cura. My buzzer starts beeping and printer freezes. But the printer works in USB when I use Pronterface. I tried this you tube video "How to fix Cura or Slicer Not CONNECTING to 3D Printer Creality Ender 3?", but in vain.

ManiRX100, 2019-10-10:
https://github.com/Ultimaker/Cura/issues/4196.. Some solutions are here.

glenmorangie007, 2019-10-10:
@ManiRX100 another email has gone off to BigTree about this issue - probably take a couple of days to resolve again !

ManiRX100, 2019-10-10:
@glenmorangie007 May be the issue is with the Cura compatibility with the latest Marlin. Meanwhile let us try the solutions in the older thread I mentioned. Are you aware of compiling this BTT V1.2 board in Arduino IDE? I am not getting any responses to that thread.

glenmorangie007, 2019-10-10:
@ManiRX100 Everything I have seen shows the only way to compile this is with VS not with Arduino IDE

ManiRX100, 2019-10-11:
A new firmware was released today. But it throws error. Has anyone tried it? Does it solve the issue with the Cura ?

glenmorangie007, 2019-10-11:
@ManiRX100 Haven't tried it yet - my firmware is working at the moment apart from the USB issue in Cura !

ManiRX100, 2019-10-11:
@glenmorangie007 Please let me know once they get back. Thanks.

glenmorangie007, 2019-10-12:
@ManiRX100 Have you tried the new version of the firmware ? Does it now work with Cura ??

ManiRX100, 2019-10-12:
@glenmorangie007 I am not able to compile the new version. It throws an error.

maddingamer, 2019-10-28:
looks similar to #40

Herb01, 2020-09-11:
I just installed the SKR MINI E3 V2 and now trying to install the usb driver with Win10. I can't install the Maple driver. Every time it complains about unknown USB device. Windows complains about unsigned driver so i disable this check. No luck. I'm running out of options. What to do

sidecontrol, 2020-10-05:
I was able to install the drivers by rebooting into unsigned mode,  hold shift while clicking reboot and go-to troubleshooting in the recovery menu, you can boot windows permitting unsigned driver installation. However the device still shows up as an unknown device but the drivers are installed.