bigtreetech/BIGTREETECH-SKR-mini-E3 #36: Compiling the BTT Marlin in Arduino IDE

ManiRX100, 2019-10-09:
Hi 
I am trying to compile the BTT E3 Marlin in the Arduino IDE. I added the generic STM32F103R series board and selected variant "STM32F103RC (48K RAM, 256K flash), Upload method: Serial, CPU Speed : 72MHz. The selection was based on the manual by BTT. But I get the following errors.
1. #error "TMC2208 Software Serial is supported only on AVR, LPC1768, STM32F1 and STM32F4 platforms."
2. fatal error: IWatchdog.h: No such file or directory
3. #error "Oops! Select an STM32F1 board in 'Tools > Board.'"
Help needed in this regard.

ManiRX100, 2019-10-10:
> Hi
> I am trying to compile the BTT E3 Marlin in the Arduino IDE. I added the generic STM32F103R series board and selected variant "STM32F103RC (48K RAM, 256K flash), Upload method: Serial, CPU Speed [...]

Hi all
Any ideas on this issue?

Neildoe, 2019-10-14:
From my understanding you cant work on 32b boards in arduino. You need Atom. Teaching Tech on youtube has great videos to get you started. his video is linked on the main listing for the board

closed 2019-10-15 by ManiRX100