bigtreetech/BIGTREETECH-SKR-mini-E3 #38: missing board mini_e3

minzukas, 2019-10-10:
Unable to build with platformio. 
Error promted

> Executing task in folder Marlin-2.0.x-SKR-Mini-E3-v1.2: platformio run <

Error: Unknown environment names 'BOARD_BIGTREE_SKR_MINI'. Valid names are 'melzi, STM32F103R_fysetc, teensy31, rambo, at90usb1286_cdc, megaatmega1280, mks_robin, BIGTREE_SKR_PRO, mks_robin_nano, STM32F103V_longer, STM32F103R, STM32F407VE_black, mks_robin_mini, at90usb1286_dfu, sanguino_atmega1284p, STM32F4, STM32F7, ARMED, BIGTREE_BTT002, STM32F103R_bigtree, megaatmega2560, teensy35, DUE, include_tree, jgaurora_a5s_a1, STM32F103R_bigtree_USB, DUE_USB, DUE_debug, SAMD51_grandcentral_m4, esp32, sanguino_atmega644p, malyanm200, LPC1768, LPC1769, linux_native, melzi_optiboot, mks_robin_lite, fysetc_f6_13'
The terminal process terminated with exit code: 1

Marlin-2.0.x-SKR-Mini-E3-v1.2/buildroot/share/PlatformIO does not contain board BOARD_BIGTREE_SKR_MINI

everything is done according to provided instructions

Stew-rt, 2019-10-10:
Ignore the instructions...
The zip already has the configuration changes necessary to build.

Only requirement *might* be to set it up to ignore the Adafruit LED library....

You may way to browse and edit Marlin/configuration.h and Marlin/configuration_adv.h though...

brew99, 2019-10-10:
change to STM32F103R_bigtree

ManiRX100, 2019-10-11:
@bigtreetech Yes... every day people are using either the wrong firmware or using the old instructions related to old firmware. Please change it.

minzukas, 2019-10-11:
I can see that new FW and guide was uploaded. Much appreciated. 
Still can't compile.

Posting errors:

*** [.pio/build/STM32F103RC_bigtree/libb5d/Wire] /home/mindaugas/Downloads/BIGTREETECH-SKR-mini-E3-master/firmware/V1.2/Marlin-2.0.x-SKR-Mini-E3-V1.2/.pio/build/STM32F103RC_bigtree/libb5d/Wire: No such file or directory
OSError: [Errno 2] No such file or directory: '/home/mindaugas/Downloads/BIGTREETECH-SKR-mini-E3-master/firmware/V1.2/Marlin-2.0.x-SKR-Mini-E3-V1.2/.pio/build/.sconsign.dblite':
  File "/home/mindaugas/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1346:
    _exec_main(parser, values)
  File "/home/mindaugas/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1309:
    _main(parser)
  File "/home/mindaugas/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1091:
    nodes = _build_targets(fs, options, targets, target_top)
  File "/home/mindaugas/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1283:
    jobs.run(postfunc = jobs_postfunc)
  File "/home/mindaugas/.platformio/packages/tool-scons/script/../engine/SCons/Job.py", line 113:
    postfunc()
  File "/home/mindaugas/.platformio/packages/tool-scons/script/../engine/SCons/Script/Main.py", line 1280:
    SCons.SConsign.write()
  File "/home/mindaugas/.platformio/packages/tool-scons/script/../engine/SCons/SConsign.py", line 109:
    syncmethod()
  File "/home/mindaugas/.platformio/packages/tool-scons/script/../engine/SCons/dblite.py", line 129:
    self._os_unlink(self._file_name)
Exception OSError: (2, 'No such file or directory', '/home/mindaugas/Downloads/BIGTREETECH-SKR-mini-E3-master/firmware/V1.2/Marlin-2.0.x-SKR-Mini-E3-V1.2/.pio/build/.sconsign.dblite') in <bound method dblite.__del__ of <SCons.dblite.dblite object at 0x7f0500509090>> ignored
============================================================= [FAILED] Took 35.54 seconds =============================================================

Environment              Status    Duration
-----------------------  --------  ------------
megaatmega2560           IGNORED
megaatmega1280           IGNORED
at90usb1286_cdc          IGNORED
at90usb1286_dfu          IGNORED
DUE                      IGNORED
DUE_USB                  IGNORED
DUE_debug                IGNORED
LPC1768                  IGNORED
LPC1769                  IGNORED
melzi                    IGNORED
melzi_optiboot           IGNORED
rambo                    IGNORED
sanguino_atmega644p      IGNORED
sanguino_atmega1284p     IGNORED
STM32F103RE              IGNORED
STM32F103RC_fysetc       IGNORED
STM32F103RC_bigtree      FAILED    00:00:35.536
STM32F103RC_bigtree_USB  IGNORED
STM32F4                  IGNORED
STM32F7                  IGNORED
ARMED                    IGNORED
STM32F103VE_longer       IGNORED
mks_robin                IGNORED
mks_robin_lite           IGNORED
mks_robin_mini           IGNORED
mks_robin_nano           IGNORED
jgaurora_a5s_a1          IGNORED
STM32F407VE_black        IGNORED
BIGTREE_SKR_PRO          IGNORED
BIGTREE_BTT002           IGNORED
teensy31                 IGNORED
teensy35                 IGNORED
STM32F103CB_malyan       IGNORED
esp32                    IGNORED
fysetc_f6_13             IGNORED
linux_native             IGNORED
SAMD51_grandcentral_m4   IGNORED
include_tree             IGNORED
======================================================== 1 failed, 0 succeeded in 00:00:35.536 ========================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

minzukas, 2019-10-11:
problems with
 #Adafruit NeoPixel@1.2.5
  https://github.com/bigtreetech/Adafruit_NeoPixel

managed to compile after commenting out.
#define NEOPIXEL_LED

Still an issue!