bigtreetech/BIGTREETECH-SKR-mini-E3 #39: Cannot compile with VS Code and PlatformIO

fluppie, 2019-10-12:
```
> Executing task in folder Marlin-2.0.x-SKR-Mini-E3-V1.2: C:\Users\Philippe\.platformio\penv\Scripts\platformio.exe run --environment STM32F103RC_bigtree_USB <

Processing STM32F103RC_bigtree_USB (platform: ststm32; framework: arduino; board: genericSTM32F103RC)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 5.6.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0), tool-stm32duino 1.0.1
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing U8glib-HAL
Using cache: C:\Users\Philippe\.platformio\.cache\a7\b7a9b473d1c239d9843ca943887cd0a7
Unpacking...
FileNotFoundError: [Errno 2] No such file or directory: 'c:\\Users\\Philippe\\OneDrive\\3D Printing\\BigTreeTech\\BIGTREETECH-SKR-mini-E3-master\\firmware\\V1.2\\Marlin-2.0.x-SKR-Mini-E3-V1.2\\.pio\\libdeps\\STM32F103RC_bigtree_USB\\_tmp_installing-gi1a573m-package\\U8glib-HAL-bugfix\\src\\clib\\HAL_example_files\\LCD_I2C_routines.c.example':
  File "C:\Users\Philippe\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Philippe\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Philippe\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Philippe\.platformio\platforms\ststm32\builder\main.py", line 91:
    target_elf = env.BuildProgram()
  File "C:\Users\Philippe\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Philippe\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 122:
    _build_project_deps(env)
  File "C:\Users\Philippe\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 47:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Philippe\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Philippe\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1043:
    project.install_dependencies()
  File "C:\Users\Philippe\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 878:
    lm.install(uri)
  File "C:\Users\Philippe\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 359:
    force=force)
  File "C:\Users\Philippe\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 701:
    track=True)
  File "C:\Users\Philippe\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 483:
    self.unpack(dlpath, tmp_dir)
  File "C:\Users\Philippe\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 218:
    return fu.unpack(dest_dir, with_progress=False)
  File "C:\Users\Philippe\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 118:
    self._unpacker.extract_item(item, dest_dir)
  File "C:\Users\Philippe\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 38:
    self._afo.extract(item, dest_dir)
  File "C:\Users\Philippe\AppData\Local\Programs\Python\Python37\lib\zipfile.py", line 1599:
    return self._extract_member(member, path, pwd)
  File "C:\Users\Philippe\AppData\Local\Programs\Python\Python37\lib\zipfile.py", line 1670:
    open(targetpath, "wb") as target:
=============================================================================================== [FAILED] Took 1.30 seconds ===============================================================================================

Anyone an idea why?
```

fluppie, 2019-10-13:
OK, this one compiles and works with a BL Touchv3.1 connected to the Z-endstop header.

https://github.com/fluppie/WORKING_Marlin-2.0.x-SKR-Mini-E3-V1.2-BLTouchv3.1

kubik256, 2019-10-13:
At first try to compile FW outside OneDrive, it can cause many strange problems. Also your full path is about 270 chars long, try to make it smaller - Something like C:\Marlin\.pio\ - should be good ;)
At last if this won't work, try to clone new Marlin at https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x
....modify configs, compile and you're done ;)

fluppie, 2019-10-13:
Yes you were right. Inside OneDrive gives big issues. I switched to Documents\Github\... and problems were resolved. I also run VS Code as admin to make sure I have enough rights during compile.
Only having a EEPROM issue at the moment with this fork: https://github.com/fluppie/BIGTREETECH-SKR-mini-E3
When saving the Z-offset to EEPROM the firmware corrupts and doesn't boot anymore after powering down the printer. So you need to reflash the MCU.

kubik256, 2019-10-13:
EEPROM doesn't work because there is no physical eeprom in the STM chip.
There have to be two ways:

**1) EEPROM on SDCARD**
#define VIRTUAL_SD_EEPROM 8
On SD card put empty file called "eeprom.dat" and it should work.

2) Flash-based EEPROM emulation
#define EEPROM_SETTINGS
#define FLASH_EEPROM_EMULATION  // Use Flash-based EEPROM emulation

More info here: https://github.com/MarlinFirmware/Marlin/issues/14684
...and in many other threads on Marlin with STM chips :)

I don't have tested yet any EEPROM setting, I will test both ways tomorrow.
Good luck ;)

kubik256, 2019-10-13:
Sorry, forget the 1st one, it was experimantal way and needs more tweaking than just setting in Configuration.h, FLASH_EEPROM_EMULATION should work.

I can't find the proper way to init EEPROM on SD card, but there could be some.

fluppie, 2019-10-13:
Hmmm it's already enabled by default. Maybe the EEPROM offset is wrong or the bootloader is (semi)corrupted?

fluppie, 2019-10-14:
The above changes seem to work for me. After enabling BL Touch the build size is 226 KB which is bigger than the 224 KB stated. Worth noting that my CPU is a 256KB one, a RCT6.

Gummibaer, 2019-10-14:
You can try 
activate #define SLIM_LCD_MENUS

Gummibaer, 2019-10-15:
BLTOUCH only useful with AUTO_BED_LEVELING = EEPROM MAX 224Kb = 229376 bytes

- STOCK Firmware from Github  = 215764 bytes
- STOCK Configuration Build  (git clone and compile no Changing)  = 215156 bytes

--- platformio.ini ---
- add lib_ignore = Adafruit NeoPixel  (no diffrends)
- add Custom Build Parameter -328 bytes

( -fno-tree-scev-cprop -fno-split-wide-types -finline-limit=3 -ffast-math )

--- Configuration.h ---
- #define BLTOUCH +5.472 bytes 
- #define AUTO_BED_LEVELING_BILINEAR +7.736 bytes 
- #define AUTO_BED_LEVELING_LINEAR +6.352 bytes 
- We NEED #define Z_SAFE_HOMING +368 bytes 

- Test Disable //#define SDSUPPORT Build FAILED
- #define SLIM_LCD_MENUS -6.512 bytes 
- #define SPEAKER +816 bytes 
- Test Disable //#define CR10_STOCKDISPLAY (if you not need [Octopi])  -38.520 bytes 

--- Configuration_adv.h ---
- Disable //#define TMC_DEBUG -10.384 bytes 
- Disable //#define EXTENDED_CAPABILITIES_REPORT  DISABLE M115 -‭664‬ bytes 

Notes
TMC_DEBUG is no longer needed in Marlin 2.0.x, but enabling it produces an extended report

ADD
#define VIRTUAL_SD_EEPROM 8 

DATA:    [==        ]  21.9% (used 10752 bytes from 49152 bytes)
PROGRAM: [========  ]  80.2% (used 210276 bytes from 262144 bytes)

Fit on Flash :D  Good Luck

JuhlTurner, 2019-10-20:
> Looks like this could be the solution:
> 
> > If your using my STM32F103RE_bigtree_USB build;
> > you should change the file pins_BIGTREE_SKR_E3_DIP.h
> > change line 38 that reads
> > #define EEPROM_START_ADDRESS uint32(0x8000000 + 256 * 1024 - 2 * EEPROM_PAGE_SIZE)
> > to:
> > //#define EEPROM_START_ADDRESS uint32(0x8000000 + 256 * 1024 - 2 * EEPROM_PAGE_SIZE) //use for F103RC
> > #define EEPROM_START_ADDRESS uint32(0x8000000 + 256 * 2048 - 2 * EEPROM_PAGE_SIZE) //use for F103RE
> > Then eeprom emulation in flash will work, and it will sit at 0x0807D000 rather than 0x0803EA00.
> > You only have to do this if our build is larger than 256KB-28KB-4KB = 224KB
> > If using _USB you probably need it.
> 
> Found in: [MarlinFirmware/Marlin#15254](https://github.com/MarlinFirmware/Marlin/issues/15254)

Changing EEPROM_START_ADDRESS worked for me.

Grayzone233, 2019-11-03:
> OK, this one compiles and works with a BL Touchv3.1 connected to the Z-endstop header.
> 
> https://github.com/fluppie/WORKING_Marlin-2.0.x-SKR-Mini-E3-V1.2-BLTouchv3.1

Get error 404 on your link.

bojanpotocnik, 2019-11-03:
@Grayzone233 looks like he changed the repository, based on the commits BLTouch v3.1 is activated in this repository: https://github.com/fluppie/BIGTREETECH-SKR-mini-E3/commits/master

Grayzone233, 2019-11-03:
@bojanpotocnik  was looking for the compiled bin file, but I can get there from what he documented in the repository. Thanks! The only bins were the original which is good to save as a backup in case the compiled does not work.

bojanpotocnik, 2019-11-03:
Good to hear that. Additionally, [my fork](https://github.com/bojanpotocnik/Marlin/tree/bugfix-2.0.x_SKR-mini-E3-V1.2_BLTouch) of the Marlin also contains binary files for BLTouch configuration.

> in case the compiled does not work.

If you happen to run into problems with booting the board using firmware.bin compiled on your machine, check MarlinFirmware/Marlin issue 15767 where we discussed the solutions.

fluppie, 2019-11-03:
> 
> 
> @bojanpotocnik was looking for the compiled bin file, but I can get there from what he documented in the repository. Thanks! The only bins were the original which is good to save as a backup in case the compiled does not work.

There is a bin file: https://github.com/fluppie/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2
Only check/change your E-steps after flashing the FW.

davidcgu, 2019-11-09:
After changing the start write point on the Eeprom emulation it not crashes anymore after using Eeprom function however seems not to be working, I mean if a value is changed and after Eeprom save, after reboot the value is gone, is it working for you guys?

Thanks in advance

fartplume001, 2019-11-18:
I've had this problem, but my corruption happned mid print. It would just stop printing. I power cycled the board and no boot. After I reloaded the firmware I clicked restore failsafe defaults. We'll see if that works.

fartplume001, 2019-11-18:
Did it again. Trying running the print through mattercontrol.

fartplume001, 2019-11-18:
Works fine through mattercontrol. I suspect running the file through the sd card is over writing the firmware space.

dburr, 2019-11-18:
I haven't been able to get flash EEPROM emulation to work at all. Have tried all of the fixes (changing start address, etc.) The best that I can get is that it no longer crashes, but I am unable to save any values to the (emulated) EEPROM. I gave up and switched to SD_EEPROM_EMULATION. Means I have to keep an SD card in the printer 24/7, but I normally do that anyway, and I normally don't print from SD so I never remove/change the SD card anyway.