bigtreetech/BIGTREETECH-SKR-mini-E3 #40: Can't connect to slicer.

PLAyerNos, 2019-10-12:
Hi All,

My new SKR mini E3 V1.2 works well except connecting to Cura.
I also tried different slicer with the same result.
There is an alarm sound coming from the board I guess and 'TMS connection error' pops on the printer display sometimes.
No problem with connecting to Pronterface.
I used communication settings from the guide:
#define SERIAL_PORT 2

#define SERIAL_PORT_2 -1

#define BAUDRATE 115200
Please help!

Regards

mrBarbecue, 2019-10-12:
Removing the Jumper next to the USB port fixed the TMC CONNECTION ERROR for me.

PLAyerNos, 2019-10-13:
No luck for me with this jumper. Thanks anyway.

kubik256, 2019-10-13:
The same here, Pronterface - ok, Cura 4.2.1 - no connection
(Ender3 with Mini E3 v 1.0)

PLAyerNos, 2019-10-13:
This message is from COM port driver:
Device USB\VID_1EAF&PID_0004\00000030 requires further installation.
I think it needs to be updated.

ManiRX100, 2019-10-19:
Hi
Anyone came across in solving this issue apart from taping the power pin in the USB cable?

PLAyerNos, 2019-10-19:
Hi,
Yes. Slice it in Cura, save it to hard drive, load it to Pronterface and print it from there 👌

maddingamer, 2019-10-20:
Same Issue for me with mini E3 v1.2 and cura 4.2.1 and 4.3.0, when I connect via USB the printer freezes with the wired noise, nothing in the display for me tho, as the display freezes too.. Any suggestions?

chaoticchoas, 2019-10-20:
So I was talking to BTT support on ali express and they mentioned to that the issue could be the display. So I started the printer without the displayed connected and it all works fine except now you don't have a display. Hope they find a fix for it.

maddingamer, 2019-10-20:
> So I was talking to BTT support on ali express and they mentioned to that the issue could be the display. So I started the printer without the displayed connected and it all works fine except now you don't have a display. Hope they find a fix for it.

Did you disconnect the cable at the back of the display or at the mainboard? Because disconnecting at the display itself doesn't work for me..

PLAyerNos, 2019-10-20:
Doesn't work for me either.

chaoticchoas, 2019-10-20:
Hm strange. I just unplugged from the motherboard while the printer was off. It seems to work fine about 80-90% of the time with the other 20-10% it seems to just freeze the printer. I get 

> Serial error: could not open port 'COM9': WindowsError(5, 'Access is denied.') 

when I try to connect to the printer. 

 Do note I have the dip version of this board and not the mini e3 v1 or v2. Here is 2 videos of mine that show what happens when I plug a usb with cura. [You will have to unmute if you want to here the alarm](https://imgur.com/a/UU6Q9rb?spm=a2g0s.imconversation.0.0.2eb03e5fQDHbSt)

ManiRX100, 2019-10-21:
@chaoticchoas The same issue I have.. The printer freezes with the buzzer sound. But all the times for me. It didnt work even once. This issue is with Mini V1.2. It appears that there is some serial port conflict. But it works in Pronterface and simplify 3D. How to check in which COM port the printer is connected?

chaoticchoas, 2019-10-21:
Same. It work with everything else but cura. I have tried the stock firmware and changing the serial ports but so far it has been the same.

mhensen, 2019-10-21:
Having the same problem but then from Octopi .
With or without the 5V pin removed!
But storing in EEPROM seems to trigger the fault..  (not sure though!)

ndemar, 2019-10-27:
I just had this issue and go bad. Started heating up the system to PLA temperatures plugged in my USB. Stepped away for a minute to fill my coffee. I came back to a smoking hot-end and the motherboard had froze.  After a few checks the system locks up when plugging in the USB.

AdamV36, 2019-11-01:
Hello, did you find any solution for freezing after connecting to PC by USB? 
My board was communicating by USB correctly in pronterface, then I sent M500 to save to EEPROM. Then the problem occured, from SD card it prints good, but when I connect it to USB whole printer freezes.

ManiRX100, 2019-11-02:
@AdamV36 I think the problem still exists. I think the old version of the BTT firmware, it was mentioned to comment out the "define EEPROM", but we will not be able to save the settings via USB. But I faced the same problem even with SD card. When I saved the Z Probe Offset setting in the printer display and pressed print from SD, the printer froze. Should we try disabling the EEPROM?

boschie, 2020-01-07:
Any solution yet about freezes with usb-connection? I had it connected on a linux Debian machine running Cura 4.4.0 (Cura was not controlling the printer) and it frooze while heating up, so it kept on heating.

ManiRX100, 2020-01-08:
@boschie I tried few days back. The problem still exists even when CURA was just minimized and not even connected to the printer. This does not happen with Pronterface, Simplify 3d or repetier host. I wonder what is the issue.

Waxonwaxoffmyson, 2020-04-26:
Same thing just happened to me. I have a raspberry pi connected with the 5v blocked. Printer froze with the heated bed stuck on. Print was melted and printer was just frozen. When I power cycled it. It displayed the thermal runaway warning. Octoprint had a communication error displayed.

Were you able to solve this?

ndemar, 2020-04-27:
I had not luck resolving this. My whole heater, fan, and thermister wiring caught fire.  Luckily the $34.00 USD  warranty covered it and replaced my Ender3.  

> Same thing just happened to me. I have a raspberry pi connected with the 5v blocked. Printer froze with the heated bed stuck on. Print was melted and printer was just frozen. When I power cycled it. [...]

fastfourier666, 2020-06-03:
I was facing this problem using the onboard USB (Mini E3 V1.2) and Repetier host. The printer would respond for 30-60 seconds before freezing and the alarm sounding.

I couldn't find a workable solution using the built-in USB so I used Serial port 2 instead - it's exposed as a 5-pin header marked "TFT" next to the processor. I'm not using a graphical TFT, just the standard Ender 3 display. To connect it to the computer, I'm using a CP2102 USB-serial. Only TX, RX and ground hooked up.

#define SERIAL_PORT -1
#define SERIAL_PORT_2 2
#define BAUDRATE 115200

Double-check that your USB-serial is using 3.3V. The STM32F103RC datasheet says these pins are NOT 5V tolerant.

I read in another thread that the USB issues are due to a conflict between the SD card and the USB interface. No idea if it's true, but this is working for me so far (and it means I don't have to have the USB cable sticking out from the electronics box at the front)

fastfourier666, 2020-06-04:
Additional: I ended up connecting it to a Pi with Octoprint using the instructions here:

[https://community.octoprint.org/t/simple-3-wire-serial-connection-from-pi3-to-skr-v1-3-board/10166](https://community.octoprint.org/t/simple-3-wire-serial-connection-from-pi3-to-skr-v1-3-board/10166)

using TX/RX/GND straight to the pi GPIO header. No USB serial adapter necessary!