bigtreetech/BIGTREETECH-SKR-mini-E3 #41: SKR E3 Mini v1.2 and Octoprint

davehardy20, 2019-10-13:
Whenever I plugin the  USB cable to the board from my OctoPi there seems to be connection issues, sometimes its fine , but mostly  1 of 2 things happens.
1. the E3 Mini board crashes
2. the board seems fine and can print from OctoPi, however the display just goes haywire.
This OctoPi  is / was perfectly fine with the Creality 1.1.5 board I used before the E3 Mini v1.2
[image]

I've had a look on the OctoPrint forums but there doesn't seem to be anyone else reporting this issue.
Is this a known problem or a new issue?

lococnc, 2019-10-14:
Mine works fine. I do have the 5v taped off so it is not powering the SKR board.

davehardy20, 2019-10-14:
@lococnc your suggestion seems to have done the trick, I taped off the 5v connector in the USB A plug and things look good now. Thanks
I would guess that the newer 32bit chip and newer electronics are more susceptible to the 5v feed from the PI, whereas the Creality 8bit electronics may not be so sensitive. Just a guess though.

ManiRX100, 2019-10-14:
@davehardy20 I am also having the issue with Cura but the USB works with Pronterface and Simplify 3D. How do you tap off the 5V connector? Remove the 5V supply line from the USB ?

pusewicz, 2019-10-14:
@ManiRX100 Simply put a bit of insulation tape on the pin in the USB plug.

[image]

davehardy20, 2019-10-14:
Just as @pusewicz  says, but I used electrical insulation tape. Works just fine

closed 2019-10-15 by davehardy20

Thinkersbluff, 2020-02-12:
@pusewicz: Thanks for the tip on how to tape-off the 5VDC pin on a USB connector.
I just had to order a 2nd SKR mini e3 v1.2, after apparently having somehow fried the X-axis driver within 3 weeks of the upgrade.  
This board exhibits various strange behaviours if powered by the Pi or PC through the USB cable before turning on the printer's main power.   I even found all my motors super hot, when I came back two days after leaving the Pi on and the printer power off.  Apparently, the TMC Connection Error did not trigger in that case, either.
Hopefully, this work around protects my new board.

pusewicz, 2020-02-12:
@Thinkersbluff Usually, hot motors mean too much current sent via the driver. Try lowering the voltage in the Marlin config.