bigtreetech/BIGTREETECH-SKR-mini-E3 #42: Updating Marlin to latest bugfix-2.0 doesn't build using ENDSTOP_INTERRUPTS_FEATURE

pandabear41, 2019-10-16:
In the mini E3 v1.2 Configuration.h ENDSTOP_INTERRUPTS_FEATURE is enabled by default and builds correctly. When updating to the latest version of Marlin bugfix-2.0 I get an error during build of 
`Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE.`

Why doesn't the version of Marlin in this repo complain about having software serial and interrupts?
Are the endstop interrupt feature that useful or can I build with them disabled?

chaoticchoas, 2019-10-17:
Yea I was having issue trying to enable it as well. I would just not enable it but enable BLTOUCH_FORCE_SW_MODE and just see if that causes your bl touch to not probe correctly or for the probe to fail. Make sure to be ready to cut the power to the printer if it fails to home and crashes into your bed. Maybe home a couple of times but trigger your bltouch early and above the bed so that you can turn of the printer if needed. You can then use g29 a couple of time after that to make sure that your bltouch can probe consistently with out fail.

GhostlyCrowd, 2019-10-22:
They edited the Sanitycheck.h to allot them to be enabled if the target was STM32F1 cpu

GhostlyCrowd, 2019-10-22:
#if HAS_TMC220x && !defined(TARGET_LPC1768) && !defined(TARGET_STM32F1) && ENABLED(ENDSTOP_INTERRUPTS_FEATURE) && !( \
       defined(X_HARDWARE_SERIAL ) || defined(X2_HARDWARE_SERIAL) \
    || defined(Y_HARDWARE_SERIAL ) || defined(Y2_HARDWARE_SERIAL) \
    || defined(Z_HARDWARE_SERIAL ) || defined(Z2_HARDWARE_SERIAL) \
    || defined(Z3_HARDWARE_SERIAL) || defined(E0_HARDWARE_SERIAL) \
    || defined(E1_HARDWARE_SERIAL) || defined(E2_HARDWARE_SERIAL) \
    || defined(E3_HARDWARE_SERIAL) || defined(E4_HARDWARE_SERIAL) \
    || defined(E5_HARDWARE_SERIAL) )
  #error "Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE."
#endif

pandabear41, 2019-10-22:
Yeah, I figured it out a few days ago. What I was mainly concerned about was why Marlin upstream didn't have those changes.

GhostlyCrowd, 2019-10-22:
I have no idea, maybe because of sensor-less probing? you cannot afford to have software serial issues.

matheusbrasil, 2019-11-01:
I've got the same error: 

> Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE.

Someone could solve this?

GhostlyCrowd, 2019-11-01:
> I've got the same error:
> 
> > Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE.
> 
> Someone could solve this?

I posted the fix for this above.

you have to edit the sanity check to match Bigtrees code, which i posted above.

matheusbrasil, 2019-11-01:
> > I've got the same error:
> > > Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE.
> > 
> > 
> > Someone could solve this?
> 
> I posted the fix for this above.
> 
> you have to edit the sanity check to match Bigtrees code, which i posted above.

Thanks! I did not understand that you had already published the fix. My mistake.

closed 2019-11-06 by pandabear41

syampolsky, 2019-11-30:
Just an note to anyone who encountered a bunch of build errors after pasting the solution code as is: don't forget to add a back-slash at the end of the first line. Apparently, the copy-paste from github removes it. go figure...