bigtreetech/BIGTREETECH-SKR-mini-E3 #43: Firmware not updating? Display not working?

Trevor159, 2019-10-16:
Maybe duplicate or related to #3 but this has more/different info

The stock display on my ender 3 only gets back light when both the usb and power supply are turned on. The power led on the board itself only turns on when both the usb and power supply have power. When the power supply is connected alone the fans spin and it seems I wired the board correct. I have tried to flash a new firmware on the board but have no way of knowing if it worked or not with the display not working and the board just seems to be defective but I do not know enough to be 100% sure.

Jimmy3d, 2019-10-17:
I am having this issue but I found out it works when I disconnect my BLtouch v3 from the servo port, do you have a bltouch installed?

Trevor159, 2019-10-17:
> I am having this issue but I found out it works when I disconnect my BLtouch v3 from the servo port, do you have a bltouch installed?

Yes, that was the issue. Now I need to figure out how to get bltouch support.

Gummibaer, 2019-10-17:
The Wiring Pinout is correct?
Which Board?
Example:
-Pinout on Board SKR Mini E3 V1.0  => S G V
-Pinout on Board SKR Mini E3 V1.2  => G V S (no changing needed)
G = Ground; S = Signal; V = Voltage

Trevor159, 2019-10-17:
Solved: the wiring diagram for the creality bltouch in the readme here https://github.com/gazcbm/Marlin-2.0.x-SKR-Mini-E3-v1.2 had the red and blue wires flipped. I had a creality v3.1 and the v1.2 board fwiw

closed 2019-10-17 by Trevor159