bigtreetech/BIGTREETECH-SKR-mini-E3 #47: Hybrid mode results in blank LCD after Initialize EEPROM

brew99, 2019-10-17:
Running OCT17th Vanilla Marlin 2.0 bugfix.   I have Linear advance enabled and EEPROM emulation, which works fine on multiple reboots even after selecting initialize EEPROM from LCD.   But, if I then enable HYBRID_THRESHOLD, it all appears to work fine until I select Initialize EEPROm from the LCD.   After that, on reboot, the LCD is blank.   Works OK after firmware load if I do "load settings", or "failsafe", but Intialize eeprom corrupts something and blank LCD on reboot

Is there a conflict with HYBRID_THRESHOLD and Eeprom emulation?   Why does it first work, but after inilatizing eeprom it blanks the LCD on reboot.

Gummibaer, 2019-10-17:
Which size have the firmware.bin file?

brew99, 2019-10-17:
not sure on actual file size, but during the compiling (at the end) it reported 87.6% 229684 bytes of 262144.   No compile errors

Gummibaer, 2019-10-17:
And this is your Problem.
Max Filsize 224kb the rest of Flash Space needed for EEPROM
229684 / 1024 = 224,3kb

Solution Disable //#define TMC_DEBUG -10.384 bytes
or use #define SLIM_LCD_MENUS -6.512 bytes

brew99, 2019-10-17:
Ok thanks for that info.   Didn't realize that 224kb was the absolute max as I didn't know how much the EEprom emulation space was reserved.

I wonder in Marlin developers could put a sanitycheck in the code when compiling, atleast to warn users that the max size has been reached.

closed 2019-10-18 by brew99