bigtreetech/BIGTREETECH-SKR-mini-E3 #48: Unresponsive Board

ecksa, 2019-10-18:
i just got a SKR mini e3 v1.2. It was working for a while... somewhat (had issues with sensorless homing endstop always on triggered state), but soon realized the board became completely unresponsive. The red led is on solid, but it will not take a new firmware update, (uploading firmware.bin to sd card) nor is it being recognized by my pc usb (it was earlier today). Nothing major happened, only firmware was uploaded to it. Is there a way to force firmware install, or factory default somehow?

HGDrenth, 2019-10-18:
I seem to also have this issue. Was printing perfectly fine this morning, got a E1 thermal runaway error causing me to restart a print and now it's dead with the symptoms described by the user above.

Gummibaer, 2019-10-19:
I had a similar problem that the board did not want flash a new firmware . 
At the beginning I generated a image of the stock SD card.
I then used the image and then did the new firmware.

You can test that.

Good Luck
[ORG_MicroSD_BigTreeTech_SKR_Mini_E3_v1.2.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/3746974/ORG_MicroSD_BigTreeTech_SKR_Mini_E3_v1.2.zip)

HGDrenth, 2019-10-20:
I did try the old firmware from @Gummibaer  but no result, led stays solid red and no others light up. Also screen stays blank (with backlight on)

ecksa, 2019-10-20:
I tried as well with no result. Everything was working at first, but i remembered i initialized eeprom from lcd, and after this is when my problems started. It also bricked 2 sd cards.

ecksa, 2019-10-20:
Got a replacement, and after setting up the firmware, and changing one setting at a time, i found it to brick after enabling eeprom, as described in several reddit/github posts. 

Grab the firmware.bin from here and put it on an empty card should bring the board back, but MAKE SURE EEPROM IS DISABLED!

HGDrenth, 2019-10-22:
I'm having no luck, next step for me is to try and flash a new bootloader on it through a STLINK v2.
Is anyone able to provide me with a bootloader.bin for a SKR mini E3 v1.2?

timryder, 2019-10-23:
> Got a replacement, and after setting up the firmware, and changing one setting at a time, i found it to brick after enabling eeprom, as described in several reddit/github posts.
>
> Grab the [...]

What do you mean by this? You mean in Marlin Configuration.h  The "#deifine EEPROM_SETTINGS" keep that commented out?

ecksa, 2019-10-23:
That's what originally bricked mine. I'm thinking it's a space thing, not sure how much flash memory the board has. If i disable tmcdebug and set menu's to none, it brings the file size down to 198k, and lets me enable eeprom. Can anyone confirm max files size of firmware.bin the printer can handle?

Gummibaer, 2019-10-23:
Max Filsize 224kb the rest of Flash Space needed for EEPROM

HGDrenth, 2019-10-23:
> That's what originally bricked mine. I'm thinking it's a space thing, not sure how much flash memory the board has. If i disable tmcdebug and set menu's to none, it brings the file size down to [...]

Can you post this firmware or img from sd?

ecksa, 2019-10-23:
[firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/3764070/firmware.zip)

timryder, 2019-10-24:
I recompiled my firmware commenting out several items that I didn't need on the LCD and I went through and searched for Program Space and commented out all optional items. the file size was 207kb and it flashes and STAYS flashed permenantly.

steverichab, 2019-12-03:
Well after many hours and tries with my BTT SKR E3 Mini V1.2 on WIndows 10 with VSCode and PIO 4.1

I found this entry on reddit( Thank You "Moringreis")  It really works and compiles ! 

https://www.reddit.com/r/ender3/comments/dojh3v/guide_for_those_upgrading_to_an_skr_e3_mini_v12/

crstrand, 2020-01-24:
@HGDrenth did you ever get this fixed?
Mine is behaving exactly the same way and it won’t even take the firmware image from the BTT repo.

HGDrenth, 2020-01-24:
@crstrand nope, unfortunatly not. I tried different firmwares from different sources, compiled my own, all to no avail. I also experimented with a 2$ ST-Link, couldn't get this to work, I suspect that the chip is empty and in a write protected mode or something.

crstrand, 2020-01-26:
@HGDrenth did you ever get a bootloader to try? I am embarking on the same voyage but would prefer not to reinvent the wheel if I don’t have to.
What was your final solution?

HGDrenth, 2020-01-26:
@crstrand I asked for a bootloader but didn't get one (source or bin). Reason was that it wasn't released as open source (yet). My 'solution' was to chuck it aside and install a better board with more memory.

crstrand, 2020-01-26:
Seeing as how I'm following in your footsteps...which board did you get? ;)-Cam

HGDrenth, 2020-01-26:
I had this one with 2209, intended to use it for another project: #Aliexpress € 26,87  12%OFF | BIGTREETECH SKR V1.3 3D Printer Board+TFT24 Touch Screen+TMC2209 TMC2208 UART TMC2130 For Ender 3/5 MKS GEN L 3D Printer Parts
https://a.aliexpress.com/_UsfZL

steverichab, 2020-01-27:
I just wanted to let those who are interested that I finally go the printer dialed in using the BigTreeTech SKR E3 Mini V1.2 (built in TMC2209's), BTT TFT35 touch screen, BLTouch and sensorless homing. The "root cause" of all of the struggles were the first **TWO** BTT SKR E3 Mini boards would not responde. Third one I received worked flawlessly and the Redit posting https://www.reddit.com/r/ender3/comments/dojh3v/guide_for_those_upgrading_to_an_skr_e3_mini_v12/  is comprehensive and very well written
Enjoy

speedygreen, 2020-01-27:
@ecksa thank you for the firmware.bin you posted 23 Oct 2019. I'm fairly new to some of this and haven't gotten into compiling my own firmware yet. My SKR mini E3 v1.2 board had been all hooked up and running fine when I decided to upgrade to the already-compiled firmware that BTT has here on github. For me, the board didn't seem to have any problem reading or writing to the SD card. I would watch the status light flash for a few seconds, but then it would stay on. Checking the SD card, firmware.bin would be gone and a newly written firmware.cur would be present. Plugging the board into my computer would yield nothing, no doop doodoo doop USB device inserted sound, and not even an unknown usb device notice. I tried several pre-compiled firmware files I found online, nothing worked until I found yours posted here. THANK YOU!

midmadn, 2020-05-24:
> Well after many hours and tries with my BTT SKR E3 Mini V1.2 on WIndows 10 with VSCode and PIO 4.1
> I found this entry on reddit( Thank You "Moringreis")  It really works and compiles !
> https://www.reddit.com/r/ender3/comments/dojh3v/guide_for_those_upgrading_to_an_skr_e3_mini_v12/

I had the same experience. I can't even count the days and hours I put into every different version of firmware that's out there and I had no luck until I stumbled upon Morningreiss reddit post in the links under one of Teaching TRech's video's. Now I've compiled the version of Marlin 2.0.5.3 at least a dozen times with never a fail. I actually just got sensorless homing working today. This stuff just doesn't need to be this hard.

mwlehmann, 2020-05-27:
> [firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/3764070/firmware.zip)

that worked finally