bigtreetech/BIGTREETECH-SKR-mini-E3 #49: Can't flash firmware - board won't boot

m4rl0ne, 2019-10-19:
I followed the instructions in this repo. Still, I can't get my customized firmware to upload. The board becomes unresponsive and nothing shows up on the screen, just the backlight. The only way to restore it is to flash the standard firmware here in this repo. Why can't I upload anything? The flash process is fine, the file is renamed afterwards. Maybe this has something to do with CR10_STOCKDISPLAY? I tried the latest Marlin 2.0 branch, tried this branch, tried various configurations, nothing works. Board is not faulty because stock firmware works.

Gummibaer, 2019-10-19:
Filesize from firmware.bin?

m4rl0ne, 2019-10-25:
Thanks to everyone for replying. I fixed my issue by simply updating the platformio plugin for atom. Now it works flawlessly.

closed 2019-10-25 by m4rl0ne

ManiRX100, 2019-10-28:
> 
> 
> This happens to me too. I had to Rebuild a new firmware.bin because my Ender 3 has an EZABL sensor for probing. I was able to do this and it took on first load off the SDCard. But somewhere while using it the machine stop responding and the beeper was stuck on. I rebooted it and there was no firmware loaded. Nothing displayed ever on the LCD. Renaming the FIRMWARE.CUR back to firmware.bin and restarting will fix it for a time. But then it all happens all over again.
> 
> Any thoughts?

My board was working for sometime without issues. Now I am experiencing the same as you mentioned. 
@m4rl0ne I use VS Code. The problem exists even with the updated version. Any idea?

timryder, 2019-10-28:
The problem it seems is you cannot have a firmware.bin larger than 224kb. Anything more than that and it doesnt fully load into the memory. It will load into like a Ram state until you power cycle it.  Make sure after you compile you check that the file size is <= 224kb. The chip uses the remained if it's flash as simulated EEPROM space.  I've successfully reflashed several times now observing the 224 rule. Seach the Configuration.h for anything that says "PROGMEM" and comment it out if you think you don't need it.

bojanpotocnik, 2019-10-29:
> My board was working for sometime without issues. Now I am experiencing the same as you mentioned.
> I use VS Code. The problem exists even with the updated version. Any idea?

@ManiRX100 I had exactly the same problem. No matter which FW I used (I took care of the size limit), I just could not produce the FW image which would boot.
Updated VS Code, Atom, Python, reinstalled PlatformIO Core and compiled command line, nothing.

Eventually I solved it by creating the virtual machine, as explained in [my comment](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/61#issuecomment-547307520) on #61. It takes you 15 minutes to install Debian/Ubuntu to VirtualBox/VMWare, create shared folder with your host machine, `pip3 install platformio`, `platformio run` in your mounted shared folder, and you will probably have the working FW minus all the headache.

ManiRX100, 2019-10-29:
@bojanpotocnik So your suggestion is to build the firmware in Linux platform and flash it in the E3 board?

bojanpotocnik, 2019-10-29:
Yes, searching trough the issues shows that few people solved this problem by restarting PC, updating PlatformIO, reinstalling the environment, etc.
As I am not willing to reinstall my development environment because of this FW, I installed bare Debian image in the virtual machine (VirtualBox or VMWare Player, both free for this usage) so that I had completely fresh environment. And it works. I was really frustrated why my _"[up-to-date Marlin branch](https://github.com/bojanpotocnik/Marlin/commits/bugfix-2.0.x_SKR-mini-E3-V1.2) with minimal changes to work on SKR Mini E3"_ does not boot, where is my config wrong... But yeah, this was the only cause.

reloxx13, 2019-11-03:
@bigtreetech pls reopen, issue is still appearing.

updating/reinstall does not work.

reloxx13, 2019-11-03:
here we go @ivankravets

we know this issue with failing firmware from windows pc´s from Tasmota firmware.
Its hapenning again in marlin since 20-30 days.

ivankravets, 2019-11-03:
Hi all,

How can I help here? Could you provide a simple PlatformIO project to reproduce this issue?

reloxx13, 2019-11-03:
> How can I help here? Could you provide a simple PlatformIO project to reproduce this issue?

nope, we cannot cause this is appearing radomly.

My workaround was to force-reinstall platformio.
But no clue why it was neccessary from one day to another.

check this for a howto reinstall:
https://github.com/MarlinFirmware/Marlin/issues/15767#issuecomment-549156764

ivankravets, 2019-11-03:
See https://github.com/MarlinFirmware/Marlin/issues/15767#issuecomment-549168089

cptsoviet, 2019-11-20:
@timryder 
> The problem it seems is you cannot have a firmware.bin larger than 224kb. Anything more than that and it doesnt fully load into the memory. It will load into like a Ram state until you power cycle it. Make sure after you compile you check that the file size is <= 224kb. The chip uses the remainer of it's flash memory as a simulated EEPROM space. I've successfully reflashed several times now observing the 224 rule. Seach the Configuration.h for anything that says "PROGMEM" and comment it out if you think you don't need it.

Thank you so so much.

Was experiencing same issues. Another symptom is attempting to change Z-offset using printer LCD/Wheel, when you click on "Save settings" marlin will completely hang and requires a powercycle. I guess it has nowhere to save. If i turned off my printer after print finished when I power it back on - blue screen. Checked my firmware, it was right at 225k. Removed boot logo without changing anything else and it went down to 223.5 ish. No more blue screen. No more hangs when saving Z-Offset. Thank you!

damaltor, 2020-11-27:
just to let you know, i had that problem again today with the most recent platform.io on vscode, compiling marlin (bugfix and non-bugfix) would make the printer not boot up. after hours of pulling out my hair i compiled the same source within linux, and it worked immediately.