bigtreetech/BIGTREETECH-SKR-mini-E3 #50: ender crashes when trying sdcard menu

kramer04, 2019-10-19:
printer crashes when i want to print something from the sdcard

mrBarbecue, 2019-10-19:
There's a part in the manuel, that says you must uncomment //#define SPEAKER in config.h or else it freezes, if you press the button. Could it be this?

kramer04, 2019-10-21:
if i uncomment it works better
but have to turn off/on the printer...
i read pdf in firmware folder but don't see what you say
where did you find it ?
thanks

mrBarbecue, 2019-10-21:
[image]

Second last picture.

timryder, 2019-10-23:
@mrBarbecue The manual says you must leave it COMMENTED or it will Freeze.  Do not uncomment this property.

Gummibaer, 2019-10-23:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2
**comment out** //#define SPEAKER, **if you don't comment out this, it will freeze when you pressed button**

mrBarbecue, 2019-10-23:
@timryder You're right. My mistake. Meant the same as you, wrote it wrong.