bigtreetech/BIGTREETECH-SKR-mini-E3 #51: function call is not allowed in constant expression

bASHer91, 2019-10-20:
How to i solve this? Followed all instructions step by step.

Thanks

[image]

HGDrenth, 2019-10-20:
I think PlatformIO had an update, try reinstalling. 
I also had to update to python 3.8 -> dont forget to tick the PATH option during installation.

bASHer91, 2019-10-20:
Ive only just installed it all, still the same :(

how do i find out what python im running?

closed 2019-10-31 by bASHer91

nickjgriff, 2020-02-04:
I have this problem with a earlier version of marlin it's only about 2 weeks old and working fine but as soon as I upload it I get the same 41 errors and the firmware is unchanged from the last time I built it ?

nicwilson58, 2020-11-22:
thank you, installing Python 3.9 in windows,  then installing,  Python for VSCode and Python Extensions for VSCode fixed this issue for me too.

150612360, 2023-02-11:
[python-3.8.0-embed-amd64.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/10713491/python-3.8.0-embed-amd64.zip)
did not work please HELP

HTMLdu, 2023-02-11:
> [python-3.8.0-embed-amd64.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/10713491/python-3.8.0-embed-amd64.zip) did not work please HELP

Download latest version which is 3.9