bigtreetech/BIGTREETECH-SKR-mini-E3 #54: Heated bed doesn't work on CR-10 (stock).

oreo-ux, 2019-10-21:
Since installing the SKR mini E3 in my CR-10 the heated bed won't heat up. The Temps. are also jumping around 20-21 degress Celsius.
Only the bed is affected (hotend works flawlessly).
I've already checked the connections and didn't find any flaws.
Maybe it's an error in the Marlin code.
Thank you in advance for your help!

kubik256, 2019-10-22:
I've setup and calibrate BED PID with `M303E-1C10S70` and this config:
```
#define PIDTEMPBED

//#define BED_LIMIT_SWITCHING

/**
 * Max Bed Power
 * Applies to all forms of bed control (PID, bang-bang, and bang-bang with hysteresis).
 * When set to any value below 255, enables a form of PWM to the bed that acts like a divider
 * so don't use it unless you are OK with PWM on your bed. (See the comment on enabling PIDTEMPBED)
 */
#define MAX_BED_POWER 255 // limits duty cycle to bed; 255=full current

#if ENABLED(PIDTEMPBED)
  //#define MIN_BED_POWER 0
  //#define PID_BED_DEBUG // Sends debug data to the serial port.

  // CREALITY ENDER 3
  #define DEFAULT_bedKp 73.95
  #define DEFAULT_bedKi 13.57
  #define DEFAULT_bedKd 268.70
```

It works fine on Ender 3, but I don't  know if CR-10 has larger / more power hungry bed. If so, external mosfet could be useful, maybe mosfet on mini E3 can't handle that amount of current.

Also I have aluminium heatsinks on all mosfets ;)

oreo-ux, 2019-10-22:
I changed the code but the problem wasn't solved. The bed was working flawlessly until I've changed the mainboard.

oreo-ux, 2019-10-22:
I've checked the corresponding connections. With my old board the bed does heat up normally = no error in the wireing. All the corresponding parts are ok. Could it be that the board has some faulty soldering?

kubik256, 2019-10-22:
Give it a try and measure voltage on heated bed output with heated bed turned on / bed temperature set ;)
[image]

oreo-ux, 2019-10-22:
I fixed it! The problem was that the board has the wires for the heated bed (Cr-10 stock one) swapped.
Just swap the two wires and it should work.
Thank you kubik256 for your help!

closed 2019-10-22 by oreo-ux