bigtreetech/BIGTREETECH-SKR-mini-E3 #55: Could the extruder and mobo fans be slowed down? (Fan1 and the other)

Chak10, 2019-10-21:
The noise is much higher than the stock, perhaps reducing speed would be better.
[image]

closed 2019-10-27 by Chak10

bojanpotocnik, 2019-10-27:
Unfortunately not as both are connected to the 24 V power supply directly.
[image]

Chak10, 2019-10-28:
@bojanpotocnik Thanks for the reply. I was thinking of using two potentiometers.

ManiRX100, 2019-11-13:
Hi 
Sorry for commenting on the closed issue. The fans are two noisy. Is it possible to use the below PWM marlin codes to slow down the fan?
//#define FAN_MIN_PWM 50
//#define FAN_MAX_PWM 128
These are commented out in configuration_adv.h. Or its not at all possible as the fans are connected to 24V supply directly ?