bigtreetech/BIGTREETECH-SKR-mini-E3 #56: No matter whatI do this file will not compile with vsc

WilliamHarned, 2019-10-21:
[image]

oreo-ux, 2019-10-21:
https://github.com/MarlinFirmware/Marlin/issues/15514
This solved the problem for me:)!

kubik256, 2019-10-22:
Configuration.h from Examples directory is not updated with the Marlin's config. You must check Configuration.h with the default one via WinMerge for structural changes ;)

WilliamHarned, 2019-10-22:
Thanks for the reply,
But not sure I understand your solution, could you explain in more detail.

Thanks,

damvcoool, 2019-10-23:
If you are using a recent copy of Marlin-BugFix-2.x (as of October 20, 2019) you might want to look  on Configuration.h around this lines (see text below)

`/**
 * Default Jerk limits (mm/s)
 * Override with M205 X Y Z E
 *
 * "Jerk" specifies the minimum speed change that requires acceleration.
 * When changing speed and direction, if the difference is less than the
 * value set here, it may happen instantaneously.
 */
//#define CLASSIC_JERK
#if ENABLED(CLASSIC_JERK)
  #define DEFAULT_XJERK 10.0
  #define DEFAULT_YJERK 10.0
  #define DEFAULT_ZJERK  0.3

  //#define LIMITED_JERK_EDITING        // Limit edit via M205 or LCD to DEFAULT_aJERK * 2
  #if ENABLED(LIMITED_JERK_EDITING)
    #define MAX_JERK_EDIT_VALUES { 20, 20, 0.6, 10 } // ...or, set your own edit limits
  #endif
#endif

#define DEFAULT_EJERK    5.0  // May be used by Linear Advance

/**
 * Junction Deviation Factor
 *
 * See:
 *   https://reprap.org/forum/read.php?1,739819
 *   http://blog.kyneticcnc.com/2018/10/computing-junction-deviation-for-marlin.html
 */
#if DISABLED(CLASSIC_JERK)
  #define JUNCTION_DEVIATION_MM 0.08  // (mm) Distance from real junction edge
#endif`

If you enable Classic_Jerk then you do not need to worry about JUNCTION_DEVIATION_MM, but if you disable Classic_Jerk (this is the default behavior) then you must have JUNCTION_DEVIATION_MM defined.
I hope this helps

fusemedia, 2019-10-24:
For me it was

not -> default_envs = STM32F103RC_bigtree
should be -> env_default = STM32F103RC_bigtree

that helped to get it compiled. Had the same issue before.

meltonpieman, 2019-10-24:
> 
> 
> For me it was
> 
> not -> default_envs = STM32F103RC_bigtree
> should be -> env_default = STM32F103RC_bigtree

It did not work for me!

WilliamHarned, 2019-10-24:
I got it to work by running vsc as an administrator.   Right-click on the
icon and click run as admin.

meltonpieman, 2019-10-24:
Hi William re

> I got it to work by running vsc as an administrator. Right-click on the icon and click run as admin.
> […](#)

Just tried it - no change. I have compared my platformio.ini and config.h with the latest BTT readme and it all matches. I am at a loss. I get the following warning messages (yellow) in the Terminal:
-------------------
FileNotFoundError: [Errno 2] No such file or directory: 'f:\\00 3D PRINTING\\3 ENDER 3 pro\\SKR mini E3\\0 BTT stuff board ver 1.2\\github\\BIGTREETECH-SKR-MINI-E3-v1.2-with-BLT-endstop\\.pio\\libdeps\\STM32F103RC_bigtree\\_tmp_installing-vfds7bdr-package\\U8glib-HAL-bugfix\\src\\clib\\HAL_example_files\\LCD_I2C_routines.c.example':
  File "C:\users\exken\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\exken\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\exken\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\users\exken\.platformio\platforms\ststm32\builder\main.py", line 91:
    target_elf = env.BuildProgram()
  File "C:\Users\exken\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\exken\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 122:
    _build_project_deps(env)
  File "C:\users\exken\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 47:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\exken\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\exken\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1043:
    project.install_dependencies()
  File "C:\users\exken\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 878:
    lm.install(uri)
  File "c:\users\exken\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 359:
    force=force)
  File "c:\users\exken\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 701:
    track=True)
  File "c:\users\exken\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 483:
    self.unpack(dlpath, tmp_dir)
  File "c:\users\exken\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 218:
    return fu.unpack(dest_dir, with_progress=False)
  File "c:\users\exken\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 118:
    self._unpacker.extract_item(item, dest_dir)
  File "c:\users\exken\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 38:
    self._afo.extract(item, dest_dir)
  File "C:\Users\exken\.platformio\python37\Lib\zipfile.py", line 1602:
    return self._extract_member(member, path, pwd)
  File "C:\Users\exken\.platformio\python37\Lib\zipfile.py", line 1673:
    open(targetpath, "wb") as target: 
-------------------
Do you get similar messages?
Am I missing something basic about getting the missing file in place?
Could you send me a text copy of your platformio.ini so that I can compare it with mine?

David

WilliamHarned, 2019-10-24:
David,
I'm at work right now but  I will get that to you when I get home.
Did you install git? https://git-scm.com/downloads

WilliamHarned, 2019-10-24:
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2019 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#define CONFIGURATION_H_VERSION 020000

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// config/examples/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a SCARA printer start with the configuration files in
// config/examples/SCARA and customize for your machine.
//

[... 1431 lines cut ...]
 *       Before starting, the nozzle moves to NOZZLE_CLEAN_START_POINT.
 *
 *   Caveats: The ending Z should be the same as starting Z.
 * Attention: EXPERIMENTAL. G-code arguments may change.
 *
 */
//#define NOZZLE_CLEAN_FEATURE

#if ENABLED(NOZZLE_CLEAN_FEATURE)
  // Default number of pattern repetitions
  #define NOZZLE_CLEAN_STROKES  12

  // Default number of triangles
  #define NOZZLE_CLEAN_TRIANGLES  3

  // Specify positions as { X, Y, Z }
  #define NOZZLE_CLEAN_START_POINT {  30, 30, (Z_MIN_POS + 1) }
  #define NOZZLE_CLEAN_END_POINT   { 100, 60, (Z_MIN_POS + 1) }

  // Circular pattern radius
  #define NOZZLE_CLEAN_CIRCLE_RADIUS 6.5
  // Circular pattern circle fragments number
  #define NOZZLE_CLEAN_CIRCLE_FN 10
  // Middle point of circle
  #define NOZZLE_CLEAN_CIRCLE_MIDDLE N

WilliamHarned, 2019-10-24:
#
# Marlin Firmware
# PlatformIO Configuration File
#
# For detailed documentation with EXAMPLES:
#
# http://docs.platformio.org/en/latest/projectconf.html
#

# Automatic targets - enable auto-uploading
# targets = upload

#
# By default platformio build will abort after 5 errors.
# Remove '-fmax-errors=5' from build_flags below to see all.
#

[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = STM32F103RC_bigtree

[common]
default_src_filter = +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared>
extra_scripts = pre:buildroot/share/PlatformIO/scripts/common-cxxflags.py
build_flags = -fmax-errors=5 -g -D__MARLIN_FIRMWARE__
lib_deps =
  U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip
  LiquidCrystal@1.3.4
  #TMCStepper@>=0.5.0,<1.0.0
  https://github.com/bigtreetech/TMCStepper
  #Adafruit NeoPixel@1.2.5
  https://github.com/bigtreetech/Adafruit_NeoPixel
  LiquidTWI2=https://github.com/lincomatic/LiquidTWI2/archive/master.zip
  Arduino-L6470=https://github.com/ameyer/Arduino-L6470/archive/dev.zip
  SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip
  SailfishRGB_LED=https://github.com/mikeshub/SailfishRGB_LED/archive/master.zip
  SlowSoftI2CMaster=https://github.com/mikeshub/SlowSoftI2CMaster/archive/master.zip

#################################
#                               #
#   Unique Core Architectures   #
#                               #
#  Add a new "env" below if no  #
# entry has values suitable to  #
#   build for a given board.    #
#                               #
#################################

#
# ATmega2560
#
[env:megaatmega2560]
platform          = atmelavr
framework         = arduino
board             = megaatmega2560
build_flags       = ${common.build_flags}
board_build.f_cpu = 16000000L
lib_deps          = ${common.lib_deps}
  TMC26XStepper=https://github.com/trinamic/TMC26XStepper/archive/master.zip
src_filter        = ${common.default_src_filter} +<src/HAL/HAL_AVR>
monitor_speed     = 250000

#
# ATmega1280
#
[env:megaatmega1280]
platform          = atmelavr
framework         = arduino
board             = megaatmega1280
[... 533 lines cut ...]
#
# Adafruit Grand Central M4 (Atmel SAMD51P20A ARM Cortex-M4)
#
[env:SAMD51_grandcentral_m4]
platform      = atmelsam
board         = adafruit_grandcentral_m4
framework     = arduino
build_flags   = ${common.build_flags} -std=gnu++17
extra_scripts = ${common.extra_scripts}
build_unflags = -std=gnu++11
lib_deps      = ${common.lib_deps}
src_filter    = ${common.default_src_filter} +<src/HAL/HAL_SAMD51>
debug_tool    = jlink

#
# Just print the dependency tree
#
[env:include_tree]
platform    = atmelavr
framework   = arduino
board       = megaatmega2560
build_flags = -c -H -std=gnu++11 -Wall -Os -D__MARLIN_FIRMWARE__
lib_deps    = ${common.lib_deps}
  TMC26XStepper=https://github.com/trinamic/TMC26XStepper/archive/master.zip
src_filter  = +<src/Marlin.cpp>

k6115907, 2019-10-25:
Guys I found to solution. I move my project to a folder in C: named compile. and it just workd. probably it has a problem if the location of the project is in Desktop.

kubik256, 2019-10-25:
> Guys I found to solution. I move my project to a folder in C: named compile. and it just workd. probably it has a problem if the location of the project is in Desktop.

This isn't problem with Desktop folder but with full-path length for all source files.
E.g.: C:\Users\very-long-named-user\Desktop\Marlin-bugfix-2.0.x\Marlin\src\gcode\..\..\..\..

So, keep base folder path as short as possible!!!
Also do not compile Marlin inside OneDrive folder unless sync is turned off, that also doesn't work ;) Obviously, if you have Desktop synced with OneDrive that will not work too.

meltonpieman, 2019-10-25:
Thank you William for the file copy and k6115907 and kubik256. Moved my project from 4 layers down in my F: drive partition to a new directory in the root C:/ directory and it compiled  ....... such a simple solution! The wonders of technology.

space-commits, 2019-10-28:
Nothing in this thread worked for me

damvcoool, 2019-10-29:
> Nothing in this thread worked for me

Which firmware are you trying to compile? and what error are you getting?

lamosca01, 2019-11-01:
Thank you k6115907 I was able compile after so many failed attempts.

Rubberduck84, 2019-11-20:
Thank you k6115907 its works for me also