bigtreetech/BIGTREETECH-SKR-mini-E3 #57: I cant compile any Marlin 2.0

Hobhub, 2019-10-22:
Processing STM32F103RC_bigtree (platform: ststm32; framework: arduino; board: genericSTM32F103RC)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 5.6.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0), tool-stm32duino 1.0.1
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing SailfishLCD
Using cache: C:\Users\seboy\.platformio\.cache\e1\f7005198ad8e3a4b619768bbbcd33ae1
Unpacking...
FileNotFoundError: [Errno 2] No such file or directory: 'c:\\Users\\seboy\\Documents\\ender 3\\BIGTREETECH-SKR-mini-E3-master\\firmware\\V1.2\\Marlin-2.0.x-SKR-Mini-E3-V1.2\\.pio\\libdeps\\STM32F103RC_bigtree\\_tmp_installing-p7c2ckfd-package\\SailfishLCD-master\\examples\\FlashForgeInterfaceBoardLCD\\FlashForgeInterfaceBoardLCD.ino':
  File "C:\Users\seboy\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\seboy\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\seboy\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\seboy\.platformio\platforms\ststm32\builder\main.py", line 91:
    target_elf = env.BuildProgram()
  File "C:\Users\seboy\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\seboy\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 122:
    _build_project_deps(env)
  File "C:\Users\seboy\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 47:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\seboy\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\seboy\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1043:
    project.install_dependencies()
  File "C:\Users\seboy\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 878:
    lm.install(uri)
  File "c:\users\seboy\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 359:
    force=force)
  File "c:\users\seboy\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 701:
    track=True)
  File "c:\users\seboy\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 483:
    self.unpack(dlpath, tmp_dir)
  File "c:\users\seboy\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 218:
    return fu.unpack(dest_dir, with_progress=False)
  File "c:\users\seboy\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 118:
    self._unpacker.extract_item(item, dest_dir)
  File "c:\users\seboy\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 38:
    self._afo.extract(item, dest_dir)
  File "C:\Users\seboy\.platformio\python37\lib\zipfile.py", line 1602:
    return self._extract_member(member, path, pwd)
  File "C:\Users\seboy\.platformio\python37\lib\zipfile.py", line 1673:
    open(targetpath, "wb") as target:
==================================================================== [FAILED] Took 1.45 seconds ====================================================================

Environment              Status    Duration
-----------------------  --------  ------------
megaatmega2560           IGNORED
megaatmega1280           IGNORED
at90usb1286_cdc          IGNORED
at90usb1286_dfu          IGNORED
DUE                      IGNORED
DUE_USB                  IGNORED
DUE_debug                IGNORED
LPC1768                  IGNORED
LPC1769                  IGNORED
melzi                    IGNORED
melzi_optiboot           IGNORED
rambo                    IGNORED
sanguino_atmega644p      IGNORED
sanguino_atmega1284p     IGNORED
STM32F103RE              IGNORED
STM32F103RC_fysetc       IGNORED
STM32F103RC_bigtree      FAILED    00:00:01.453
STM32F103RC_bigtree_USB  IGNORED
STM32F4                  IGNORED
STM32F7                  IGNORED
ARMED                    IGNORED
STM32F103VE_longer       IGNORED
mks_robin                IGNORED
mks_robin_lite           IGNORED
mks_robin_mini           IGNORED
mks_robin_nano           IGNORED
jgaurora_a5s_a1          IGNORED
STM32F407VE_black        IGNORED
BIGTREE_SKR_PRO          IGNORED
BIGTREE_BTT002           IGNORED
teensy31                 IGNORED
teensy35                 IGNORED
STM32F103CB_malyan       IGNORED
esp32                    IGNORED
fysetc_f6_13             IGNORED
linux_native             IGNORED
SAMD51_grandcentral_m4   IGNORED
include_tree             IGNORED
=============================================================== 1 failed, 0 succeeded in 00:00:01.453 ===============================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

I've tried adding missing files manually, vscode has firewall and administrator, please help.

Hobhub, 2019-10-22:
update: it works when I disable sailfishlcd
[common]
default_src_filter = +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared>
extra_scripts = pre:buildroot/share/PlatformIO/scripts/common-cxxflags.py
build_flags = -fmax-errors=5 -g -D__MARLIN_FIRMWARE__
lib_deps =
  U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip
  LiquidCrystal@1.3.4
  #TMCStepper@>=0.5.0,<1.0.0
  https://github.com/bigtreetech/TMCStepper
  #Adafruit NeoPixel@1.2.5
  https://github.com/bigtreetech/Adafruit_NeoPixel
  LiquidTWI2=https://github.com/lincomatic/LiquidTWI2/archive/master.zip
  Arduino-L6470=https://github.com/ameyer/Arduino-L6470/archive/dev.zip
  #SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip
  SailfishRGB_LED=https://github.com/mikeshub/SailfishRGB_LED/archive/master.zip
  SlowSoftI2CMaster=https://github.com/mikeshub/SlowSoftI2CMaster/archive/master.zip

Wood0702, 2020-04-08:
Thanks for posting your problem- this fixed the issue of compiling  for me too

Wood0702, 2020-04-08:
@bigtreetech Please could you add a note to your guide that if we uses a fresh version of marlin then to take the above steps.

comment | #SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip