bigtreetech/BIGTREETECH-SKR-mini-E3 #61: Skr Mini E3 1.2 problem to flash

thedark20, 2019-10-23:
Hi;
i try to flash my SKR mini E3 1.2 with newest firmware i took here.

I configure all requirement like it is specified here in the readme :  https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2

but after insert the sdcard, i see the led flashing and nothing.. blue screen.. i got the cur file on sdcard also with the bin. 
i don't understand what i miss 
thanks for help me to understand and solve this things

[Marlin-2.0.x-SKR-Mini-E3-V1.2 config files.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/3764491/Marlin-2.0.x-SKR-Mini-E3-V1.2.config.files.zip)

os3dprinting, 2019-10-24:
Got the same issue... Firmware compiles with 5 Warnings and a strange hint on the maple usb part.
`C:\users\bfeat\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\STM32ADC\src\utility\util_adc.c:10:30: warning: 'adc_result' initialized and declared 'extern'
 extern volatile unsigned int adc_result = 0;
 A                             ^~~~rchiving .pio\build\STM32F103RC_bigtree\lib395\libEEPROM.a`

[image]

os3dprinting, 2019-10-24:
@thedark20 so we get the same warnings, in my case there’s no difference between setting adafruit neopixel on or off in libignore.

os3dprinting, 2019-10-25:
I removed Ardunio, Atom, Platformio, VSC and Git, restarted the computer and setup a new build environment. Display is working now, the 5 compiling warnings stay the same.

thedark20, 2019-10-25:
i used an another marlin 2.x.x from another guy and i got a firmware wich work good. 
https://github.com/gazcbm/Marlin-2.0.x-SKR-Mini-E3-v1.2/tree/bltouch-improvements
i modify for my need, and no problem at all. Look like they are a problem in the BBT new files for 1.2.

m4rl0ne, 2019-10-25:
Had the same issues. This is a problem with platformio. Try updating it, restarting your computer or using a different IDE.

swilkens, 2019-10-27:
> since i didn't use adafruit, i leave it in the libignore variable in platforme.io. If i didn't do that, i can't complie.
> i got this warning :
> [image]
> 
> the code is compiled but that's all, after nothing to the screen. If i use a previous marlin 2.x.x, i got the TMC connection error.

These errors show up if you use the latest Marlin-2.0.x-bugfix (that needs TMCStepper 0.5.0 or higher) for this board in combination with the TMCStepper library from https://github.com/bigtreetech/TMCStepper.

This is because latest marlin-2.0.x-bugfix needs an updated version of the TMCStepper library. The actual fix is that https://github.com/bigtreetech/TMCStepper is brought up to date with the current upstream.

Alternatively, you can try to use upstream TMCStepper and merge the modifications from https://github.com/bigtreetech/TMCStepper. But really @bigtreetech should update https://github.com/bigtreetech/TMCStepper to work with the latest marlin-2.0.x

pmjdebruijn, 2019-10-27:
I've had success building current 2.0.x, the repo below includes a fully automated build script:
https://github.com/pmjdebruijn/BIGTREETECH-SKR-mini-E3

swilkens, 2019-10-27:
> I've had success building current 2.0.x, the repo below includes a fully automated build script:
> https://github.com/pmjdebruijn/BIGTREETECH-SKR-mini-E3

`# discovered from BigTreeTech reference firmware sources`

I gather these changes are not defined in the guide?

pmjdebruijn, 2019-10-27:
Nope, thought some of them aren't critical to be able to get a functioning firmware of course.

bojanpotocnik, 2019-10-28:
@pmjdebruijn I have a very strange situation here.

Your script does not work for me. Marlin checkout is at [Confirm object cancellation (#15660)](https://github.com/MarlinFirmware/Marlin/commit/9aff30da0c82b504561166dce60442acc33ddeb8), PlatformIO version 4.1.0rc1. After flashing the produced firmware.bin, only the backlight is lit and it does not boot (#49 exactly). The produced FW size is 216172 bytes.

Your firmware.bin, however, does work.

Can you please share the archive of your working directory, so I could diff yours and mine to find out what is the cause of the problems?

bojanpotocnik, 2019-10-29:
I solved it.

Created a fresh Debian virtual machine, installed Python and Platformio, checked out your repository and run the script. The same version of Python, the same version of Platformio, the same process - if done on the fresh VM it works, but compiled on my machine it doesn't, with the same generate FW size.

> I removed Ardunio, Atom, Platformio, VSC and Git, restarted the computer and setup a new build environment. Display is working now, the 5 compiling warnings stay the same.

> Had the same issues. This is a problem with platformio. Try updating it, restarting your computer or using a different IDE.

I could/can not believe that such things could be the cause of the problem.
Oh PlatformIO, _Cross-platform build system without external dependencies to the OS software_...

pmjdebruijn, 2019-10-30:
If you take a close look at the script, your system does not need to have PlatformIO installed, the script purposefully avoids any pre-existing PlatformIO installs, and spawns a Python VirtualEnv in which a fresh PlatformIO is installed.

blackasdf, 2019-11-04:
Please let me know if i'm saying and doing something completely wrong as i have no experience in this. Using a virtual machine i was able to compile BTT firmware and also official 2.0 bugfix. However i was getting this error : "\platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\STM32ADC\src\utility\util_adc.c:10:30: warning: 'adc_result' initialized and declared 'extern' extern volatile unsigned int adc_result = 0;"
After some minor research i've downloaded the Arduino_STM32 package from github and found util_adc.c under "Arduino_STM32/STM32F1/libraries/STM32ADC/src/utility and pasted it to /users/MyName/.platformio/packages/framework-arduinoststm32-maple/STM32F1/libraries/STM32ADC/src/utility/ " overwriting the old util_adc.c that came with the latest version of VSCode. Next i tried again to compile Marlin and it finished successfully without any error or warning.

bojanpotocnik, 2019-11-04:
@pmjdebruijn I noted the venv in the script (I examined it well to see what am I doing wrong). Despite the venv the problem persisted, the cause and the solution - or better to say workaround - was later discovered in [MarlinFirmware/Marlin/#15767](https://github.com/MarlinFirmware/Marlin/issues/15767).

@blackasdf Those are warnings (yellow/orange) not errors (red) and not critical. Usually there are also a few about unused u8g2 initialization sequences. As you discovered, although they are present the firmware compiles just fine. I suppose those minor "typos" will be fixed in the upstream Marlin in the near future.

thedark20, 2019-11-04:
hi, i have updated my VS studio and plateformio and no error no warning. I didn't try with the last version of marlin 2.x.x but https://github.com/gazcbm/Marlin-2.0.x-SKR-Mini-E3-v1.2/tree/bltouch-improvements

DrTobogganMD, 2019-11-08:
I also cannot get this to load onto my printer. I'm using an Ender 5, so a few tweaks are needed (I can't use the included .bin file). The compile works fine, but then the printer refuses to boot.  The included BIN boots fine, but obviously will not work.

codedmind, 2019-11-08:
Anyone can help how to get TRU-LEV 600 configuration in the firmware?
When i try to use this firmware or marling some of the options are missing or can't find them...

Thanks

swordfish45, 2019-11-08:
Downgrade of platformio worked for me.

4.1.0 = fail
4.0.3 = success

https://github.com/MarlinFirmware/Marlin/issues/15767#issuecomment-549156764

DrTobogganMD, 2019-11-08:
Yea, Keith confirmed on the Ender 5 FB Group that it's with the Release version of PIOCore 4.1 and STM32 hardware.  They're pushing a fix to the marlin branch today he said. https://www.facebook.com/groups/ender5creality/permalink/1525744094245515/

swordfish45, 2019-11-08:
You can apply the patch yourself

https://github.com/MarlinFirmware/Marlin/issues/15767#issuecomment-551820528

pmjdebruijn, 2019-11-09:
@ivankravets patch worked fine for me:
https://github.com/ivankravets/Marlin/commit/c14f732017a727e987074afad1b03d7a66b20d38

codedmind, 2019-11-09:
@pmjdebruijn To me too :)
Thanks @ivankravets

cv636, 2019-11-13:
Similar issue, super frustrating. I have VS Code 1.40.0 and PIO Home 3.0.0 Core 4.1.0.  I build the firmware and place it on SD card, the motherboard flashes and freezes on a blue lcd screen.  I've tried multiple forks and all end the same way.  I did find a fork that had a firmware.bin already available and plugged that in, and to my surprise it worked. I tried to compile without making any changes and i'm back to the blue screen. Does anyone have any suggestions?

pmjdebruijn, 2019-11-13:
As far as I can tell, this issue has been definatively resolved, and the fix was committed to upstream Marlin a few days ago:
https://github.com/MarlinFirmware/Marlin/commit/4ff708cb3a2fe6702f709c4632d3653ac57460d9
So likely your configuration is wrong/incomplete. Or your output firmware.bin is too large, it should be less than 226 or 228 if I recall correctly.

My script is probably a better (as in more complete) reference than the official docs (also includes a recent prebuilt firmware.bin):
https://github.com/pmjdebruijn/BIGTREETECH-SKR-mini-E3-V1.2
My script essentially does everything, it only needs a bourne shell and python3 with venv.

cv636, 2019-11-13:
I'd agree as well, but not sure what would be the problem here. 
I downloaded a "fixed" version of marlin with some BL touch settings. The fimrware.bin file that was in the github fork worked perfectly. I compiled my own firmware.bin WITHOUT any changes (should've been the same one in the CompiledFimware folder) but that one failed. Looks like it may be an issues with VSCode / PIO

As for filesize, both (working and nonworking) firmware.bin files were 215KB in size

Once my current print is done I'll test out your fork and hope for the best.

ivankravets, 2019-11-14:
See https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/pull/105

Could someone re-test this repo using the latest PlatformIO Core 4.1 or above? https://github.com/ivankravets/BIGTREETECH-SKR-mini-E3

realitybrief, 2019-11-15:
Ivan, compiled with 4.1 and works great, thank you!

axrhart, 2019-11-15:
I tried this repo for the v1.2 firmware yesterday with a fresh PlatformIO installation on Arch to no success. I also tried [this adapted marlin fork](https://github.com/morningreis/Marlin-SKR-E3-Mini-1.2) and got a blue screen and no boot beyond bltouch testing. The firmware that came with the board still flashes and works fine.

So this problem is still relevant for some users but might be purely a PlatformIO issue?

closed 2019-11-16 by bigtreetech

fullrespect, 2019-11-29:
i am getting the same problem, received the SKR Mini v1.2 for Ender 3. Flashed the software i have compiled, and blue screen. 
The firmware that came with the board still flashes and works fine.
Any straight forward solution for this ?

ivankravets, 2019-11-29:
You need to use the latest source code from https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3

fullrespect, 2019-11-29:
thanks, i will check right now. Downloaded the v1.2 firmware, using VSC + GIT + PatformIO.
Not changed anything in the firmware i downloaded, just compile it as is. We will see if it will make properly and boot up my board.

fullrespect, 2019-11-29:
this is magic, yesterday it won't boot. Today - same thing i did, no changes to the firmware. Compiled, flashed, sheeebang - all booted up... What the heck ? I am pleased, but why it did not work yesterday ?

ivankravets, 2019-11-29:
Looks like you used old source code. Please do not forget to CLEAN project if you used old source code.

fullrespect, 2019-11-29:
How to clean the project in Visual Studio Code ?

ivankravets, 2019-11-29:
Left side bar > PlatformIO > Proejct Tasks > Clean or from CLI `pio run -t clean`

HeathNL, 2019-12-09:
> I've had success building current 2.0.x, the repo below includes a fully automated build script:
> https://github.com/pmjdebruijn/BIGTREETECH-SKR-mini-E3-V1.2

Heey Pascal! ik ben van plan jou file te gaan gebruiken voor mijn Ender 3. is dit gewoon op de SD kaart zetten en flashen?

Ik heb namelijk behoorlijk lompe bed clips op mijn bed zitten en alle bestanden die ik tot nu toe geprobeerd heb levelen dus op de bed clips. ben gisteren 5 uur bezig geweest met zelf een programma schijven maar compleet mislukt vandaar.

pmjdebruijn, 2019-12-09:
> Heey Pascal! ik ben van plan jou file te gaan gebruiken voor mijn Ender 3. is dit gewoon op de SD kaart zetten en flashen?
>
> Ik heb namelijk behoorlijk lompe bed clips op mijn bed zitten en alle [...]

Given this is a public forum, I'll stick to English. But you can flash my firmware.bin as you normally would, but at your own risk of course. My firmware build avoids the outer edges for bed level corners and bltouch automatic bed leveling, so you should be fine there, though I never tested with clips myself.

Don't forgot to pay attention to the README, as my later firmwares require a BLtouch to be installed using the Creality official bracket.

HeathNL, 2019-12-09:
Ah yeah sorry. I tried it but it did not work :( im trying for 3 days now to get this Marlin working but it wont >:( im at the point to burn this printer down >:(

pmjdebruijn, 2019-12-09:
Basically every firmware.bin I'll upload has produced at least one decent print on my Ender 3.

What specifically doesn't work? If you're having issue applying the firmware.bin, make sure the filesystem on your SD card isn't broken. Modern operating systems can sometimes be rather tolerant of semi broken filesystems, while the STM bootloader (or other embedded software for that matter) may not be as forgiving, which is why a filesystem issue might not be so obvious when using the card on your computer.

HeathNL, 2019-12-09:
Flashing the firmware works, but as soon as i press auto home it says "Home Z first" and no matter what i try nothing helps...and when i press bed level it says "stopped"

pmjdebruijn, 2019-12-09:
@HeathNL and you have a working BLtouch (it visibly finishes it's selftest on power-on) connected to your PROBE and SERVO sockets?

HeathNL, 2019-12-10:
> @HeathNL and you have a working BLtouch (it visibly finishes it's selftest on power-on) connected to your PROBE and SERVO sockets?

When i put on my printer the BLTouch pin is going in and out tot a few times. When i push "bed leveling" in the printer of goes to the lift front corner hits the X and Y sensor and goes to the middel of the bed, and than my printer says " home Z first" and of stops.

pmjdebruijn, 2019-12-10:
> When i put on my printer the BLTouch pin is going in and out tot a few times. When i push "bed leveling" in the printer of goes to the lift front corner hits the X and Y sensor and goes to the [...]

That did not answer whether you have the BLtouch connected to PROBE as oppossed to Z-STOP... For my firmware it should be connected to PROBE, as detailed in the README.

HeathNL, 2019-12-10:
> That did not answer whether you have the BLtouch connected to PROBE as oppossed to Z-STOP... For my firmware it should be connected to PROBE, as detailed in the README.

I used the teach tech video on YouTube. Zo i THINK i have it in the servo one and the Z-stop.

Can this be the problem?

pmjdebruijn, 2019-12-10:
@HeathNL of course that's the problem.

HeathNL, 2019-12-10:
> @HeathNL of course that's the problem. My README explicitly states you need to have the BLTouch connected to PROBE and that Z-STOP is ignored.

Im going to check this right away when im at home!! I think i've missed it because i was so frustrated because it is not working. I let you know!

HeathNL, 2019-12-10:
It worked!! All i have to do is fix the offset stuff and hope it prints well. Thank you very much!!

HeathNL, 2019-12-11:
The 3Dtouch is working now!! But when i press bed leveling and than start a print, i dont see the Z lead moving and i have like 0.1mm level difference in my bed.

pmjdebruijn, 2019-12-11:
Oh you have BLtouch clone, keep in mind I've only tested this using a genuine BLtouch 3.1, but with that said...

I'm puzzled by your statement "when i press bed levelling", this is not how you use a bltouch (or clone presumably)...

Typically you setup your slicer to insert a G29 after the G28 in your start gcode.

HeathNL, 2019-12-12:
> Oh you have BLtouch clone, keep in mind I've only tested this using a genuine BLtouch 3.1, but with that said...
>
> I'm puzzled by your statement "when i press bed levelling", this is not how you use [...]

Aaaah oké. Well i have a Trianglelab 3Dtouch and Many say it is the same as the BLTouch. I have no clue how to put in G codes in the cura slicer. I just put in my settings and press slice and i print it =| 

I have watched a few video's on YouTube and they all press bed leveling first and than start the print.

swordfish45, 2019-12-12:
> Aaaah oké. Well i have a Trianglelab 3Dtouch and Many say it is the same as the BLTouch. I have no clue how to put in G codes in the cura slicer. I just put in my settings and press slice and i [...]

In Cura under machine profile there is start and end gcode. Add g29 after the g28 line

HeathNL, 2019-12-13:
> In Cura under machine profile there is start and end gcode. Add g29 after the g28 line

I found it and added the G29 code, TNX! But i still cant see the Z lead moving while printing after leveling.

pmjdebruijn, 2019-12-13:
Please be a little more forthcoming with information, at every turn you provide the least amount of information possible for us to "get" what might be going on.

But you seem to imply that the printer successfully moves the Z lead while probing the 9 points on your printbed? If so, your Z stepper is likely connected properly, and thus I'd suspect slicer settings.

pmjdebruijn, 2019-12-13:
Please take a look at: https://www.youtube.com/watch?v=Y8WlB8wUHuU
Also, posting a video of your issue will likely be quite enlightening...

HeathNL, 2019-12-13:
> Please take a look at: https://www.youtube.com/watch?v=Y8WlB8wUHuU
> Also, posting a video of your issue will likely be quite enlightening...

Yea im sorry for that, ill try to give some more info also im going to start a xyz cube soon so ill try to film the start.

I got a Ender 3 pro with yellow springs, glass bed, Trianglelab 3Dtouch and a bigtreetech skr mini e3 v1.2 board. Printing with inofill3D pro1 filament.

When i added the G29 code after the G28 and he does that! When i load the print it heats up and goes bed leveling. But when it starts to print he puts down the line and i dont see the Z lead moving for compansation.

pmjdebruijn, 2019-12-13:
Well, that another matter entirely... so you don't see your Z lead moving during a layer, not between layers...

And to be honest I've never obviously noticed that, because keep in mind if all is well those changes in Z should be really tiny. Larger changes should be addressed by "bed level corners", not auto bed leveling.

So in my also somewhat limited experience it seems you're looking for a problem where there isn't one?

HeathNL, 2019-12-13:
> Well, that's another matter entirely... so you don't see your Z lead moving during a layer, not between layers...
> 
> And to be honest I've never obviously noticed that, because keep in mind if all is well those changes in Z should be really tiny. Larger changes should be addressed by "bed level corners", not auto bed leveling.
> 
> So in my also somewhat limited experience it seems you're looking for a problem where there isn't one?

Well, if it should compansate 0.1mm (thats how curvy my bed is and cant get it better even with corner leveling) i was thinking you should see the Z lead moving because he has to compensatie 0.1mm right? Tell me if in wrong!! Because than in worried about nothing  :)

pmjdebruijn, 2019-12-14:
How large is the print you're testing this with? Keep in mind if you have a 0.1mm dip over the full length of your bed, this will be next to nothing on a 5 cm print. You may want to try printing a 20x20cm square.

HeathNL, 2019-12-30:
> How large is the print you're testing this with? Keep in mind if you have a 0.1mm dip over the full length of your bed, this will be next to nothing on a 5 cm print. You may want to try printing a 20x20cm square.

Sorry for my late reaction! Everything seems to work doe so far but sometimes when i start the print he does 1 bed level point and the 3dprobe is blinking and or just starts to print (it does work so...) 

But now my extruder starts clicking after i installed the skr mini v1.2 board with TMX2209 UART, my extruder is cslibrated, my nozzle is not to close to bed, my speed is between 50 and 60 so not that slow and i have no clogs.

Hou have any idea?

pmjdebruijn, 2019-12-30:
Just seeing one "bed level point" in the center means it's just doing Z homing, so really no bed leveling at all, which means you forgot to add G29, or are re-using a previously generated gcode before you added G29. If it does have G29, you BLtouch might be off, you can adjust it using the hexnut at the top, as already mentioned on my repository.

As far as step skipping goes on the extruder, if you'd have researched the TMC drivers a little, you'd have noticed it has multiple modes, like the popular silent but weaker StealthChop and the more powerful but louder SpreadCycle. If I recall correctly you can switch between via the menu, you might try SpreadCycle.

If you'd have browsed recent Marlin commits, you might have noticed:
https://github.com/MarlinFirmware/Marlin/commit/329f5b4df97111a7f8019292aa3ac6713e0656ff
HYBRID_THRESHOLD makes Marlin automatically switch between the two, but the upstream thresholds seem excessively conservative to me, making the printer louder than it needs to be.

[12 more replies]