bigtreetech/BIGTREETECH-SKR-mini-E3 #62: Printer Menu beeps and switches between homescreen and settings.

fusemedia, 2019-10-24:
I followed the steps, compiled firmware with Marlin 2.0 en flashed the SKR mini E3 v1.2 and everything works fine. Auto home, printing.

But when the bed or nozzle starts to heat up the menu starts beeping and switches between homescreen and settings untill the temperature has reached. After that it now and then beeps when giving input to the  nozzle or bed.

Already had some tip about disabling TEMPPIDBED but that didn't helped.

MattFryer, 2019-10-31:
I had this exact same issue. Took me a while to work out what the problem was but eventually I realised the pins on the bottom of the SKR mini E3 are slightly longer than the stock Creality board which means they come too close to the metal of the electronics case.

Unscrew the board and put some kapton tape (or similar) under where the board goes. Then screw the board back in and you should have sorted the problem.

fusemedia, 2019-10-31:
Oh my man Matt, could hugh you. I was planning on rewiring, checking cables of the bed if they wheren't damaged or bothered by interfering somehow.

Just did it, quick with some black ductape and screwed the board very loosely so it has more clearance and YES! Heated the bed and no beeping (so far)! THX!!

fusemedia, 2019-10-31:
A bit to hasty with my comment ... but I'll keep searching on this path for a solution. Bummer.

MattFryer, 2019-10-31:
Could be that the sharp pins are poking through the soft electrical tape? File the pins down?

I spent a day trying to work it out. After several hours I found if I unscrewed the metal mound the LCD is attached to from the main printer frame it stopped. Give that a go and if it stops then it will definitely be something touching the metal frame that shouldn't be.

fusemedia, 2019-10-31:
"unscrewed the metal mound the LCD is attached to from the main printer frame it stopped. Give that a go and if it stops then it will definitely be something touching the metal frame that shouldn't be."

Will do, believe it's in the right direction. Will test tomorrow.

fusemedia, 2019-11-04:
busy weekend and my printer is at the office, with Halloween didn't get to it.

But now I can confirm 100% you're right. I left the board outside the case and no bedheating - menu beeping issues at all. So I'm gonna file some pins down, look for a better insulator and even suspect the case of the SD card touching the outercase? Anyway, issue detected and now can work my way to a solution.

MANY MANY thx Matt!

MattFryer, 2019-11-04:
No problem. 

I would be surprised if the SD card slot touching the case was causing it as the outside of the slot should be grounded anyway. Got to be worth a try though.  I'll try bridging mine to the case when I get the chance and see what happens. If it turns out to be the problem then one of these might be the solution: [https://www.myminifactory.com/object/3d-print-sd-card-saver-ender-3-76934](https://www.myminifactory.com/object/3d-print-sd-card-saver-ender-3-76934). 

I was planning on printing one of these at some point as I've been known to miss the slot and post the SD into the electronics case instead.

fusemedia, 2019-11-04:
Well if you have an ender 3 (maybe it counts for 5 as well) you can print the teaching tech all in one plastic case. When it comes to conductivity it must solve all problems ;-)
https://www.thingiverse.com/thing:3688967

MattFryer, 2019-11-04:
Unfortunately, mine is an Ender 5 and the electronics enclosure is completely different.

awdorrin, 2019-12-01:
I am experiencing the same issue with an Ender 5 and SKR E3 Mini V1.2 board.
After reading yoru suggestions above, I did go through with a pair of diagonal cutters and trim all of the through board leads. This drastically reduced the menu switching, but did not eliminate it.
Disconnecting the LCD control from the main-frame, however, has appeared to completely eliminate.

I assume there must be some sort of ground loop through the frame, and that the SKR board is perhaps not as well designed as it could be in regards to noise. (Seeing other drawing similar conclusions.)

Will have to take a look at how the LCD control board is mounted to that panel, and see if maybe some non-conductive washers might resolve the issue.

Thanks to you both for sharing your experiences, its been very helpful to me.

AndyinMission, 2020-01-04:
I have this issue too on an ender-5, mini e3 1.2 board, bltouch, Marlin 2.x.

I tried isolating the bottom of the board from the printer case witjh 2 layers of tape and also replacing the encoder part itself.  Neither helped.  I found a couple other people on reddit who just installed the board and also see this.  Both on ender-5 printers.

AndyinMission, 2020-01-05:
Update - I probed around with the multi-meter and the board didn't seem to be ground whatsoever.  It was just floating.  I ran a wire from the SD card's frame solder point, which seemed to be soldered to the board's ground plane/layer, to the ground on the input from the mains, which is tied to the frame and PSU chassis.  After a 1 hr print just now the issue did not re-occur.  Will monitor but until this modification it was occurring essentially all the time.  So maybe it's fixed.

rooterkyberian, 2020-03-17:
Same issue, tape on bottom of board under connectors didn't help.

BIQU SKR E3 Mini v1.2 in stock Ender 3

Brainstorm4ever, 2020-06-06:
installed it today on the ender 5 with bltouch. 
mini e3 1.2 

Any solution that fixes this? (using the latest bugfix version)

txt4nk, 2020-06-06:
Brainstorm, print this piece
[Ender5 Screen Offset EMI Fix_6mm.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4740924/Ender5.Screen.Offset.EMI.Fix_6mm.zip)

and install it like this:
[image]

txt4nk, 2020-06-06:
The issue is this line of boards do not have EMI filtering (or good filtering at least..) so they transmit the EMI through the chassis. If the metal frame that holds the screen touches the metal frame (even through the screws, tried shims and it did not work) the screen gets effected as it is severely sensive to the EMI.

Brainstorm4ever, 2020-06-06:
> Brainstorm, print this piece (print it solid, 100% infill for strength. I printed it at 40% first and it was too flimsy).
> [Ender5 Screen Offset EMI Fix_6mm.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/4740924/Ender5.Screen.Offset.EMI.Fix_6mm.zip)
> 
> You will need six M5x8 bolt & six M5 t-nuts to install it as seen below:
> [image]

Thanks. I will try it after. i get those m5 bolts and tnuts
Thanks

txt4nk, 2020-06-06:
No problem. You can simply pull the screen holder off the printer frame and set it on the table until you get the screws and t nuts

txt4nk, 2020-06-07:
I posted the wrong screw and t nut size, they are M4. 

Also I put ferrite chokes between a few connections and just put the screen directly back on the printer without that printed stand off. This seems to have worked too, I removed the stand off and there's no problems anymore.

I put them in the following places:
Wall & PSU
PSU & Main Board
Main Board & Hot End
Main Board & Heated Bed
Main Board & Hot End Fan

The following probably won't apply to you, as I've added these all myself.
PSU & Buck converters
PSU & LED Light

triwats, 2021-01-04:
Thanks for this guys, having this issue myself, will give it a go!

gorbaloid, 2021-01-23:
I had this issue but with the original Ender Creality 3 screen.. i took the potentiometer knob off and the noise stopped..guessing it weighs too much or the sensitivity for the click is too high.

Sjoerd, 2021-02-27:
Same issue here!

EastRiverCreature, 2021-04-06:
I had the same issue and followed the adive in this thread - I covered the base of the PCB in Kapton tape. Unfortunately this did not fix the issue completeley. However the hint adding ferrite to the ribbon cable did the trick. I have cut off the connector and could slide the ferrite over and crimped it on again. 

Thanks to everyone in this thread sharing the advise!

PropDusty, 2021-07-06:
same issue after putting together 4 units Ender 3 standard.. the 5th one had this issue none of the rest were effected. It seams to be coming off the frame most prob a small voltage from the power supply or the main board some where touching. . I noticed on the display a little tab still there used for the pole to stop the knob washer like on all nobs was not fully cut off. I cut this off and the beeps lessoned. I took the display off the frame and wala its gone. I put nylon washers on the 2 screws attaching the display to the frame and the beeps and screen ticks stopped.. I still want to hunt down why because none of my other printers do this.

Na1w, 2022-03-16:
I had the same issue with mine, made a quick modification to marlins debounce mechanism and it completely solves the issue for me (with no negative impacts what I’ve noticed so far).

Patch available here if someone else may find it useful. 

[https://gist.github.com/Na1w/6709640aff5a6a85b8f749f183ae2d2f](https://gist.github.com/Na1w/6709640aff5a6a85b8f749f183ae2d2f)