bigtreetech/BIGTREETECH-SKR-mini-E3 #63: Unable to get NeoPixels to work

h00etn, 2019-10-24:
Downloaded the firmware.bin for the v1.2 (where it says that neopixel are fixed). But no matter where I connect the LED-strip, it doesn't turn on. And M150 just gives "unknown command".

I have tried to use:
- the left three pins (from top to bottom: GND, 5V, Signal)
- the right three pins (from top to bottom: GND, 5V, Signal) and with the jumper off
- the pins inside the Neopixel 'slot' (from left to right: 

And I have also tried to edit the firmware (from here) within Atom (as I have done before). In that version I saw that neopixel was commented out so I uncommented it. But I'm not able to upload the updated firmware. The build goes fine, but the uploading fails.

What should I do?

TheGadge, 2019-11-09:
Did you ever sort your neopixel? Have you ordered the dcdc addon board?

bojanpotocnik, 2019-11-18:
Today I had an interesting observation. I used [my fork](https://github.com/bojanpotocnik/Marlin) (compiled FW included, check README) where NeoPixels are enabled (among other tweaks) and obviously I didn't want to push the commits until I tested everything works.

When I attached my WS2812B LED strip with 10 LEDs, they didn't work at all. I thought it was a configuration issue, but after a lot of head scratching and triple-checking the firmware I finally tested with just 1 LED. And it works! This is very strange to me, because LEDs by themselves - when OFF - are not supposed to consume so much current... Just my 5 cents, if someone would have similar problems.

> - the left three pins (from top to bottom: GND, 5V, Signal)
> - the right three pins (from top to bottom: GND, 5V, Signal) and with the jumper off

Which pins do you mean by that ([BTT SKR MINI E3 V1.2PIN.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V1.2/BTT%20SKR%20MINI%20E3%20V1.2PIN.pdf))? 🤔

> the pins inside the Neopixel 'slot' (from left to right: GND, SGNL, 5V)

This is the correct connector. As shown in [The Notement of BTT SKR MINI E3 V1.2.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V1.2/The%20Notement%20of%20BTT%20SKR%20MINI%20E3%20V1.2.pdf) you connect NeoPixel to the `Neopixel` connector where left-most pin is GND, middle Signal (PC7) and right-most +5V (looking at the board so that the USB connector is on the right side). Then you move the `Neo-PWR` jumper to the right position so that you connect +5V from the `Neopixel` connector to the board supply - **IF** you have a small enough number of LEDs so that the onboard switcher can handle it (in my case, 10 is too much).

> And M150 just gives "unknown command".

This happens if you don't [enable and configure](https://github.com/bojanpotocnik/Marlin/commit/a7b7193e6f0273df197a4404e01378038672a938#diff-8270513bcf2a548cd418cdfe123f27d3R2159) `NEOPIXEL_LED`, as you already discovered. NeoPixel will not work in this case. `NEOPIXEL_STARTUP_TEST` is also a good feature to enable - it let you see if your NeoPixel work immediately after powering the printer (it just lits them Red.... Green.... Blue.... OFF).

> But I'm not able to upload the updated firmware. The build goes fine, but the uploading fails.

I suppose you've run into [_Self-compiled firmware not working_](https://github.com/bojanpotocnik/Marlin/issues/1), which is now already solved. If anything other than that, more verbose description would be welcome.