bigtreetech/BIGTREETECH-SKR-mini-E3 #65: Massive HotEnd temp YoYo ing

timryder, 2019-10-25:
I'm having an impossible time getting my stock HotEnd Inman ender 3 to keep a constant temp. it will yo yo +-10 degs from Target and eventually thermal runaway protection kicks in.  I've tried several times to PID tune with no success.  Any thoughts?

brew99, 2019-10-25:
Post your current PID numbers.   Also take a good look at your heater cartridge and thermocouple to see that they are fully inserted in the hotend block

timryder, 2019-10-27:
I have tried several times to tune and the numbers nearly always come out the same. In didn't remove any hardware between the stock creality board and this. there should be no reason I'm having any issues regulating temps when 1hr. prior on the stock board I wasn't.

swilkens, 2019-10-27:
Have you tried using the P, I, D values from the creality board? The hotend remainds the same, logically the same values should apply - as the system you are controlling hasn't changed.

timryder, 2019-10-27:
Yes of course,. the SKR-Mini-E3 by default already uses the Creality Tuning values.  That did not help. Think I got it fixed so far.... more to come

timryder, 2019-10-29:
Not sure which operation fixed it... but I took out my HotEnd wires from the board, pried open the crappy screw terminals so I could put wires in fully.  Swapped my two wires in their terminals and added a 10uF capacitor right at the screw terminals for the HotEnd. All of the above fixed it.

closed 2019-10-29 by timryder