bigtreetech/BIGTREETECH-SKR-mini-E3 #66: SD Cards bigger than 8GB don't work

damvcoool, 2019-10-25:
I have noticed that currently this board does not appear to be able to handle SD cards that are 8GB or bigger.

I can use the 8GB card to upgrade Marlin, but past this point the SD card does not initialize. I am currently using a 2GB SD to print.

I did not have this issue with the same build of Marlin (Bugfix-2.x from Octoboer 23 2018)  in the Creality Board.

TheGadge, 2019-10-27:
mine came with, and I've flashed several times with a 128mb?? (1.2)

damvcoool, 2019-10-28:
Mine came with a 128MB MicroSD Card as well, I have V1.2, however i do find it strange that only less than 2GB Card work with this Board.

TheGadge, 2019-10-28:
I presume they’re exact same format?

damvcoool, 2019-10-28:
Yes, I format all my cards using Windows FAT32, with 64K allocation 
[image]

TheGadge, 2019-10-28:
Well this is worrying. I do hope they’re not presuming everyone uses octoprint etc, which I do, but many don’t. I’m currently looking into using an adafruit neopixel stick under the extruder housing and making use of our new little socket, so that’s me today lol. Hopefully we’ll hear back soon about this issue.

damvcoool, 2019-10-28:
I will try to format the cards again using a different allocation unit, however these work fine with the Creality 1.1.3 board.
Only SD cards bigger than 8GB seem to be affected.

dani1995ar, 2019-10-31:
I might be wrong, but I think this is bebcause 32bit processors can only handle up to 4 GB of memory. I know this sd is not RAM but it is somehting that came to my mind while reading this post

damvcoool, 2019-11-01:
no that's not it...specially since the original creality board uses an 8 Bit processor.

swilkens, 2019-11-10:
F.Y.I. 

SD cards with a capacity above 2 Gb are "High Capacity" or "eXtended Capacity" cards, they are physically different than the < 2Gb cards and require newer hardware to read them.

damvcoool, 2019-11-10:
> F.Y.I.
> 
> SD cards with a capacity above 2 Gb are "High Capacity" or "eXtended Capacity" cards, they are physically different than the < 2Gb cards and require newer hardware to read them.

Does this mean that the SKR Mini E3 v1.2 is not compatible with SD HC?

TheGadge, 2019-11-10:
“Newer hardware” better be in a board that’s less than a month old. :/

lococnc, 2019-11-14:
I am using a 16GB with success. However I do have to reformat every few times I load a firmware.bin file as it seems to corrupt the card

dp250f, 2020-01-15:
> 
> 
> > F.Y.I.
> > SD cards with a capacity above 2 Gb are "High Capacity" or "eXtended Capacity" cards, they are physically different than the < 2Gb cards and require newer hardware to read them.
> 
> Does this mean that the SKR Mini E3 v1.2 is not compatible with SD HC?

I was able to use some spare pins to enable LCD SD card (SD_DETECT_PIN and LCD_SDSS), and even that cannot read my 16GB sd card. I thought SPI was SPI when it comes to SD card access, but I guess there's something more to it.