bigtreetech/BIGTREETECH-SKR-mini-E3 #67: Maple driver

FredLQ, 2019-10-26:
Windows 7 doesn't recognize the card, and i can't have it use the drivers supplied by bigtreetech. I have a device called maple in my peripherical list but it has no driver and when i tell windows to check into the directories i downloaded from github/bigtreetech, it says that it finds nothing :/

kramer04, 2019-11-01:
same here on windows 10.
in configuration.h these two lines are uncommented
#define SERIAL_PORT 2
#define SERIAL_PORT_2 -1
win10 detects connection on usb but says there is a malfunction

Kecajus, 2019-11-06:
my Win 10 recognize Maple 3, but cannot automatically install drivers (code :28).
When Manually installing 
> mapleDrv-USBdriver
from  gitHub is stops displaying "INF file doesnt have digital signature.
 lack of driver does not allow me to connect to pc :( ( pronterface etc.:)