bigtreetech/BIGTREETECH-SKR-mini-E3 #68: Just fitted v1.2 mini. V6 trianglelabs hotend fluctuations after several pids autos?

TheGadge, 2019-10-27:
Hi. I've just fitted this. Had a v6 on it for months without problems.
I've added the BTT firmware with 0 5 defined as required.

I'm getting variations of up to 10 degrees with loads of autotune done, adjusted in marlin 2.x and it's not really changed. I can't print because the temp won't settle even close.

Also, why is bltouch enabled by default? Should I disable soft pwm as I don't use one, yet.
Many thanks.
[image]

timryder, 2019-10-27:
You're not alone. I have the same issues with my stock hot end and I can't print either. I've Auto tuned everything till I'm blue in the face...I suspect it's an electrical issue on the board. I'm investigating it myself and I'll pay my findings.

timryder, 2019-10-27:
I just switched the two wires on my HotEnd and added a 10uF capacitor right at the screw terminals where the HotEnd plugs into the board.  I'm getting a nice clean signal which can be PID controlled permenantly.  

Start by removing the wires from the screw terminals on the board and manually pry open them. If you look they're like a clamp style terminal not a gate like high quality ones. So shove something in there to open it all the way up. Then even though it should not matter, SWITCH the two HotEnd wires and make sure you shove ALL the exposed wire in the screw terminals and screw down HARD. Making darn sure there's a good connection.  

Report back if that fixes it for you.  If not then the capacitor option....

closed 2019-11-01 by TheGadge