bigtreetech/BIGTREETECH-SKR-mini-E3 #69: I just had this issue and almost start a fire.  Started heating up the system to PLA temperatures using the controller, plugged in my USB.  I Stepped away for a minute to fill my coffe mug. I came back to a horrible smell and smoking hot-end. and the screen frozen up at 72c on then head. Oddly the bed was cold . After a few checks on a second printer, I found the board locks up when plugging in the USB from my pi or laptop. Also if you set a temp from the controller and plug into the USB it keeps powering the nozzle  while locked.  My ender3 block and heater at toast now.

ndemar, 2019-10-27:
I just had this issue and almost start a fire.  Started heating up the system to PLA temperatures using the controller, plugged in my USB.  I Stepped away for a minute to fill my coffe mug. I came back to a horrible smell and smoking hot-end. and the screen frozen up at 72c on then head. Oddly the bed was cold . After a few checks on a second printer, I found the board locks up when plugging in the USB from my pi or laptop. Also if you set a temp from the controller and plug into the USB it keeps powering the nozzle  while locked.  My ender3 block and heater at toast now.

_Originally posted by @ndemar in https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/40#issuecomment-546716942_

maddingamer, 2019-10-28:
#40

maddingamer, 2019-10-28:
sounds similar to #40

maddingamer, 2019-10-28:
#40

Waxonwaxoffmyson, 2020-04-26:
Same thing just happened to me. I have a raspberry pi connected with the 5v blocked. Printer froze with the heated bed stuck on. Print was melted and printer was just frozen. When I power cycled it. It displayed the thermal runaway warning.  Octoprint had a communication error displayed.

Were you able to solve this?