bigtreetech/BIGTREETECH-SKR-mini-E3 #70: Screen doesn't turn on when all five cables of BLTouch are connected

Rogerz79, 2019-10-28:
I am fairly certain that the cables are in the correct position and not shorted to each other, When ever I plug in both cable for the BLTouch v3,1 into the SKR Mini v1,2 the screen refuses to power on, If i unplug either the Z end stop cable or the servo cable the scree turns on and the system boots normally. Any help is appreciated

bojanpotocnik, 2019-10-28:
Black and brown/blue wires (both GND) of the BLTouch are connected together in the BLTouch itself. That means that if you do not connect the brown wire and only connect other 4, the BLTouch would also work (and the same goes for leaving black disconnected while connecting brown).

For me, the first (and only, in fact) cause of the problem you are describing is that you have +5V of the BLTouch connected to the GND on the board. And then when you connect the other 2 pins (the probe), you basically short +5V via BLTouch to the GND.
This would happen if you connected the most common (Antclabs, Trianglelab, ...) BLTouch as shown on [this image](https://github.com/gazcbm/Marlin-2.0.x-SKR-Mini-E3-v1.2/raw/master/docs/crealitybltouch.jpg) instead of as shown on [this image](https://github.com/gazcbm/Marlin-2.0.x-SKR-Mini-E3-v1.2/raw/master/docs/1570702845960.png) (source [gazcbm/Marlin-2.0.x-SKR-Mini-E3-v1.2](https://github.com/gazcbm/Marlin-2.0.x-SKR-Mini-E3-v1.2/blob/master/README.md)).

See also [gazcbm/#4](https://github.com/gazcbm/Marlin-2.0.x-SKR-Mini-E3-v1.2/issues/4), #43 and #58, which are all basically the same problem. There is certainly some confusion going on with the BLTouch connection, which I understand as neither the schematic and pin desription does not match (#71).

Rogerz79, 2019-10-28:
you sir were 100% correct, Thank you!

bojanpotocnik, 2019-10-29:
I would suggest closing the issue if the problem is solved.
Maybe you even add a sentence what exactly was the cause (blue was in the middle pin instead of on the left?) for others searching the same issue.

MarcusDLG, 2019-10-29:
Not solved! I have the same issue and my bltouch is wired appropriately. If i flash the firmware, the lcd lights up and shows text, but as soon as I power cycle the lcd lights up but fails to show any text. Just unplugged the bltouch and power cycled and still nothing.

bojanpotocnik, 2019-10-29:
So currently it does or does not matter if the BLTouch is plugged or unplugged?
If you put this [firmware.bin](https://github.com/bojanpotocnik/Marlin/raw/bugfix-2.0.x_SKR-mini-E3-V1.2/.pio/build/STM32F103RC_bigtree/firmware.bin) on the SD card and then turn on the Ender 3, does the LCD show logo after a 10-20 seconds? Try with BLTouch connected and disconnected.

TheGadge, 2019-10-29:
Sorry to jump in here @bojanpotocnik but is there some kind of noob guide for setting this up on the 1.2 board please? I get the placement of wires from your example above, the 2nd image being correct? But do I completely replace the z stop with the bl touch then? And firmware changes etc. I've managed to customise the firmware for trianglelabs v6 with great results with petg, so pretty confident of changing marlin. Many thanks.

MarcusDLG, 2019-10-29:
@bojanpotocnik So I've isolated this incident to the SD card and I'm opening a new ticket. I have an 8gb card and everytime I initialize the sd card after turning on it causes the freeze. Troubleshooted with the 128mb card that came with the board and was still able to replicate. Once the SD card is inserted and "initialize media" has been pressed, it freezes, the sound in the attached video begins to play, all stepper motors freeze and my screen becomes unresponsive until I reflash firmware.

MarcusDLG, 2019-10-29:
I know this is getting repetitive with me replying to myself, but here's my idea and hopefully someone chimes in to agree. 

My firmware build size was 229kb, and I saw on a different thread that it needs to be under 224. I was triggering the sd card issue everytime I was trying to write to eeprom. Testing now and will update this entry with findings.

Rogerz79, 2019-10-29:
*Gasp* My issue thread has been high-jacked! but its all good. Im glad marcus got his board working. as for mr gage yes, you do completely replace the z stop plug with the bl touch. I pulled the wire completely off from the z stop switch and have nothing coming out of the top of my mother board box. I used this video as a reference. 
https://www.youtube.com/watch?v=sUlqrSq6LeY&t=9s
https://www.youtube.com/watch?v=-XUQKQnUNig&t=481s 
just note that the wiring is wrong in the video as pointed out by our good friend here bojan and to compile the firmware used the guide here from big tech tree but their marlin wouldn't compile for me so i got a fresh one from marlin.  
[firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/3786098/firmware.zip)
i ended up with this bin file and your welcomed to use it. note that you might still have to change your x and y position of your bltouch in the firmware. mine was -43, -16

sazzwi, 2019-10-30:
Not sure if this is related or if anyone has seen this, hoping someone has a simple answer, I have a 1.2 SKR mini e3 board with a v3.0 Bltouch (from Creality) on a Ender 3 and no luck with the above firmware.   It loads, screen works and the BL touch does the self test.  When I attempt to level the bed the the pin never extends.  This worked with the 1.1.4 board.

bojanpotocnik, 2019-10-30:
@sazzwi can you try with [this firmware](https://github.com/bojanpotocnik/Marlin/tree/bugfix-2.0.x_SKR-mini-E3-V1.2_BLTouch) (there is a link under the image about BLTouch connection). If it won't work then create a new issue so we stop kidnapping this one which is in fact resolved.

closed 2019-11-01 by Rogerz79

Oniketsu, 2020-05-20:
> Black and brown/blue wires (both GND) of the BLTouch are connected together in the BLTouch itself. That means that if you do not connect the brown wire and only connect other 4, the BLTouch would [...]
----
Just to clarify, if you have a Creality version of the Bltouch the wiring is wrong, and the blue and red wires have to be swapped on this board?