bigtreetech/BIGTREETECH-SKR-mini-E3 #74: Can't Change Max Temp

space-commits, 2019-10-28:
I tried changing the max hotend temp to 300c, but through the LCD screen I can't go above 260c. To test I changed the PLA pre-heat temps and that didn't change either.

I did get a firmwar.cur file so it seems the firmware was booted.

What am I doing wrong?

timryder, 2019-10-29:
In Configuration.h search for "#define HOTEND_0_MAXTEMP". Set values appropriately.

ADS312, 2019-11-06:
I have a BIGTREETECH SKR PRO v1.1 running Marlin 2.0 bugfix.   I have exactly the same problem.  I have Marlin configured to a max_temp of 275.   However, no matter via console or gcode setting to, as an example, 270, it will max at 260 every time.   Also, on the manual LCD controller, it will not let me go above 260 for the nozzle temp setting.   Other than #define HOTEND_0_MAXTEMP are there any other places in the firmware that could limit the temp setting?

wm16685, 2020-02-11:
I am having the exact same problem!

wm16685, 2020-02-11:
So I think I figured out the issue for some if not all of us. The problem with me is that I am using the touchscreen with the board  (tft35 v3) and I am guessing you guys are too. If so you have to copile firmware for the touchscreen and change the #define HEAT_MAX_TEMP in the config.h file to whatever you need it to. You are going to want to change the 2nd number as the first is the bed temp. Of course you also have to change this for the firmware for the board too. The board will allow it to go to the set temp but the screen won't until you do this and it worked for me. I hope this helps you guys as this took me a while to figure out but it is the solution for most of us I think. If you have any questions feel free to email me at willemc1988@gmail.com and I will get back to you as I think I am an expert on this topic now lol.

Good Luck!

wm16685, 2020-02-11:
> So I think I figured out the issue for some if not all of us. The problem with me is that I am using the touchscreen with the board (tft35 v3) and I am guessing you guys are too. If so you have to copile firmware for the touchscreen and change the #define HEAT_MAX_TEMP in the config.h file to whatever you need it to. You are going to want to change the 2nd number as the first is the bed temp. Of course you also have to change this for the firmware for the board too. The board will allow it to go to the set temp but the screen won't until you do this and it worked for me. I hope this helps you guys as this took me a while to figure out but it is the solution for most of us I think. If you have any questions feel free to email me at [willemc1988@gmail.com](mailto:willemc1988@gmail.com) and I will get back to you as I think I am an expert on this topic now lol.
> 
> Good Luck!