bigtreetech/BIGTREETECH-SKR-mini-E3 #75: Missing binary operator before token "4013 [273, 39]

Vern-K, 2019-10-29:
Hello, help please, I'm newish to coding, I haven't played with it since the Apple IIGS...... I have a SKR mini E3 V1.2 on the way (maybe if it ever gets here) for my Ender 3. So far I have made the changes outlined in the instructions, with changes made to a thermistor table for the Slice Engineering 450C, which I made on my first try to take this on, since then the table shows up in the list, set it up as number 67 in the list. I was having other issues with software (as in using the wrong stuff and other such things) I had closed all that out and deleted the files to start over, had started in Atom then moved to VSC, couldn't find the temp table folder but #67 was showing up as 450C thermistor from SliceEngineering, so set it as that and changed the max temp and moved on. Then when I went to compile everything I got the 1 error. I'm more of a hardware person so please use small words and crayons please but would like some help with getting this ready and navigating around what I will need to know like how do I upload what would be needed so it can be seen what I'm messing up. Thanks

bojanpotocnik, 2019-10-29:
This error indicates that the preprocessor has encountered some invalid syntax in your preprocessor directives (#if, #endif, #define, etc.). Please archive changed files(s) and add them, so we can take a look. Or at least full error code description with file and line number.

Vern-K, 2019-10-29:
So that was the full error code, not sure where to find the file, hit save and went back and had some other errors, not really sure what I did, was in the middle of looking and had a client to take care of.... Having a few issues finding my way around like how to keep it from opening the stuff I want closed.... think I found what I needed to keep it from having V1.0 open... So once I find where I was working on how do I save it and what folder will I want to post?

Vern-K, 2019-10-29:
[Marlin-bugfix-2.0.xmodified.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/3786166/Marlin-bugfix-2.0.xmodified.zip)

ok I think this is it, got everything closed out, restarted computer since it's all been open for a few days... Looks like all my changes were saved, it's going to be the BTT SKR mini E3 V1.2 one

McPoopyPantz, 2019-10-29:
I have the same issue.. i can compile every other firmware, but when trying to compile marlin with this board it just simply errors out the same way

bojanpotocnik, 2019-10-29:
@Vern-K the preprocessor semantic error in your code is:
- `#define MOTHERBOARD MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V1_2` shall be `#define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V1_2` in Configuration.h

Other configuration errors/suggestions:
In platformio.ini:
 - `env_default = STM32F103RC_bigtree` shall be `default_envs = STM32F103RC_bigtree` (`[platformio]`)
- `#TMCStepper@>=0.5.0,<1.0.0` must be `https://github.com/bigtreetech/TMCStepper` (`[common]`)
- `#Adafruit NeoPixel@1.2.5` must be `https://github.com/bigtreetech/Adafruit_NeoPixel` (`[common]`)
- `platform          = ststm32` must be `platform          = ststm32@5.6.0` as there is already 5.7.0 (`[env:STM32F103RC_bigtree]`)

In Configuration.h:
`//#define SERIAL_PORT_2 -1` must be `#define SERIAL_PORT_2 -1` (uncommented)

I would suggest checking out (rebase) the latest Marlin bugfix-2.0.x which has this few fixes in place. You can also compare it with my [bugfix-2.0.x_SKR-mini-E3-V1.2](https://github.com/bojanpotocnik/Marlin/tree/bugfix-2.0.x_SKR-mini-E3-V1.2), which I rebase every few days from the Marlin upstream and then (re)apply the minimal configuration changes so that it works on SKR Mini E3.

Vern-K, 2019-10-30:
Ok going though and seeing if I can find everything, I know I changed a couple of the things via instructions, and the  #define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V1_2 part reads just as you posted it already, I'm only getting one error and it's showing it in boards.h Marlin\src\core. Have you added the info for the Slice Engineering thermistor? I'm going to try and find these changes first then download the new file since I should work on finding my way around this if I'm going to be modifying my set up...

Vern-K, 2019-10-30:
[image]
Still getting the same error

bojanpotocnik, 2019-10-30:
> the #define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V1_2 part reads just as you posted it already

Sorry, I have written my semantic error description in a way that the actual difference can be overlooked:
- `#define MOTHERBOARD MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V1_2` shall be
- `#define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V1_2`

note the double `MOTHERBOARD` in your code
> [Marlin-bugfix-2.0.xmodified.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/3786166/Marlin-bugfix-2.0.xmodified.zip)

Preprocessor directives are directly replaced with their definition. And unfortunately, use of an undefined macro in an `#if` is substituted with 0 (unless warnings are turned on with `-Wundef`), that is why it is important to check if they are even defined with `#ifdef` or `#if defined()`. Marlin does that for the `MOTHERBOARD` macro, but not for its content.

Lets take a look at what happens in the code: 
```c
#if defined(MOTHERBOARD) && (MOTHERBOARD == BOARD_BTT_SKR_MINI_E3_V1_2)
    #include "stm32/pins_BTT_SKR_MINI_E3_V1_2.h"
#endif // Actually #elseif, but isolate only this conditional for simplicity
```
`defined(MOTHERBOARD)` is **`true`** as it for sure is defined as `MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V1_2`.
Now comes the first pass of preprocessor substitution, where the macros are replaced with their definitions, in our case:
- `#define MOTHERBOARD MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V1_2`
- `#define BOARD_BTT_SKR_MINI_E3_V1_2    4013  // BigTreeTech SKR Mini E3 V1.2 (STM32F103RC)`

so we have:
```c
#if defined(MOTHERBOARD) && (MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V1_2 == 4013)
    #include "stm32/pins_BTT_SKR_MINI_E3_V1_2.h"
#endif
```
now comes the second pass as there are still macros to be replaced. Note that `MOTHERBOARD` was undefined at the `#define MOTHERBOARD MOTHERBOARD BOARD_BT...`, so it is substituted with 0:
```c
#if defined(MOTHERBOARD) && (0 4013 == 4013)
    #include "stm32/pins_BTT_SKR_MINI_E3_V1_2.h"
#endif
```
and now the preprocessor sees:
```c
#if (0 4013 == 4013)
```
and it is like - there is a missing operator before `4013`. Shall it be `0 + 4013`, `0 * 4013`, `0 || 4013`, `0 && 4013`, what? It does not know what to do, but it is trying to help you at its best - showing you that the `4013` comes from the definition `BOARD_BTT_SKR_MINI_E3_V1_2`.

Vern-K, 2019-10-30:
@bojanpotocnik Thank you, it was a success but did come up with 5 other errors, is it something I should just leave, or where and what should I comment out to get rid of them?  Now if I hit save it should save it in the master folder so when it comes time to load it I just open it and put the file (will have to rewatch video to see file name) and put that on a SD card and put that in and power on right? Where would I double check the info on #67 temp sensor to make sure it has the right info?
[image] 

Thanks again, was kind of a duh moment for me not catching the double motherboard right away... Now I just need to go though the stuff for the TFT35 that I have coming with the board

closed 2019-10-30 by Vern-K