bigtreetech/BIGTREETECH-SKR-mini-E3 #77: Make displays other than CR10_stockdisplay work with E3 mini too...

McPoopyPantz, 2019-10-30:
I tried to use the board with my Ender-2 and this was such a negative surprise...
Ender-2 uses MKS_MINI_12864

opiswahn, 2019-11-02:
Second that! Please make that work FAST...

GuzLightyear, 2019-11-06:
I'm trying to make the MKS_MINI_12864 from my CR-20 work with the board. So far, i have only achieved a dotted screen which i think means the screen gets initialised but the SPI communication isn't working as expected. So far i have added this

`  #elif ENABLED(MKS_MINI_12864)
    #define BEEPER_PIN     PA15
    #define BTN_ENC       PB6
    #define KILL_PIN      PB7
    #define DOGLCD_A0     PB8
    #define DOGLCD_CS     PB9
    // #define DOGLCD_SCK     PA5
    // #define DOGLCD_MOSI    PA7
    // #define SCK_PIN          PA5
    // #define MISO_PIN         PA6
    // #define MOSI_PIN         PA7
    // #define ENABLE_SPI1
    #define FORCE_SOFT_SPI   // SPI MODE3

    // GLCD features
    // Uncomment screen orientation
    //#define LCD_SCREEN_ROT_90
    //#define LCD_SCREEN_ROT_180
    //#define LCD_SCREEN_ROT_270

    // not connected to a pin
    #define LCD_BACKLIGHT_PIN -1

    #define BTN_EN2        PA9
    #define BTN_EN1        PA10`

below the   #if ENABLED(CR10_STOCKDISPLAY) in the pins definition of the board, and wired the board accordingly using this page as reference:
https://reprap.org/wiki/MKS_MINI_12864

I'm using the SPI header just above the expansion header to get the MOSI and SCK signals, but i have not had success yet.

ufficioprogettiperduti, 2019-12-25:
This is an urgent matter! We want to be able to use the board with other printers!

GuzLightyear, 2019-12-25:
> 
> 
> This is an urgent matter! We want to be able to use the board with other printers!

Sorry, I did not update the query. It's working on the latest Marlin 2.0 revision, [to quote from my reddit post](https://www.reddit.com/r/BIGTREETECH/comments/dtliyz/skr_mini_e3_and_mks_mini_12864_lcd/f8bripb/?context=3)

I got it working! In the latest revision of marlin there's already support implemented for the MKS_MINI_12864 in the SKR Mini E3. You just need to use individual cables to wire the single connector of the board to the double connectors of the screen, but it works.

You can use the pinouts in this page: https://reprap.org/wiki/MKS_MINI_12864 to get the pins on the screen, and the pins that are on pins_BTT_SKR_E3_DIP.h (take note that the pins here are seen as you would see them if the board had the stepper drivers on the bottom)

/** Creality Ender-2 display pinout
 *                   _____
 *               5V | · · | GND
 *      (MOSI) PB7  | · · | PB8  (LCD_RS)
 *    (LCD_A0) PB9  | · · | PA10 (BTN_EN2)
 *            RESET | · · | PA9  (BTN_EN1)
 *   (BTN_ENC) PB6  | · · | PA15 (SCK)
 *                   -----
 *                    EXP1
 */

You'll have to set your screen to MKS_MINI_12864 on Configuration.h

And i got it working finally.